llvm/circt #682: Handle Registers in LowerTypes

seldridge, 2021-02-25:
Aggregate *registers* are currently not being lowering in FIRRTL Dialect's type lowering pass. I think this is a straightforward addition of a visitor for wire ops in the pass.

Consider:
```scala
circuit Bar :
  module Bar:
    input a: {a: UInt<1>}
    input clk: Clock
    output b: {a: UInt<1>}

    reg x: {a: UInt<1>}, clk

    x.a <= a.a
    b.a <= x.a
```

This is currently crashing when running:

```bash
circt-translate -import-firrtl Bar.fir | circt-opt -pass-pipeline='firrtl.circuit(firrtl.module(firrtl-lower-types))'
```

With:

```
Assertion failed: (entry && "bundle lowering was not set"), function getBundleLowering, file ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp, line 524.
```

This is basically the same thing as #680, just split out to potentially expose more parallel work.

lattner, 2021-02-28:
Random comment, but in addition to this bug, we should make LowerTypes clealy emitOpError on things it doesn't know about yet.  This will make it more nice/resilient to future firrtl dialect extensions.

seldridge, 2021-03-02:
Closed via #686.

closed 2021-03-02 by seldridge