llvm/circt #701: [FIRRTL] wire inside always block

drom, 2021-03-02:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_a: {inp_b: {inp_j: SInt<11>}, inp_f: SInt<19>}
    reg _tmp6: UInt<25>, clock with: (reset => (reset, UInt<25>(0)))
    _tmp6 <= and(inp_a.inp_b.inp_j, SInt<25>(12345))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// skipped defines
module top_mod(
  input        clock, reset,
  input [10:0] inp_a_inp_b_inp_j,
  input [18:0] inp_a_inp_f);

  reg [24:0] _tmp6;	// a_top_mod.fir:6:5

  always_ff @(posedge clock) begin	// a_top_mod.fir:6:5
    if (reset)	// a_top_mod.fir:6:5
      _tmp6 <= 25'h0;	// a_top_mod.fir:6:{5,56}
    else	// a_top_mod.fir:6:5
      wire [24:0] _tmp = {{14{inp_a_inp_b_inp_j[10]}}, inp_a_inp_b_inp_j};	// a_top_mod.fir:7:14
      _tmp6 <= _tmp & 25'h3039;	// a_top_mod.fir:7:{11,14,37}
  end // always_ff @(posedge)
  `ifndef SYNTHESIS	// a_top_mod.fir:6:5
    initial begin	// a_top_mod.fir:6:5
      `INIT_RANDOM_PROLOG_	// a_top_mod.fir:6:5
      `ifdef RANDOMIZE_REG_INIT	// a_top_mod.fir:6:5
        if (~reset)	// a_top_mod.fir:6:5
          _tmp6 = `RANDOM;	// a_top_mod.fir:6:5
      `endif
    end // initial
  `endif
  wire [24:0] _tmp = {{14{inp_a_inp_b_inp_j[10]}}, inp_a_inp_b_inp_j};	// a_top_mod.fir:7:14
endmodule
```

An output Verilog has an error, declaring `wire` iside `always_ff` block.

```verilog
%Error: a_top_mod_new.sv:66:7: syntax error, unexpected wire
   66 |       wire [24:0] _tmp = {{14{inp_a_inp_b_inp_j[10]}}, inp_a_inp_b_inp_j};  
      |       ^~~~
```

darthscsi, 2021-03-02:
Additionally the else statement has two statements and no begin/end

lattner, 2021-03-04:
ooh, this is definitely my bug.  Will take a look ,thanks!

lattner, 2021-03-04:
The begin/end problem is fixed in 867438f

lattner, 2021-03-04:
I mean c46ec9e

lattner, 2021-03-04:
Fixed in c29ca85, thanks for tracking this down!

closed 2021-03-04 by lattner