llvm/circt #706: Support for inferring result types

GeorgeLyon, 2021-03-02:
Operations like `array_create`, `array_get`, `constant`, and likely many others can participate in result type inference. I haven't dug into this enough to figure out what is needed to enable this but the C API entry point is here: https://github.com/llvm/llvm-project/blob/9a2e2de15f108f943ae50e6183719a4af81104e8/mlir/include/mlir-c/IR.h#L259

I wouldn't mind if someone took a stab at this :)

GeorgeLyon, 2021-03-18:
Here is a link to some relevant tablegen: https://github.com/llvm/llvm-project/blob/5a8d5a2859d9bb056083b343588a2d87622e76a2/mlir/include/mlir/Interfaces/InferTypeOpInterface.td#L32

fabianschuiki, 2021-05-20:
I have started to look into this in the context of FIRRTL operations. The interface is pretty handy and adds a nice new builder method. Usage in a parser is a bit awkward at the moment because the generated builder aborts with an assertion on type inference failure rather than giving the parser a chance to exit gracefully. (Started a [discussion upstream](https://llvm.discourse.group/t/fallible-op-builders-and-return-type-inference/3504) to see if there's any interest to streamline this a bit).

lattner, 2021-05-20:
nice

lattner, 2021-05-21:
It is unrelated to the firrtl dialect, but I'd also love to see this get adopted by the comb dialect, to clean up the builders for things like comb::ExtractOp

fabianschuiki, 2021-05-21:
Yes that would be great! I guess almost all comb ops would benefit from this, e.g. `AddOp` and others which just trivially forward their one op type (if that's not handled by the same return type intf yet).

closed 2024-09-14 by GeorgeLyon