llvm/circt #708: [CI] Unable to pass LLHD tests when building with GCC and BUILD_SHARED_LIBS

youngar, 2021-03-02:
When adding the `BUILD_SHARED_LIBS` option to the GCC build job in our CI, there is a problem with the JIT not being able to find symbols defined by the LLHD wrappers library:

```
JIT session error: Symbols not found: [ allocEntity, allocSignal, driveSignal ]
Failed invocation of llhd_init: Failed to materialize symbols: { (main, { _mlir_llhd_init, root, llhd_init, _mlir_root }) }Program aborted due to an unhandled Error:
Failed to materialize symbols: { (main, { _mlir_llhd_init, root, llhd_init, _mlir_root }) }
PLEASE submit a bug report to bugs.llvm.org and include the crash backtrace.
```

Some notes:
1. In this build MLIR is built with Clang, while CIRCT is built with GCC
2. `circt-llhd-signals-runtime-wrappers` is a SHARED library in CMakeLists.txt, so `BUILD_SHARED_LIBS` should not affect how this library itself is built.
3. Setting `__attribute__((visibility("default")))` on each function did not fix the problem
4. There is the same problem, but on Windows here: https://github.com/llvm/circt/issues/375

mikeurbach, 2021-03-18:
I was just testing this build scenario because I'd like the Python bindings to build whenever BUILD_SHARED_LIBS is enabled. It seems like there is a solution here, which I just hacked up. Basically, LLHD is using the MLIR execution engine in these tests, and there is a parameter to specifically point to some shared libraries for the execution engine to load. When I specifically give the path to libcirct-llhd-signals-runtime-wrappers.so to this parameter, these tests pass.

It looks like in MLIR they are plumbing through a command line option to specify such shared libraries: https://github.com/llvm/llvm-project/blob/main/mlir/test/mlir-cpu-runner/async.mlir#L10. I will try to do the same thing here, and add an option to `llhd-sim` similar to how `mlir-cpu-runner` works upstream. I guess `llhd-sim` will work for the use-cases it currently works for without the option, but by specifying it, we should be able to fix this test and the windows issue.

mikeurbach, 2021-03-19:
Closed by #789

closed 2021-03-19 by mikeurbach