llvm/circt #709: Fix GCC Warnings

youngar, 2021-03-02:
There are currently some warnings when building with GCC, listed below.  Most of these should be pretty easy to fix.

1. Control reaches end of `sv::CaseZOp::getLetter`
```
In file included from /home/runner/work/circt/circt/include/circt/Dialect/SV/SVOps.h:43:0,
                 from /home/runner/work/circt/circt/lib/Dialect/SV/SVOps.cpp:13:
/home/runner/work/circt/circt/build_gcc/include/circt/Dialect/SV/SV.h.inc: In static member function ‘static char circt::sv::CaseZOp::getLetter(circt::sv::CaseZOp::PatternBit, bool)’:
/home/runner/work/circt/circt/build_gcc/include/circt/Dialect/SV/SV.h.inc:609:5: warning: control reaches end of non-void function [-Wreturn-type]
     }
```

2.  In `firrtl::verifyMemOp()` portNamesSet may be uninitialized

This problem manifests multiple times, possibly everywhere SmallDenseSet is used.

```
In file included from /home/runner/work/circt/circt/llvm/install/include/mlir/Support/InterfaceSupport.h:17:0,
                 from /home/runner/work/circt/circt/llvm/install/include/mlir/IR/StorageUniquerSupport.h:16,
                 from /home/runner/work/circt/circt/llvm/install/include/mlir/IR/AttributeSupport.h:17,
                 from /home/runner/work/circt/circt/llvm/install/include/mlir/IR/Attributes.h:12,
                 from /home/runner/work/circt/circt/llvm/install/include/mlir/IR/BuiltinAttributes.h:12,
                 from /home/runner/work/circt/circt/include/circt/Dialect/FIRRTL/FIRRTLDialect.h:17,
                 from /home/runner/work/circt/circt/include/circt/Dialect/FIRRTL/FIRRTLOps.h:16,
                 from /home/runner/work/circt/circt/lib/Dialect/FIRRTL/FIRRTLOps.cpp:13:
/home/runner/work/circt/circt/llvm/install/include/llvm/ADT/DenseMap.h: In function ‘mlir::LogicalResult verifyMemOp(circt::firrtl::MemOp)’:
/home/runner/work/circt/circt/llvm/install/include/llvm/ADT/DenseMap.h:1165:18: warning: ‘*((void*)(& portNamesSet)+16).llvm::SmallDenseMap<mlir::Attribute, llvm::detail::DenseSetEmpty, 8, llvm::DenseMapInfo<mlir::Attribute>, llvm::detail::DenseSetPair<mlir::Attribute> >::LargeRep::NumBuckets’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     return Small ? InlineBuckets : getLargeRep()->NumBuckets;
            ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/runner/work/circt/circt/llvm/install/include/llvm/ADT/DenseMap.h:1156:18: warning: ‘*((void*)(& portNamesSet)+16).llvm::SmallDenseMap<mlir::Attribute, llvm::detail::DenseSetEmpty, 8, llvm::DenseMapInfo<mlir::Attribute>, llvm::detail::DenseSetPair<mlir::Attribute> >::LargeRep::Buckets’ may be used uninitialized in this function [-Wmaybe-uninitialized]
     return Small ? getInlineBuckets() : getLargeRep()->Buckets;
            ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

3. Possibly uninitialized value in `firrtl::parseType()`
```
/home/runner/work/circt/circt/lib/Dialect/FIRRTL/FIRRTLTypes.cpp: In function ‘mlir::ParseResult parseType(circt::firrtl::FIRRTLType&, mlir::DialectAsmParser&)’:
/home/runner/work/circt/circt/lib/Dialect/FIRRTL/FIRRTLTypes.cpp:148:35: warning: ‘fieldIntName’ may be used uninitialized in this function [-Wmaybe-uninitialized]
             nameStr = llvm::utostr(fieldIntName);
```

4. Possibly uninitialized value in `handshake::executeHandshakeFunction()`

```
[ 75%] Building CXX object lib/Dialect/LLHD/Simulator/CMakeFiles/obj.CIRCTLLHDSimEngine.dir/Engine.cpp.o
/home/runner/work/circt/circt/tools/handshake-runner/Simulation.cpp: In function ‘void executeHandshakeFunction(circt::handshake::FuncOp&, llvm::DenseMap<mlir::Value, llvm::Any>&, llvm::DenseMap<mlir::Value, double>&, std::vector<llvm::Any>&, std::vector<double>&, std::vector<std::vector<llvm::Any> >&, std::vector<double>&)’:
/home/runner/work/circt/circt/tools/handshake-runner/Simulation.cpp:568:27: warning: ‘time’ may be used uninitialized in this function [-Wmaybe-uninitialized]
       timeMap[out] = time + 1;
```

5. Control reaches end of `handshake::getFIRRTLType`

```
[ 84%] Building CXX object lib/Conversion/HandshakeToFIRRTL/CMakeFiles/obj.CIRCTHandshakeToFIRRTL.dir/HandshakeToFIRRTL.cpp.o
/home/runner/work/circt/circt/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp: In lambda function:
/home/runner/work/circt/circt/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp:62:7: warning: control reaches end of non-void function [-Wreturn-type]
       })
```

6. Control reaches end of `handshake::StdExprBuilder::visitStdExpr()`

```
/home/runner/work/circt/circt/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp: In member function ‘bool {anonymous}::StdExprBuilder::visitStdExpr(mlir::CmpIOp)’:
/home/runner/work/circt/circt/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp:633:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
```
 
7. `ESIRTLBuilder::declareCosimEndpoint()` defined but not used
```
/home/runner/work/circt/circt/lib/Dialect/ESI/ESIPasses.cpp:190:19: warning: ‘circt::rtl::RTLModuleExternOp {anonymous}::ESIRTLBuilder::declareCosimEndpoint()’ defined but not used [-Wunused-function]
 RTLModuleExternOp ESIRTLBuilder::declareCosimEndpoint() {
                   ^~~~~~~~~~~~~
[ 89%] Built target obj.CIRCTFIRRTLTransforms
```

8. FIRParser::parseGetLocation(mlir::Location&)’ defined but not used 

```
/home/runner/work/circt/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:268:13: warning: ‘mlir::ParseResult {anonymous}::FIRParser::parseGetLocation(mlir::Location&)’ defined but not used [-Wunused-function]
 ParseResult FIRParser::parseGetLocation(Location &loc) {
             ^~~~~~~~~
/home/runner/work/circt/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:263:13: warning: ‘mlir::ParseResult {anonymous}::FIRParser::parseGetLocation(llvm::SMLoc&)’ defined but not used [-Wunused-function]
 ParseResult FIRParser::parseGetLocation(SMLoc &loc) {
             ^~~~~~~~~
In file included from /home/runner/work/circt/circt/include/circt/Dialect/SV/SVOps.h:43:0,
                 from /home/runner/work/circt/circt/lib/Translation/ExportVerilog/ExportVerilog.cpp:19:
/home/runner/work/circt/circt/build_gcc/include/circt/Dialect/SV/SV.h.inc: In static member function ‘static char circt::sv::CaseZOp::getLetter(circt::sv::CaseZOp::PatternBit, bool)’:
/home/runner/work/circt/circt/build_gcc/include/circt/Dialect/SV/SV.h.inc:609:5: warning: control reaches end of non-void function [-Wreturn-type]
     }
```

9. Various function in Export verilog not used

```
/home/runner/work/circt/circt/lib/Translation/ExportVerilog/ExportVerilogFIRRTL.cpp:961:13: warning: ‘{anonymous}::SubExprInfo {anonymous}::ExprEmitter::emitUnary(mlir::Operation*, const char*, bool)’ defined but not used [-Wunused-function]
 SubExprInfo ExprEmitter::emitUnary(Operation *op, const char *syntax,
             ^~~~~~~~~~~
/home/runner/work/circt/circt/lib/Translation/ExportVerilog/ExportVerilogFIRRTL.cpp:940:13: warning: ‘{anonymous}::SubExprInfo {anonymous}::ExprEmitter::emitVariadic(mlir::Operation*, {anonymous}::VerilogPrecedence, const char*)’ defined but not used [-Wunused-function]
 SubExprInfo ExprEmitter::emitVariadic(Operation *op, VerilogPrecedence prec,
             ^~~~~~~~~~~
/home/runner/work/circt/circt/lib/Translation/ExportVerilog/ExportVerilogFIRRTL.cpp:875:13: warning: ‘{anonymous}::SubExprInfo {anonymous}::ExprEmitter::emitBinary(mlir::Operation*, {anonymous}::VerilogPrecedence, const char*, bool)’ defined but not used [-Wunused-function]
 SubExprInfo ExprEmitter::emitBinary(Operation *op, VerilogPrecedence prec,
             ^~~~~~~~~~~
```

9.  Control reaches end of firtool's `processBuffer()`

```
/home/runner/work/circt/circt/tools/firtool/firtool.cpp: In function ‘mlir::LogicalResult processBuffer(std::unique_ptr<llvm::MemoryBuffer>, llvm::raw_ostream&)’:
/home/runner/work/circt/circt/tools/firtool/firtool.cpp:173:1: warning: control reaches end of non-void function [-Wreturn-type]
 };
```

lattner, 2021-03-02:
Wow ok.  Well it sounds like llvm_unreachable needs to be used in a bunch of places.  I'd recommend splitting off the DenseSet related issues to a discussion on the llvm phabricator.

fabianschuiki, 2021-03-24:
Updated list based on CI on 2021-03-24:

1.
    LLHD time parser:
    ```
    /home/runner/work/circt/circt/lib/Dialect/LLHD/IR/LLHDDialect.cpp: In static member function ‘static mlir::LogicalResult circt::llhd::TimeAttr::verifyConstructionInvariants(mlir::Location, mlir::Type, llvm::ArrayRef<unsigned int>, llvm::StringRef)’:
    /home/runner/work/circt/circt/lib/Dialect/LLHD/IR/LLHDDialect.cpp:481:21: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
   if (timeValues[0] < 0 || timeValues[1] < 0 || timeValues[2] < 0)
    /home/runner/work/circt/circt/lib/Dialect/LLHD/IR/LLHDDialect.cpp:481:42: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
   if (timeValues[0] < 0 || timeValues[1] < 0 || timeValues[2] < 0)
    /home/runner/work/circt/circt/lib/Dialect/LLHD/IR/LLHDDialect.cpp:481:63: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
   if (timeValues[0] < 0 || timeValues[1] < 0 || timeValues[2] < 0)
    ```
2.
    FIRRTL parser unused functions:
    ```
    /home/runner/work/circt/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:268:13: warning: ‘mlir::ParseResult {anonymous}::FIRParser::parseGetLocation(mlir::Location&)’ defined but not used [-Wunused-function]
     ParseResult FIRParser::parseGetLocation(Location &loc) {
                 ^~~~~~~~~
    /home/runner/work/circt/circt/lib/Dialect/FIRRTL/Import/FIRParser.cpp:263:13: warning: ‘mlir::ParseResult {anonymous}::FIRParser::parseGetLocation(llvm::SMLoc&)’ defined but not used [-Wunused-function]
     ParseResult FIRParser::parseGetLocation(SMLoc &loc) {
                 ^~~~~~~~~
    ```
3.
    ExportVerilog MergeOp emission function not being used:
    ```
    /home/runner/work/circt/circt/lib/Translation/ExportVerilog/ExportVerilog.cpp:1750:6: warning: ‘void {anonymous}::StmtEmitter::emitMergeOp(circt::comb::MergeOp)’ defined but not used [-Wunused-function]
     void StmtEmitter::emitMergeOp(MergeOp op) {
          ^~~~~~~~~~~
    ```
4.
    ESI RTL Builder unused function:
    ```
    /home/runner/work/circt/circt/lib/Dialect/ESI/ESIPasses.cpp:191:19: warning: ‘circt::rtl::RTLModuleExternOp {anonymous}::ESIRTLBuilder::declareCosimEndpoint()’ defined but not used [-Wunused-function]
     RTLModuleExternOp ESIRTLBuilder::declareCosimEndpoint() {
                       ^~~~~~~~~~~~~
    ```

darthscsi, 2021-03-26:
All cases closed.

closed 2021-03-26 by darthscsi