llvm/circt #718: TypeAlias type ?

teqdruid, 2021-03-04:
As part of the typedef conversation #645, @darthscsi came up with a [few ideas](https://github.com/llvm/circt/pull/645#pullrequestreview-601256121) which avoid the implicit typedef that I assumed in that PR. Although I initially favored an operation attribute (which would have to be on every op which could potentially become a Verilog statement), I now favor the wrapper type option to "annotate" an arbitrary type with a suggested name. I say suggested since the most straightforward implementation would have both name and structural equivalence semantics, so the same name could map to multiple types.

The tradeoff here is all operations would have to "look-through" the TypeAliasType since we don't want to have to unwrap and rewrap all the time. I don't know if there's a trivial way to do this. Incidentally, this pattern of "wrapper" types is becoming quite common. It'd be nice to support this "look-through" semantic on all annotation wrapper types, though I don't know how useful that'd be in practice.

I don't think we have any good options for this. How bad is this one?

teqdruid, 2021-03-10:
If I understood correctly, @lattner thinks this is the way to go. It introduces complexity, but there's likely no way around that. Is this a fair statement Chris?

mikeurbach, 2021-04-29:
I have started looking into this, and after doing the obvious work to define a new TypeAlias type in tablegen, I started playing around with it a bit. For motivation, I'm trying to name rtl.structs, and ultimately do something in ExportVerilog to use the pretty name. Leaving aside ExportVerilog, I have one consideration that I think is worth discussing at this point.

Given that I can define a type alias like `!rtl.typealias<foo, struct<...>>`, my question is how to actually use such a value in existing operations that don't know about type aliases. For example, consider a `rtl.struct_extract` from a `!rtl.struct` wrapped in `!rtl.typealias`:

```mlir
func (%arg0: !rtl.typealias<foo, struct<a: i1>>) {
  %0 = rtl.struct_extract %arg0["a"]
}
```

This will complain very clearly:

```
'rtl.struct_extract' op operand #0 must be RTL struct type, but got '!rtl.typealias<foo, !rtl.struct<a: i1>>
```

Now, we already discussed having something like `rtl::getCanonicalType` to "see through" the type aliases, and I've implemented it. I can think of at least two different ways to use that to resolve this issue:

1. Define an operation that unwraps a value of `!rtl.typealias` type and returns a value of its canonical type. This would play really well with the dialect conversion framework, as the framework would materialize this operation when needed. Unfortunately, we don't really use the dialect conversion framework in circt, especially at the RTL level, so transforms seem to be on their own to introduce this operation where necessary.
2. Define a type constraint in ODS that allows either a) a specified RTL type, or b) a type alias whose canonical type is the specified RTL type. This keeps the problem confined in the type system, but any operation that has a specific type constraint, and might see a type alias, would need to have its current type constraint replaced by the wrapper type constraint mentioned here. If the operation definition uses custom parsers and printers, those may also need to be updated, or the operation switched to specify an `assemblyFormat` to let MLIR handle it.

What other approaches might work?

I'll discuss a little bit more about option 2, because I initially wanted to experiment this way, and so far the experiment has gone decently well.

The RTL dialects already has types for RTLIntegerType, RTLValueType, RTLNonInoutType, and I have already taught those to "see through" type aliases to the canonical type. So the problem is more confined to things like `!rtl.struct_extract` that explicitly name `StructType` as one of the operands. Let me show how it would look to support `StructType` or a `TypeAlias` that canonicalizes to `StructType`:

First, define a type constraint in ODS, parameterized by the desired inner type:

```mlir
class TypeAliasOr<Type innerType> :
    TypeConstraint<Or<[innerType.predicate, CPred<
        "getCanonicalType($_self).isa<" # innerType.cppClassName # ">()">]>>;
```

Then, define an instance of this type constraint for `StructType`:

```mlir
def StructOrAliasType : TypeAliasOr<StructType>;
```

Now, in definitions of operations, we can replace `StructType` with `StructOrAliasType`. This allows us to create operations like the above in a legal way. After having worked this through for `StructType` and `struct_extract`, it doesn't seem like the end of the world. This TypeAlias stuff is a new feature, and we can do this mechanical change either incrementally as new ops want to use it, or wholesale.

If anyone wants to take a look at the experimental branch, I've pushed all the changes, including the `TypeAliasOr` experiment here: https://github.com/llvm/circt/pull/983. At this point, I can work a type alias all the way down through the ESI passes and into ExportVerilog, and now I'm thinking about how to handle this there.

teqdruid, 2021-04-29:
I would prefer option 2, but I also cannot think of another approach. It sounded like @lattner had some experience with this -- IIRC in persisting C/C++ `typedef`s in clang.

mikeurbach, 2021-04-29:
I was looking at https://clang.llvm.org/docs/InternalsManual.html#canonical-types, which I think is what was mentioned in the ODM where we discussed this.

mikeurbach, 2021-09-08:
This was done a while ago, and I've opened https://github.com/llvm/circt/issues/1642 to track the remaining issue with this type.

closed 2021-09-08 by mikeurbach