llvm/circt #720: temporaries output in the middle of blocks

darthscsi, 2021-03-04:
Consider the test case below.  A temporary is made for 610, which is output after the first if block.  Verilog requires variable declarations to be at the beginning of blocks.

module {
  rtl.module @foobar(%clock: i1, %arg1: i1, %arg2: i1, %arg3: i1) {
    sv.always posedge %clock  {
        sv.if %arg1  {
          sv.fatal
        }
        //this forces a common subexpression to be output out-of-line
        %610 = comb.and %arg1, %arg2 : i1
        %611 = comb.and %arg3, %610 : i1
        sv.if %610  {
          sv.fatal
        }
        sv.if %611  {
          sv.fatal
        }
    }
    rtl.output
  }
}

darthscsi, 2021-03-04:
Section 9.8.1 of 1364-2005
https://www.eg.bucknell.edu/~csci320/2016-fall/wp-content/uploads/2015/08/verilog-std-1364-2005.pdf

lattner, 2021-03-04:
Verilog is really annoying, sigh.  I'll take care of this tonight.  Thank you for catching this!

lattner, 2021-03-05:
Fixed by be85f3c.  This is a big change though and may have regressed other things (I'm sorry in advance that I still don't know verilog).  This also broke mergeop emission when it is in a procedural region, but I don't think that will be a blocking issue.   That is tracked by Issue 726

closed 2021-03-05 by lattner

darthscsi, 2021-03-05:
Made a new bug for the remaining cast.  #728