llvm/circt #724: [ExportVerilog] Wire declaration and definition "_tmp" output twice

teqdruid, 2021-03-05:
I haven't tracked this down, but in a certain condition this line is emitted twice (identically) in different areas in the same scope:

```
  wire [31:0][7:0] _tmp = {{_T_9}, {_T_10}, {_T_11}, {_T_12}, {_T_13}, {_T_14}, {_T_15}, {_T_16}, {_T_17}, {_T_18}, {_T_19}, {_T_20}, {_T_21}, {_T_22}, {_T_23}, {_T_24}, {_T_25}, {_T_26}, {_T_27}, {_T_28}, {_T_29}, {_T_30}, {_T_31}, {_T_32}, {_T_33}, {_T_34}, {_T_35}, {_T_36}, {_T_37}, {_T_38}, {_T_39}, {_T_40}};
```

The wire name points to https://github.com/llvm/circt/blob/main/lib/Translation/ExportVerilog/ExportVerilog.cpp#L956. I'll look into this further later, but I figured there is a decent chance that would be enough for @lattner to know what's going on.

lattner, 2021-03-05:
I don't know offhand, do you have a testcase?

teqdruid, 2021-03-05:
Not a simple one... Yet.

teqdruid, 2021-03-05:
```
sv.interface @IValidReady_Struct  {
  sv.interface.signal @data : !rtl.struct<foo: !rtl.array<384xi1>>
}
rtl.module @structs(%clk: i1, %rstn: i1) {
  %0 = sv.interface.instance : !sv.interface<@IValidReady_Struct>
  sv.interface.signal.assign %0(@IValidReady_Struct::@data) = %s : !rtl.struct<foo: !rtl.array<384xi1>>
  %c0 = rtl.constant 0 : i8
  %c64 = rtl.constant 100000 : i64
  %c52 = rtl.constant 100000 : i52
  %16 = rtl.bitcast %c64 : (i64) -> !rtl.array<64xi1>
  %58 = rtl.bitcast %c0 : (i8) -> !rtl.array<8xi1>
  %90 = rtl.array_concat %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %58, %16, %16 : !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<8xi1>, !rtl.array<64xi1>, !rtl.array<64xi1>
  %s = rtl.struct_create (%90) : !rtl.struct<foo: !rtl.array<384xi1>>
}
```

teqdruid, 2021-03-05:
I'm not sure if this necessarily involves interfaces or not, I just whittled this down from a larger one which involved interfaces.

teqdruid, 2021-03-05:
The above testcase emits:

```
interface IValidReady_Struct;
  struct packed {logic [383:0] foo; } data;
endinterface

module structs(
  input clk, rstn);

  IValidReady_Struct                                     _T();  // testcase.mlir:5:8
  wire               [63:0]                              _T_0;  // testcase.mlir:10:9
  wire               [7:0]                               _T_1;  // testcase.mlir:11:9
  wire               struct packed {logic [383:0] foo; } _T_2;  // testcase.mlir:13:8

  wire [383:0] _tmp = {_T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_0, _T_0};     // testcase.mlir:12:9
  assign _T.data = '{foo: _tmp};
  assign _T_0 = 64'h186A0;      // testcase.mlir:8:10, :10:9
  assign _T_1 = 8'h0;   // testcase.mlir:7:9, :11:9
  wire [383:0] _tmp = {_T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_1, _T_0, _T_0};     // testcase.mlir:12:9
  assign _T_2 = '{foo: _tmp};   // testcase.mlir:13:8
endmodule
```

teqdruid, 2021-03-05:
If I move the `sv.interface.signal.assign` after the `struct_create`, it works.

teqdruid, 2021-03-05:
Also, if I comment out the block starting here https://github.com/llvm/circt/blob/main/lib/Translation/ExportVerilog/ExportVerilog.cpp#L958 it works fine.

lattner, 2021-03-13:
Fixed in fa201796, thanks!

closed 2021-03-13 by lattner