llvm/circt #728: splitting variadic expressions in a procedural block makes a temporary at the wrong place

darthscsi, 2021-03-05:
splitting variadic expressions in a procedural block makes a temporary at the wrong place.  This is a follow-on to #720 
```
rtl.module @foo(%clock: i1, %a: i1 {rtl.name = "asdfasdfasdfasdfafa"}, %b: i1 {rtl.name = "gasfdasafwjhijjafija"})  {
    sv.always posedge %clock  {
          sv.fwrite "force output"
        %cond = comb.and %a, %b, %a, %b, %a, %b : i1
        sv.if %cond  {
          sv.fwrite "this cond is split"
        }
    }
    rtl.output 
  }
```

lattner, 2021-03-05:
Thx, *shakes fist at verilog*, I'll take a look tonight or this weekend at the latest

darthscsi, 2021-03-05:
Verilog doesn't always make me sad, sometimes I'm not thinking about it.

lattner, 2021-03-07:
Yes, that sounds about right.  this is what compilers are supposed to take care of for us.  :-)

Fixed in e2f3f69, thanks again!

closed 2021-03-07 by lattner