llvm/circt #729: sv.ifdef.procedural isn't as procedural as other procedural blocks

darthscsi, 2021-03-05:
ifdef.procedural blocks aren't true procedural blocks, in that you can't declare variables in them safely.  Consider:
```
var a;
printf
ifdef.procedural cond
  var b
endif
printf
ifdef.procedural cond
  var c
endif
```
vars could either be explicit in the ir or generated during lowering.  If cond is true, the result is invalid verilog.  variables declarations would have to be hoisted to otuside the ifdef to be at the beginning of the enclosing block.  For temporaries from common subexpression emission or from long-line splitting, this would be a surprising result (contents inside the ifdef generated output outside the ifdef).

lattner, 2021-03-05:
Is this a semantic issue or a verilog printing issue?  The verilog printer tries to handle this today (but misses for the "too long expression" case as you pointed out). If we have other cases that are problematic, we could emit them as:

`ifdef x
begin
...
end
`endif

Which would allow var declarations IIUC.

darthscsi, 2021-03-05:
yea, it just seems non-ifdef like to generate output outside the ifdef.  One could reasonably want to do:
```
var b : i9
ifdef.proc
  var a : i6
  b= math on a
else
  var a : i9
  b = math on a
endif
```
My first problem is right now, without outputting declarations early, the legality of the if block depends on context.  If we do output declarations early, then we have to rename one of them, which would be surprising in the output, if this is expected to work like an actual ifdef.

lattner, 2021-03-07:
In practice, you can't actually do this.  The equivalent code would end up getting printed by the verilog emitter like this:

logic b : i9
logic a : i6
logic a_0 = i9

ifdef.proc
  b= math on a
else
  var a : i9
  b = math on a_0
endif

Please let me know if there is an IR example that isn't printing right.  I think we're good here.

lattner, 2021-03-14:
I'm not aware of a problem here.  Please include a testcase if there is a problem, thx

closed 2021-03-14 by lattner