llvm/circt #733: Handshake <--> ESI connector

teqdruid, 2021-03-06:
Handshake has ready/valid semantics. ESI has valid/ready semantics. Let's marry them.

mikeurbach, 2021-03-06:
I investigated this once, so let me share what I found.

This will need to occur after Handshake has been lowered. The valid/ready signals are implicit in the Handshake dialect, but to integrate with ESI, we need actual Values for valid/ready signals.

The integration should probably happen at the top level module in the lowered circuit. Technically every Handshake component has valid/ready semantics, but my understanding of ESI is that we wouldn't really want it between every Fork, Join, etc. We would want it integrated just at the top level of the whole circuit.

One point of friction I found was that Handshake is lowering to FIRRTL, which is a closed world that we can't just inject ESI into.  We've discussed before that Handshake could lower to RTL/SV directly, and that's definitely an option to open up the world.

We really want to do this integration at the FIRRTL(ish) level of abstraction, because the valid/ready values are explicit in the type system. Once FIRRTL is lowered to RTL/SV, there is no explicit association between the data signal and its valid/ready signals.

It seems like we'd want to use ESI channel types in the top level module's ports, which FIRRTL wouldn't allow. If it makes sense to put ESI channel types as the ports of the top level module, we might need to use an RTL module, at least in the top level. We could then cast back to FIRRTL within the module body, so most of the lowering can be reused.

Finally, I was thinking about when this should be done in the pipeline. We could generate ESI stuff when we lower Handshake down to the (FIR)RTL level. Or, we could have a separate pass to do this after Handshake has been lowered. A separate pass is what I was thinking.

teqdruid, 2021-03-07:
> The integration should probably happen at the top level module in the lowered circuit. Technically every Handshake component has valid/ready semantics, but my understanding of ESI is that we [...]

Yeah, ESI is not intended for every connection. I think it's appropriate to have a 'ToESIFromHandshake' and a 'ToHandshakeFromESI' op which get lowered to `esi.wrap.vr` and `esi.unwrap.vr` whenever in the lowering the valid/ready signals get realized. Presumably, this would just be at module boundaries (at least at first).

> It seems like we'd want to use ESI channel types in the top level module's ports, which FIRRTL wouldn't allow. If it makes sense to put ESI channel types as the ports of the top level module, we [...]

This is exactly what I was thinking, though I didn't know enough about the FIRRTL dialect to have an informed thought. I would assume that handshake lowers itself to a FIRRTL module which some ready/valid signals, so it's really just a question of building that gasket.

BTW: ESI does not yet lower RTLModule's ESI channel's ports, it only does this for RTLExternModules and uses SV interfaces. The thought was that for modules defined in CIRCT, data/valid/ready would be more appropriate since modules are pretty heavyweight.

> Finally, I was thinking about when this should be done in the pipeline. We could generate ESI stuff when we lower Handshake down to the (FIR)RTL level. Or, we could have a separate pass to do this after Handshake has been lowered. A separate pass is what I was thinking.

I think this boils down to the use case. If you just want to connect the handshake ports which are exposed out to the world, a post-handshake-lowering pass could work. If, however, we eventually want to use ESI within Handshake to construct snoop-able long haul connections, to build NoCs, to build telemetry/debug which implicitly tunnels out to the top level, then I think the ops I mentioned above which get lowered when Handshake lowers to FIRRTL is better. Also, for the former use case, the Handshake module could be hooked up just like extern modules are now, using the `esi.cosim` op to define the endpoints.

Out curiosity, does the FIRRTL dialect allow RTLModule instances inside of its modules?

mikeurbach, 2021-03-08:
Thanks for the input @teqdruid.

>  does the FIRRTL dialect allow RTLModule instances inside of its modules?

I don't think so; FIRRTL operations only accept FIRRTL types, so it's an explicitly closed world in the type system. When FIRRTL lowers itself to RTL/SV, it first replaces all of the FIRRTLModules with RTLModules, and keeps the operations within each module in the FIRRTL dialect. There is a second pass that then lowers the internal operations to RTL/SV.

> then I think the ops I mentioned above which get lowered when Handshake lowers to FIRRTL is better

This approach makes sense, but I am still a little hazy how much work it will be to fit this into the existing lowering. I will continue to ponder what this looks like, since this seems like the right way to do a deep integration, which could be super cool in the long run.

> Also, for the former use case, the Handshake module could be hooked up just like extern modules are now

This actually sounds pretty appealing to me as a first quick-and-dirty experiment. It seems like the least obtrusive means to connect the dots in a useful way. I might look into this route first and see if it bears fruit.

lattner, 2021-03-09:
Agreed, FIRRTL has a closed world assumption.  I'd recommend moving Handshake off of FIRRTL now that the lower level things are up to speed.  We still need a good seq dialect and other mid level abstractions of course.

teqdruid, 2021-03-09:
I'm planning on writing that seq motivation document within the next 24
hours.

Is there any particular reason that FIRRTL doesn't support RTL module
instantiations as its FFI?

~John

lattner, 2021-03-09:
FIRRTL dialect does support "FExtModuleOp" which allows talking to external verilog modules, even ones that are parameterized.  There is no support for talking to RTL dialect modules but such a thing could be added in theory.  I don't want to do it until there is a specific reason though, because we're trying to keep FIRRTL minimal and focused and push the cool stuff elsewhere.

mikeurbach, 2021-03-09:
> I'd recommend moving Handshake off of FIRRTL

Agreed, and I think this has been discussed before. Besides `BundleType`, all of the Handshake lowerings are using low-FIRRTL constructs [to avoid being too specific to FIRRTL](https://llvm.discourse.group/t/handshake-lowerings-for-merge-like-and-branch-ops/2054), so this should be relatively straightforward to implement. However, I consider this a separate issue, so I opened one here: https://github.com/llvm/circt/issues/742.

I think we can keep holding FIRRTL at arms length in this case as well. I don't really have time to refactor HandshakeToFIRRTL, but I do want to experiment with a Handshake <--> ESI connector. I think that can be done in a way that is mostly FIRRTL-agnostic, and can be refactored easily when Handshake does start lowering to something else.

mikeurbach, 2021-04-13:
I think the discussion in #742 has evolved to encompass the original scope of this ticket. Shall we close this @teqdruid?

teqdruid, 2021-04-13:
#742 is a superset of this. I don't think we need to lower directly to rtl/comb/seq in order to accomplish this. Handshake just needs to produce an RTLModuleOp with ESI ports to wrap the FIRRTL module. Unfortunately, the new build wrapper functions in the ESI dialect won't work for this purpose since they assume RTLModule[Extern]Op. We should really introduce a `ModuleLike` op interface.

Anyway, if #742 gets implemented, it should close this one as well so I'd prefer this one stay open since we don't know how quickly #742 will happen.

mikeurbach, 2021-04-13:
Ok fair enough.

> Handshake just needs to produce an RTLModuleOp with ESI ports to wrap the FIRRTL module. 

I do think this could be done in a standalone pass, and I see the value in that without getting into all of #742 .

mikeurbach, 2021-04-13:
I think another approach is to lower Handshake all the way down to RTL/Comb/SV, and then run the new build wrapper functions in ESI, which should work with that representation.

teqdruid, 2021-04-13:
Yeah, that'd work. It was the intent of the wrapper functions -- as they exist today -- to be used by a driver application which knows about the ports, not the handshake code itself. Building the wrapper before lowering feels more "MLIR-ish". So I think which makes sense depends on the use case.

mikeurbach, 2021-11-10:
I worked on this once a while ago, and got partway there. Since there has been some Handshake <--> ESI talks recently, I dusted off what I had and put it here: https://github.com/llvm/circt/pull/2127. It's a Handshake-specific pass that uses the wrapper functionality @teqdruid put in a while back on the top level `hw.module`.

I didn't even add a test yet, but here is how it looks...

Given an input to the Handshake flow like this:

```mlir
builtin.func @add(%a: i32, %b: i32) -> (i32) {
  %0 = arith.addi %a, %b : i32
  return %0 : i32
}
```

You can run: 

```
circt-opt test.mlir -lower-std-to-handshake -canonicalize -lower-handshake-to-firrtl | firtool -format=mlir -lower-to-hw | circt-opt -handshake-wrap-esi=top-module=add
```

The output looks like this:

```mlir
module  {
  hw.module @arith_addi_in_ui32_ui32_out_ui32(%arg0_valid: i1, %arg0_data: i32, %arg1_valid: i1, %arg1_data: i32, %arg2_ready: i1) -> (arg0_ready: i1, arg1_ready: i1, arg2_valid: i1, arg2_data: i32) {
    %0 = comb.add %arg0_data, %arg1_data : i32
    %1 = comb.and %arg0_valid, %arg1_valid : i1
    %2 = comb.and %arg2_ready, %1 : i1
    hw.output %2, %2, %1, %0 : i1, i1, i1, i32
  }
  hw.module @add(%arg0_valid: i1, %arg0_data: i32, %arg1_valid: i1, %arg1_data: i32, %arg2_valid: i1, %out0_ready: i1, %outCtrl_ready: i1, %clock: i1, %reset: i1) -> (arg0_ready: i1, arg1_ready: i1, arg2_ready: i1, out0_valid: i1, out0_data: i32, outCtrl_valid: i1) {
    %arith_addi.arg0_ready, %arith_addi.arg1_ready, %arith_addi.arg2_valid, %arith_addi.arg2_data = hw.instance "arith_addi" @arith_addi_in_ui32_ui32_out_ui32(arg0_valid: %arg0_valid: i1, arg0_data: %arg0_data: i32, arg1_valid: %arg1_valid: i1, arg1_data: %arg1_data: i32, arg2_ready: %out0_ready: i1) -> (arg0_ready: i1, arg1_ready: i1, arg2_valid: i1, arg2_data: i32)
    hw.output %arith_addi.arg0_ready, %arith_addi.arg1_ready, %outCtrl_ready, %arith_addi.arg2_valid, %arith_addi.arg2_data, %arg2_valid : i1, i1, i1, i1, i32, i1
  }
  hw.module @add_esi(%arg0_data: !esi.channel<i32>, %arg1_data: !esi.channel<i32>, %arg2_valid: i1, %outCtrl_ready: i1, %clock: i1, %reset: i1) -> (arg2_ready: i1, out0_data: !esi.channel<i32>, outCtrl_valid: i1) {
    %rawOutput, %valid = esi.unwrap.vr %arg0_data, %pearl.arg0_ready : i32
    %rawOutput_0, %valid_1 = esi.unwrap.vr %arg1_data, %pearl.arg1_ready : i32
    %chanOutput, %ready = esi.wrap.vr %pearl.out0_data, %pearl.out0_valid : i32
    %pearl.arg0_ready, %pearl.arg1_ready, %pearl.arg2_ready, %pearl.out0_valid, %pearl.out0_data, %pearl.outCtrl_valid = hw.instance "pearl" @add(arg0_valid: %valid: i1, arg0_data: %rawOutput: i32, arg1_valid: %valid_1: i1, arg1_data: %rawOutput_0: i32, arg2_valid: %arg2_valid: i1, out0_ready: %ready: i1, outCtrl_ready: %outCtrl_ready: i1, clock: %clock: i1, reset: %reset: i1) -> (arg0_ready: i1, arg1_ready: i1, arg2_ready: i1, out0_valid: i1, out0_data: i32, outCtrl_valid: i1)
    hw.output %pearl.arg2_ready, %chanOutput, %pearl.outCtrl_valid : i1, !esi.channel<i32>, i1
  }
}
```

I say this is partway there because are a couple top-level ports that don't get wrapped. For the clock and reset, there is no need. But there are two pairs of valid-ready pairs added to every Handshake top-level. One to control the start of the "function call" (`arg2_valid`/`arg2_ready`), and one to for the "function call" to indicate it is done (`outCtrl_valid`/`outCtrl_ready`). I'm not sure what we should do about these...

One thought would be to add ESI channels with a NoneType, to be faithful to Handshake's semantics. We could add this on top of the ports found by `esi::findValidReadySignals`. Or we could teach `esi::findValidReadySignals` to look for valid-ready pairs that may not have a data signal.

More generally, if we are turning the top-level ports into ESI channels, I'm not sure if these extra signals are even necessary. Perhaps they could be surgically removed by my ESI-wrapping pass, or perhaps we could add a flag to the Handshake lowering to not generate them. Or maybe they still serve a purpose?

This is still half baked, but thought I'd share.