llvm/circt #742: [Handshake] Move HandshakeToFIRRTL lowerings to target RTL/Comb/Seq dialects

mikeurbach, 2021-03-09:
The HandshakeToFIRRTL lowering existed before most of the RTL and Comb dialects existed (and the Seq dialect does not exist as of today). Targeting FIRRTL was a practical choice, but pains have been taken [to avoid being too specific to FIRRTL](https://llvm.discourse.group/t/handshake-lowerings-for-merge-like-and-branch-ops/2054) so that one day we can target the CIRCT-native dialects to model hardware at that level (RTL/Comb/Seq). As these dialects mature, it is worth moving Handshake off FIRRTL. The RTL/Comb/Seq dialects are more flexible and facilitate broader integrations within the CIRCT ecosystem.

Practically speaking, this issue can and should be tackled in small pieces. To support this, it would be great to land a proper Handshake integration test before we start working on this. I have a second iteration of #464 that I can share to restart the integration testing discussion.

mikeurbach, 2021-03-26:
A good first step might be to lower to RTLModule instead of FIRRTLModule, but keep the rest the same. I'm not sure how practical this is, but it would open up the type system.

teqdruid, 2021-04-10:
With the addition of `seq.combreg` do we have everything we need to do this?

mikeurbach, 2021-04-10:
I have been thinking about this... it's close. It'd be nice to have an abstraction for a [pipeline](https://github.com/llvm/circt/issues/587), so we can say things like "give me a two-slot non-transparent buffer". Maybe in `seq.pipeline`? It comes up for `handshake.buffer`, as well as memories, if we need to match up read/write latencies. That's certainly not necessary to get started, everything in HandshakeToFIRRTL should be representable with `comb` and `seq.compreg` today.

One thing I was thinking about--I'd be interested in a lowering that uses ESI channels for all the modules' ports. This wasn't in the initial scope of this ticket, but it seems like a great fit.

teqdruid, 2021-04-10:
It'd be interesting to model all of the handshake internal connections as
ESI connections. ESI already has the buffer operation of which you speak...

mikeurbach, 2021-04-10:
Yeah, the more I think on it, the more interesting leaning into ESI for stuff below Handshake in abstraction but above "RTL" makes sense. It's relatively straightforward to swap in right now. All the Handshake ops lower to FIRRTL modules with BundleType ports and SubfieldOp to get out the Values. This could be swapped for ChannelType ports and UnwrapVROp to get out the values. 

Like you say, then all the other good ESI stuff is available throughout the lowering process.

mikeurbach, 2021-04-10:
Updating the title to reflect that this would be a new, separate, incremental addition. AFAICT the HandshakeToFIRRTL path is just starting to work end-to-end, and I definitely want to use that path 🙂