llvm/circt #754: [FIRRTL] mem

drom, 2021-03-11:
The following FIRRTL code:

```python
circuit top_mod :
  module top_mod :
    input clock : Clock
    input r0addr : UInt<11>
    input w0addr : UInt<10>
    input w0data : UInt<7>
    output r0data : UInt<5>
    mem u0 :
      data-type => UInt<9>
      depth => 42
      read-latency => 0
      write-latency => 1
      reader => r0
      writer => w0
      read-under-write => undefined
    u0.r0.en <= UInt<1>(1)
    u0.r0.addr <= r0addr
    u0.r0.clk <= clock
    u0.w0.en <= UInt<1>(1)
    u0.w0.addr <= w0addr
    u0.w0.mask <= UInt<1>(1)
    u0.w0.data <= w0data
    u0.w0.clk <= clock
    r0data <= u0.r0.data
```

produces the following Verilog output:

```verilog
// Standard header to adapt well known macros to our needs.
`ifdef RANDOMIZE_GARBAGE_ASSIGN
  `define RANDOMIZE
`endif
`ifdef RANDOMIZE_INVALID_ASSIGN
  `define RANDOMIZE
`endif
`ifdef RANDOMIZE_REG_INIT
  `define RANDOMIZE
`endif
`ifdef RANDOMIZE_MEM_INIT
  `define RANDOMIZE
`endif
`ifndef RANDOM
  `define RANDOM $random
`endif

// Users can define 'PRINTF_COND' to add an extra gate to prints.
`ifdef PRINTF_COND
  `define PRINTF_COND_ (`PRINTF_COND)
`else
  `define PRINTF_COND_ 1
`endif

// Users can define 'STOP_COND' to add an extra gate to stop conditions.
`ifdef STOP_COND
  `define STOP_COND_ (`STOP_COND)
`else
  `define STOP_COND_ 1
`endif

// Users can define INIT_RANDOM as general code that gets injected into the
// initializer block for modules with registers.
`ifndef INIT_RANDOM
  `define INIT_RANDOM
`endif

// If using random initialization, you can also define RANDOMIZE_DELAY to
// customize the delay used, otherwise 0.002 is used.
`ifndef RANDOMIZE_DELAY
  `define RANDOMIZE_DELAY 0.002
`endif

// Define INIT_RANDOM_PROLOG_ for use in our modules below.
`ifdef RANDOMIZE
  `ifdef VERILATOR
    `define INIT_RANDOM_PROLOG_ `INIT_RANDOM
  `else
    `define INIT_RANDOM_PROLOG_ `INIT_RANDOM #`RANDOMIZE_DELAY begin end
  `endif
`else
  `define INIT_RANDOM_PROLOG_
`endif

module top_mod(
  input         clock,
  input  [10:0] r0addr,
  input  [9:0]  w0addr,
  input  [6:0]  w0data,
  output [4:0]  r0data);

  wire [4:0] _2Er0data2Eoutput;	// a_top_mod.fir:2:3
  reg  [8:0] u0[41:0];	// a_top_mod.fir:8:5
  wire [5:0] u0_r0_addr;	// a_top_mod.fir:8:5
  wire [8:0] u0_r0_data;	// a_top_mod.fir:8:5
  wire [5:0] u0_w0_addr;	// a_top_mod.fir:8:5
  wire       u0_w0_en;	// a_top_mod.fir:8:5
  wire       u0_w0_clk;	// a_top_mod.fir:8:5
  wire [8:0] u0_w0_data;	// a_top_mod.fir:8:5
  wire       u0_w0_mask;	// a_top_mod.fir:8:5
  wire [5:0] u0_r0_addr_6;	// a_top_mod.fir:8:5
  wire [5:0] u0_w0_addr_7;	// a_top_mod.fir:8:5
  wire       u0_w0_en_8;	// a_top_mod.fir:8:5
  wire       u0_w0_clk_9;	// a_top_mod.fir:8:5

  wire [5:0] _T = 6'h2A;	// a_top_mod.fir:8:5
  wire [5:0] _T_0 = u0_r0_addr;	// a_top_mod.fir:8:5
  wire [8:0] _T_1 = u0[_T_0];	// a_top_mod.fir:8:5
  `ifdef RANDOMIZE_GARBAGE_ASSIGN	// a_top_mod.fir:8:5
    assign u0_r0_data = _T_0 < _T ? _T_1 : `RANDOM;	// a_top_mod.fir:8:5
  `else
    assign u0_r0_data = _T_1;	// a_top_mod.fir:8:5
  `endif
  wire _T_2 = u0_w0_clk;	// a_top_mod.fir:8:5
  wire [8:0] _T_3 = u0_w0_data;	// a_top_mod.fir:8:5
  wire _T_4 = u0_w0_en & u0_w0_mask;	// a_top_mod.fir:8:5
  wire [8:0] _T_5 = u0[u0_w0_addr];	// a_top_mod.fir:8:5
  always_ff @(posedge _T_2) begin	// a_top_mod.fir:8:5
    if (_T_4)	// a_top_mod.fir:8:5
      _T_5 <= _T_3;	// a_top_mod.fir:8:5
  end // always_ff @(posedge)
  `ifndef SYNTHESIS	// a_top_mod.fir:8:5
    initial begin	// a_top_mod.fir:8:5
      `INIT_RANDOM_PROLOG_	// a_top_mod.fir:8:5
      `ifdef RANDOMIZE_MEM_INIT	// a_top_mod.fir:8:5
        integer u0_initvar;
        for (u0_initvar = 0; u0_initvar < 42; u0_initvar = u0_initvar+1)
          u0[u0_initvar] = `RANDOM;	// a_top_mod.fir:8:5
      `endif
    end // initial
  `endif
  assign u0_r0_addr = u0_r0_addr_6;	// a_top_mod.fir:8:5
  assign u0_w0_addr = u0_w0_addr_7;	// a_top_mod.fir:8:5
  assign u0_w0_en = u0_w0_en_8;	// a_top_mod.fir:8:5
  assign u0_w0_clk = u0_w0_clk_9;	// a_top_mod.fir:8:5
  assign u0_r0_addr_6 = r0addr[5:0];	// a_top_mod.fir:17:16
  assign u0_w0_en_8 = 1'h1;	// a_top_mod.fir:16:17, :19:14
  assign u0_w0_addr_7 = w0addr[5:0];	// a_top_mod.fir:20:16
  assign u0_w0_mask = 1'h1;	// a_top_mod.fir:16:17, :21:16
  assign u0_w0_data = {2'h0, w0data};	// a_top_mod.fir:22:16
  assign u0_w0_clk_9 = clock;	// a_top_mod.fir:23:15
  wire [8:0] _T_10 = u0_r0_data;	// a_top_mod.fir:24:12
  assign _2Er0data2Eoutput = _T_10[4:0];	// a_top_mod.fir:24:12
  assign r0data = _2Er0data2Eoutput;	// a_top_mod.fir:2:3
endmodule
```

With the following issues:

## Clock reassignment

Getting:

```verilog
  input         clock,
...
  wire       u0_w0_clk;	// a_top_mod.fir:8:5
...
  wire       u0_w0_clk_9;	// a_top_mod.fir:8:5
...
  wire _T_2 = u0_w0_clk;	// a_top_mod.fir:8:5
...
  always_ff @(posedge _T_2) begin	// a_top_mod.fir:8:5
    ...
  end // always_ff @(posedge)
...
  assign u0_w0_clk = u0_w0_clk_9;	// a_top_mod.fir:8:5
...
  assign u0_w0_clk_9 = clock;	// a_top_mod.fir:23:15
```

Expecting:

```verilog
  input         clock,
...
  always @(posedge clock) begin	// a_top_mod.fir:8:5
    ...
  end // always_ff @(posedge)
```

##

lattner, 2021-03-13:
We now generate this, it looks like the unpacked dimension order is still different:

```
module top_mod(
  input         clock,
  input  [10:0] r0addr,
  input  [9:0]  w0addr,
  input  [6:0]  w0data,
  output [4:0]  r0data);

  reg  [8:0] u0[41:0];	// z.fir:8:5
  wire [8:0] u0_r0_data;	// z.fir:8:5
  wire [5:0] _T_0;	// z.fir:17:16
  wire [8:0] _T_1;	// z.fir:22:16

  wire [8:0] _T = u0[_T_0];	// z.fir:8:5
  `ifdef RANDOMIZE_GARBAGE_ASSIGN	// z.fir:8:5
    assign u0_r0_data = _T_0 < 6'h2A ? _T : `RANDOM;	// z.fir:8:5
  `else
    assign u0_r0_data = _T;	// z.fir:8:5
  `endif
  always_ff @(posedge clock)	// z.fir:8:5
    u0[w0addr[5:0]] <= _T_1;	// z.fir:8:5, :20:16
  `ifndef SYNTHESIS	// z.fir:8:5
    initial begin	// z.fir:8:5
      `INIT_RANDOM_PROLOG_	// z.fir:8:5
      `ifdef RANDOMIZE_MEM_INIT	// z.fir:8:5
        integer u0_initvar;
        for (u0_initvar = 0; u0_initvar < 42; u0_initvar = u0_initvar+1)
          u0[u0_initvar] = `RANDOM;	// z.fir:8:5
      `endif
    end // initial
  `endif
  assign _T_0 = r0addr[5:0];	// z.fir:17:16
  assign _T_1 = {2'h0, w0data};	// z.fir:22:16
  wire [8:0] _T_2 = u0_r0_data;	// z.fir:24:12
  assign r0data = _T_2[4:0];	// z.fir:2:3, :24:12
endmodule
```

drom, 2021-03-13:
Looks way better!

lattner, 2021-03-13:
Is the dimension ordering still a problem?  @darthscsi is it ok for us to just change this in the printer?  It is probably a bug due to refactoring of the logic, I'm not sure if there is a semantic effect.

lattner, 2021-03-13:
Nuked a few more wires, and tightened up the #ifdef with a macro:

```
module top_mod(
  input         clock,
  input  [10:0] r0addr,
  input  [9:0]  w0addr,
  input  [6:0]  w0data,
  output [4:0]  r0data);

  reg  [8:0] u0[41:0];	// z.fir:8:5
  wire [5:0] _T_0;	// z.fir:17:16
  wire [8:0] _T_1;	// z.fir:22:16

  wire [8:0] _T = RANDOMIZE_GARBAGE_ASSIGN_BOUND_CHECK(_T_0, u0[_T_0], 6'h2A);	// z.fir:8:5
  always_ff @(posedge clock)	// z.fir:8:5
    u0[w0addr[5:0]] <= _T_1;	// z.fir:8:5, :20:16
  `ifndef SYNTHESIS	// z.fir:8:5
    initial begin	// z.fir:8:5
      `INIT_RANDOM_PROLOG_	// z.fir:8:5
      `ifdef RANDOMIZE_MEM_INIT	// z.fir:8:5
        integer u0_initvar;
        for (u0_initvar = 0; u0_initvar < 42; u0_initvar = u0_initvar+1)
          u0[u0_initvar] = `RANDOM;	// z.fir:8:5
      `endif
    end // initial
  `endif
  assign _T_0 = r0addr[5:0];	// z.fir:17:16
  assign _T_1 = {2'h0, w0data};	// z.fir:22:16
  assign r0data = _T[4:0];	// z.fir:2:3, :24:12
endmodule
```

darthscsi, 2021-03-14:
We can change it in the printer, endian discussion and bitcast would care.
I don't remember if we defined an order for unpacked arrays.  But bound
order is visible in other semantics (Cast, slice, stream, etc).

drom, 2021-03-16:
@lattner 

> Nuked a few more wires, and tightened up the #ifdef with a macro:
> 
> ```
> module top_mod(
> ...
>   wire [8:0] _T = RANDOMIZE_GARBAGE_ASSIGN_BOUND_CHECK(_T_0, u0[_T_0], 6'h2A);	// z.fir:8:5
> ...
> endmodule
> ```

Good Nuking! But there where some innocent casualties. You `RANDOMIZE_GARBAGE_ASSIGN_BOUND_CHECK` is missing a `back-tick`

prithayan, 2021-03-16:
@drom Added the back-tick

lattner, 2021-03-16:
Thank you for catching this @prithayan !

lattner, 2021-03-20:
Retitling since there is only one issue left here.  @darthscsi can you take this since you've worked on the type printing stuff?

prithayan, 2021-03-24:
PR to fix this ExportVerilog issue, https://github.com/llvm/circt/pull/815

closed 2021-03-24 by prithayan