llvm/circt #755: Verilog Or Reduction Improperly Inlines Nots

seldridge, 2021-03-11:
Verilog emission is generating things like "or reduction" "not", e.g., `|~foo` which is invalid Verilog.

Consider:

```mlir
module  {
  firrtl.circuit "Foo" {
    firrtl.module @Foo(%a: !firrtl.uint<8>, %b: !firrtl.flip<uint<1>>) {
      %0 = firrtl.not %a : (!firrtl.uint<8>) -> !firrtl.uint<8>
      %1 = firrtl.orr %0 : (!firrtl.uint<8>) -> !firrtl.uint<1>
      firrtl.connect %b, %1 : !firrtl.flip<uint<1>>, !firrtl.uint<1>
    }
  }
}
```

This produces:

```verilog
module Foo(
  input  [7:0] a,
  output       b);

  assign b = |~a;	// sunder/Foo.fir:2:3, :6:{10,14}
endmodule
```

lattner, 2021-03-12:
Is it ok with parens?  `|(~a)`?

seldridge, 2021-03-12:
Yeah, using parens there should be fine. :+1: (Ideally, these rules are somewhere in the emitter and it'll aggressively inline when possible, adding parentheses, but only if needed or if we add other rules which make things clearer for a human reader.)

lattner, 2021-03-12:
Fixed in 48617dba thanks!

closed 2021-03-12 by lattner