llvm/circt #762: [FIRRTL] Fix the syntax of `firrtl.constant` to match `rtl.constant`

lattner, 2021-03-12:
We currently use:

```
 %c5_ui8 = firrtl.constant(5 : ui8) : !firrtl.uint<8>
```
it would be waaay nicer to have:

```
 %c5_ui8 = firrtl.constant 5 : !firrtl.uint<8>
```

Also, the implementation of this node sometimes stores the attributes with i8 type, sometimes with ui8, etc.  It would be good to make this consistent and verify it.

Ayushsunny, 2021-04-23:
Hey @lattner I wanna fix this

lattner, 2021-04-23:
Sure, please create a PR!

Ayushsunny, 2021-04-23:
> Sure, please create a PR!
Sir, I have made a PR for this issue 
could you please review it https://github.com/llvm/circt/pull/954

lattner, 2021-05-05:
For others following along, the direction proposed isn't the right patch. I don't consider this bug to be taken by anyone.

lattner, 2021-05-09:
This is actually surprisingly difficult because of the way `parseAttribute` works, but [this MLIR patch](https://reviews.llvm.org/D102120) will allow addressing this.

lattner, 2021-05-12:
Finally implemented in 2d1c29e8

closed 2021-05-12 by lattner