llvm/circt #773: [FIRRTL] crash

drom, 2021-03-16:
The following FIRRTL program

```python
; seed: 497
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_a: UInt<5>
    input inp_b: {inp_c: {inp_j: UInt<5>}, inp_d: {inp_e: SInt<7>, inp_i: SInt<2>}, inp_g: {inp_h: UInt<14>}}
    input inp_f: UInt<13>
    output tmp8: SInt<25>
    reg tmp6: UInt<5>, clock with: (reset => (reset, UInt<5>(0)))
    reg tmp9: UInt<13>, clock
    reg tmp10: SInt<8>, clock
    wire tmp11: SInt<23>
    reg tmp12: SInt<1>, clock with: (reset => (reset, SInt<1>(0)))
    reg tmp13: SInt<11>, clock
    reg tmp15: UInt<8>, clock
    wire _tmp17: UInt<12>
    wire tmp18: SInt<11>
    reg tmp19: UInt<13>, clock with: (reset => (reset, UInt<13>(0)))
    reg tmp23: SInt<11>, clock with: (reset => (reset, SInt<11>(0)))
    tmp6 <= bits(inp_b.inp_g.inp_h, 4, 0)
    tmp8 <= mux(head(tmp6, 1), SInt<23>("h1d0137"), asSInt(UInt<25>("b100101111111101010010010")))
    tmp9 <= xor(inp_f, inp_b.inp_c.inp_j)
    tmp10 <= neg(UInt<7>("h26"))
    tmp11 <= asSInt(UInt<23>("o5674546"))
    tmp12 <= validif(head(inp_f, 1), SInt<1>("b-1"))
    tmp13 <= shr(tmp8, 14)
    tmp15 <= not(tmp10)
    _tmp17 <= add(UInt<11>("b110001010"), tmp15)
    tmp18 <= rem(tmp13, asSInt(UInt<24>(9972122)))
    tmp19 <= bits(inp_f, 12, 0)
    tmp23 <= validif(head(asUInt(SInt<20>("o-71634")), 1), tmp13)
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
firtool
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
 #0 0x00000000004bc921 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004bc9d8 PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004ba9d8 llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004bc374 SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007fb20ef94a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #5 0x0000000000694d5e llvm::PointerIntPair<void*, 1u, int, llvm::pointer_union_detail::PointerUnionUIntTraits<mlir::Dialect*, mlir::MLIRContext*>, llvm::PointerIntPairInfo<void*, 1u, llvm::pointer_union_detail::PointerUnionUIntTraits<mlir::Dialect*, mlir::MLIRContext*> > >::getInt() const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/PointerIntPair.h:61:57
 #6 0x0000000000695db3 bool llvm::PointerUnion<mlir::Dialect*, mlir::MLIRContext*>::is<mlir::Dialect*>() const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/PointerUnion.h:181:34
 #7 0x000000000069113c mlir::Dialect* llvm::PointerUnion<mlir::Dialect*, mlir::MLIRContext*>::dyn_cast<mlir::Dialect*>() const /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/PointerUnion.h:195:5
 #8 0x0000000000687b59 mlir::Identifier::getDialect() /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/MLIRContext.cpp:801:1
 #9 0x00000000006a96f7 mlir::OperationName::getDialect() const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:328:56
#10 0x00000000006a176e mlir::Operation::getDialect() /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:214:67
#11 0x00000000006a27fb mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:557:32
#12 0x00000000007d23e9 mlir::OperationFolder::tryToFold(mlir::OpBuilder&, mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::function_ref<void (mlir::Operation*)>) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:198:13
#13 0x00000000007d1d6e mlir::OperationFolder::tryToFold(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, llvm::function_ref<void (mlir::Operation*)>, bool*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:101:13
#14 0x00000000007b7f09 (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>, int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:191:21
#15 0x00000000007b8234 mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:258:35
#16 0x00000000007b8120 mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:238:1
#17 0x00000000006ed6ae (anonymous namespace)::Canonicalizer::runOnOperation() /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:35:3
#18 0x00000000008a4563 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:403:36
#19 0x00000000008a478f mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:444:15
#20 0x00000000008a68cf mlir::PassManager::run(mlir::Operation*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:896:74
#21 0x000000000040dbfc processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) (../../llvm/circt/build/bin/firtool+0x40dbfc)
#22 0x000000000040b203 main (../../llvm/circt/build/bin/firtool+0x40b203)
#23 0x00007fb20ea02b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#24 0x000000000040b56e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: строка 25: 25590 Segmentation fault      (стек памяти сброшен на диск) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

The issue appears at random.

lattner, 2021-03-17:
Thanks @drom, valgrind shows lots of interesting things, working on it

lattner, 2021-03-17:
I believe that I fixed this in 88866c32, please verify.  Thank you for the report!

drom, 2021-03-17:
Rerunning regression overnight.

drom, 2021-03-17:
Look good!

closed 2021-03-17 by drom