llvm/circt #774: [FIRRTL] instance binding reassignment

drom, 2021-03-17:
The following FIRRTL program

```python
circuit top_mod :
  module mod_0 :
    input clock: Clock
    input reset: UInt<1>
    input d: UInt<1>
    output q: UInt<1>
    q <= d
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input d: UInt<1>
    output q: UInt<1>
    inst U0 of mod_0
    U0 is invalid
    U0.clock <= clock
    U0.reset <= reset
    U0.d <= d
    q <= U0.q
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.
...
module mod_0(
  input  clock, reset, d,
  output q);

  assign q = d;	// a_top_mod.fir:2:3
endmodule

module top_mod(
  input  clock, reset, d,
  output q);

  wire _T;	// a_top_mod.fir:13:5
  wire _T_0;	// a_top_mod.fir:13:5
  wire _T_1;	// a_top_mod.fir:13:5
  wire U0_q;	// a_top_mod.fir:13:5

  assign _T = clock;	// a_top_mod.fir:13:5
  assign _T = 1'h0;	// a_top_mod.fir:13:5, :14:5
  assign _T_0 = reset;	// a_top_mod.fir:13:5
  assign _T_0 = 1'h0;	// a_top_mod.fir:13:5, :14:5
  assign _T_1 = d;	// a_top_mod.fir:13:5
  assign _T_1 = 1'h0;	// a_top_mod.fir:13:5, :14:5
  mod_0 U0 (	// a_top_mod.fir:13:5
    .clock (_T),
    .reset (_T_0),
    .d     (_T_1),
    .q     (U0_q)
  );
  assign q = U0_q;	// a_top_mod.fir:8:3
endmodule
```

Which has incorrect reassignment or `_T`, `_T_0` and `_T_1` temporal wires.

Compiled with `firrtl-1.5.0` produces this Verilog:

```verilog
module mod_0(
  input   d,
  output  q
);
  assign q = d;
endmodule
module top_mod(
  input   clock,
  input   reset,
  input   d,
  output  q
);
  wire  U0_d;
  wire  U0_q;
  mod_0 U0 (
    .d(U0_d),
    .q(U0_q)
  );
  assign q = U0_q;
  assign U0_d = d;
endmodule
```

prithayan, 2021-03-17:
This is the IR in RTL dialect. The invalidate construct is lowered to ```false``` constant in RTL dialect. 
```  
  module attributes {firrtl.mainModule = "top_mod"}  {
  sv.verbatim "// Standard header to adapt well known macros to our needs."
  sv.verbatim ""
  rtl.module @mod_0(%clock: i1, %reset: i1, %d: i1) -> (%q: i1) {
    rtl.output %d : i1
  }
  rtl.module @top_mod(%clock: i1, %reset: i1, %d: i1) -> (%q: i1) {
    %false = rtl.constant false
    %0 = comb.merge %clock, %false : i1
    %1 = comb.merge %reset, %false : i1
    %2 = comb.merge %d, %false : i1
    %U0.q = rtl.instance "U0" @mod_0(%0, %1, %2) : (i1, i1, i1) -> i1
    rtl.output %U0.q : i1
  }
}
```

darthscsi, 2021-03-17:
Input is high-firrtl, which isn't supported yet.

lattner, 2021-03-18:
I think the problem is that we're not lowering rtl.merge correctly.  I'm going to stop its formation and remove it.

lattner, 2021-03-18:
Actually, that isn't necessary, will just fix it.

lattner, 2021-03-18:
We now generate incorrect verilog instead of invalid verilog (yay progress) after rtl.merge emission has been fixed:

```
module top_mod(
  input  clock, reset, d,
  output q);

  wire _T;	// z.fir:13:5
  wire _T_0;	// z.fir:13:5
  wire _T_1;	// z.fir:13:5
  wire U0_q;	// z.fir:13:5

  assign _T_1 = clock;	// z.fir:13:5
  assign _T_1 = 1'h0;	// z.fir:13:5, :14:5
  assign _T_0 = reset;	// z.fir:13:5
  assign _T_0 = 1'h0;	// z.fir:13:5, :14:5
  assign _T = d;	// z.fir:13:5
  assign _T = 1'h0;	// z.fir:13:5, :14:5
  mod_0 U0 (	// z.fir:13:5
    .clock (_T_1),	// z.fir:13:5
    .reset (_T_0),	// z.fir:13:5
    .d     (_T),	// z.fir:13:5
    .q     (U0_q)
  );
  assign q = U0_q;	// z.fir:8:3
endmodule
```

The remaining problem is the lowering of the `invalid` op.  We currently lower it as zero, but I think SFC has some fancier lowering -- sometimes to random, sometimes to zero, sometimes to nothing.  @seldridge do you know what the right thing here is ?

darthscsi, 2021-03-18:
It lowers to zero.  But multiple assigns are independent parallel tasks and
multiple assigns to a single netlist is a race condition.  What we are
missing is preserving the last-assign semantics during lowering.  The
invalid assignment should never generate rtl as the normal assigns
post-dominate.

darthscsi, 2021-03-18:
Or but another way, we are missing incorrectly turning a procedural region
(firrtl) into a graph region.  This is the same problem we had with printf.

seldridge, 2021-03-18:
The issue here is that the last assignment needs to win. The `ExpandWhensAndCheck` pass is resolving this to "static single connect" where everything is only connected to exactly once. Here, last assignment should win.

This specific example is really no different from:

```scala
a <= b
a <= c
```

The last assignment should win here and this reduces to:
```scala
a <= c
```

Note, that in the event that something is conditionally valid, you wind up with a` validif` statement. Consider:

```scala
a is invalid
when cond:
  a <= b
```

This is lowered to:

```scala
a <= validif(cond, b)
```

Eventually that gets stripped out and the compiler will choose to optimize this some direction. The following is what happens. 

```scala
a <= b
```

Note! This is **not** _invalid becomes zero_.

seldridge, 2021-03-18:
One clarification: if we had something that was doing the equivalent work of `ExpandWhensAndCheck` (which is a requirement of lowering from High FIRRTL to Mid FIRRTL) then we'd be in "static single connect" and this wouldn't be a problem. 

Reiterating @darthscsi's point, I view the real issue as this being High FIRRTL and we haven't started to tackle that yet. It would be good if we had a real error here as opposed to weird Verilog, though.

lattner, 2021-03-19:
Yeah ok that makes more sense to me, I didn't realize this was high firrtl.  This is equivalent to the `when` elimination problem, and will fit very naturally into its design.  That is somewhere on my todo list to take a look at, I'll keep this in mind as part of that, thanks!

youngar, 2021-04-02:
I'm going to put together an ExpandWhens pass that handles last connect semantics

lattner, 2021-04-03:
Sound sgood

youngar, 2021-04-22:
This is working now, you have to enable the `--expand-whens` pass when using firtool.  See #885.