llvm/circt #775: [FIRRTL] mem (core dump)

drom, 2021-03-17:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input r0en: UInt<1>
    mem tmp41:
      data-type => UInt<8>
      depth => 10
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
      reader => r0
      writer => w0
    tmp41.r0.clk <= clock
    tmp41.r0.en <= r0en
    tmp41.r0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.clk <= clock
    tmp41.w0.en <= r0en
    tmp41.w0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.mask <= asUInt(SInt<25>(0))
    tmp41.w0.data <= asUInt(SInt<25>(0))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
firtool
malloc(): unsorted double linked list corrupted
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
malloc(): unsorted double linked list corrupted
./process1.sh: line 25: 29968 Aborted                 (core dumped) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

prithayan, 2021-03-17:
Valgrind output
```
==18367== Memcheck, a memory error detector
==18367== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==18367== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==18367== Command: firtool -enable-lower-types -lower-to-rtl -verilog t.fir -o output.v
==18367==
==18367== Invalid write of size 8
==18367==    at 0x6241BF: mlir::IROperand<mlir::OpOperand, mlir::detail::OpaqueValue>::removeFromCurrent() (UseDefLists.h:193)
==18367==    by 0x763E85: mlir::IROperand<mlir::OpOperand, mlir::detail::OpaqueValue>::~IROperand() (UseDefLists.h:147)
==18367==    by 0x763811: mlir::OpOperand::~OpOperand() (UseDefLists.h:250)
==18367==    by 0x76388A: mlir::detail::TrailingOperandStorage::~TrailingOperandStorage() (OperationSupport.h:465)
==18367==    by 0x7618DD: mlir::detail::OperandStorage::~OperandStorage() (OperationSupport.cpp:242)
==18367==    by 0x750C65: mlir::Operation::~Operation() (Operation.cpp:188)
==18367==    by 0x750D86: mlir::Operation::destroy() (Operation.cpp:205)
==18367==    by 0x751863: llvm::ilist_traits<mlir::Operation>::deleteNode(mlir::Operation*) (Operation.cpp:429)
==18367==    by 0x6A9850: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false,
 void>, false, false>) (ilist.h:268)
==18367==    by 0x75B9D8: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(mlir::Operation*) (ilist.h:272)
==18367==    by 0x751A1A: mlir::Operation::erase() (Operation.cpp:478)
==18367==    by 0x971424: deleteDeadness(llvm::MutableArrayRef<mlir::Region>, (anonymous namespace)::LiveMap&) (RegionUtils.cpp:321)
==18367==  Address 0x66b0790 is 0 bytes inside a block of size 128 free'd
==18367==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18367==    by 0x750D92: mlir::Operation::destroy() (Operation.cpp:206)
==18367==    by 0x751863: llvm::ilist_traits<mlir::Operation>::deleteNode(mlir::Operation*) (Operation.cpp:429)
==18367==    by 0x6A9850: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false,
 void>, false, false>) (ilist.h:268)
==18367==    by 0x75B9D8: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(mlir::Operation*) (ilist.h:272)
==18367==    by 0x751A1A: mlir::Operation::erase() (Operation.cpp:478)
==18367==    by 0x971424: deleteDeadness(llvm::MutableArrayRef<mlir::Region>, (anonymous namespace)::LiveMap&) (RegionUtils.cpp:321)
==18367==    by 0x9713E1: deleteDeadness(llvm::MutableArrayRef<mlir::Region>, (anonymous namespace)::LiveMap&) (RegionUtils.cpp:318)
==18367==    by 0x97173E: runRegionDCE(llvm::MutableArrayRef<mlir::Region>) (RegionUtils.cpp:367)
==18367==    by 0x97321D: mlir::simplifyRegions(llvm::MutableArrayRef<mlir::Region>) (RegionUtils.cpp:698)
==18367==    by 0x96B3DA: (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>, int) (GreedyPatternRewriteDriver.cpp:205)
==18367==    by 0x96B64D: mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&, unsigned int) (GreedyPatternRewriteDriver.cpp:258)
==18367==  Block was alloc'd at
==18367==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==18367==    by 0x7507F3: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, unsigned int) (Operation.cpp:136)
==18367==    by 0x7505B9: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, mlir::RegionRange) (Operation.cpp:95)
==18367==    by 0x75052F: mlir::Operation::create(mlir::OperationState const&) (Operation.cpp:85)
==18367==    by 0x5D0DDF: void mlir::OpBuilder::createOrFold<circt::sv::ReadInOutOp, mlir::Value&>(llvm::SmallVectorImpl<mlir::Value>&, mlir::Location, mlir::Value&) (Builders.h:419)
==18367==    by 0x5D0C64: std::enable_if<circt::sv::ReadInOutOp::hasTrait<mlir::OpTrait::OneResult>(), mlir::Value>::type mlir::OpBuilder::createOrFold<circt::sv::ReadInOutOp, mlir::Value&>(mlir::Location, mli
r::Value&) (Builders.h:434)
==18367==    by 0x5D03DE: std::enable_if<circt::sv::ReadInOutOp::hasTrait<mlir::OpTrait::OneResult>(), mlir::Value>::type circt::ImplicitLocOpBuilder::createOrFold<circt::sv::ReadInOutOp, mlir::Value&>(mlir::V
alue&) (ImplicitLocOpBuilder.h:99)
==18367==    by 0x5A42FF: (anonymous namespace)::FIRRTLLowering::getLoweredValue(mlir::Value) (LowerToRTL.cpp:1244)
==18367==    by 0x5A4D14: (anonymous namespace)::FIRRTLLowering::getLoweredAndExtendedValue(mlir::Value, mlir::Type) (LowerToRTL.cpp:1263)
==18367==    by 0x5AFBE0: (anonymous namespace)::FIRRTLLowering::visitStmt(circt::firrtl::ConnectOp) (LowerToRTL.cpp:2455)
==18367==    by 0x5AFB4A: _ZZN5circt6firrtl11StmtVisitorIN12_GLOBAL__N_114FIRRTLLoweringEN4mlir13LogicalResultEJEE19dispatchStmtVisitorEPNS4_9OperationEENKUlT_E_clINS0_9ConnectOpEEES5_S9_ (FIRRTLVisitors.h:162)
==18367==    by 0x5AFAAF: llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>& llvm::TypeSwitch<mlir::Operation*, mlir::LogicalResult>::Case<circt::firrtl::ConnectOp, circt::firrtl::StmtVisitor<(anonymous
[... cut ...] torageUniquer::StorageAllocator&) (STLExtras.h:185)
==18367==
==18367== LEAK SUMMARY:
==18367==    definitely lost: 144 bytes in 1 blocks
==18367==    indirectly lost: 0 bytes in 0 blocks
==18367==      possibly lost: 0 bytes in 0 blocks
==18367==    still reachable: 67,624 bytes in 2 blocks
==18367==         suppressed: 0 bytes in 0 blocks
==18367== Reachable blocks (those to which a pointer was found) are not shown.
==18367== To see them, rerun with: --leak-check=full --show-leak-kinds=all
```

drom, 2021-03-17:
```
firtool
==5443== Memcheck, a memory error detector
==5443== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==5443== Using Valgrind-3.16.1 and LibVEX; rerun with -h for copyright info
==5443== Command: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
==5443== 
==5443== Conditional jump or move depends on uninitialised value(s)
==5443==    at 0x4A3070: (anonymous namespace)::FIRRTLModuleLowering::runOnOperation() (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FB54: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FC7F: mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x6A2D63: mlir::PassManager::run(mlir::Operation*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x40B8A5: processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x4080C2: main (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443== 
==5443== Conditional jump or move depends on uninitialised value(s)
==5443==    at 0x4A3093: (anonymous namespace)::FIRRTLModuleLowering::runOnOperation() (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FB54: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FC7F: mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x6A2D63: mlir::PassManager::run(mlir::Operation*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x40B8A5: processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x4080C2: main (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443== 
==5443== Conditional jump or move depends on uninitialised value(s)
==5443==    at 0x4A30A0: (anonymous namespace)::FIRRTLModuleLowering::runOnOperation() (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FB54: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FC7F: mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x6A2D63: mlir::PassManager::run(mlir::Operation*) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x40B8A5: processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x4080C2: main (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443== 
==5443== Invalid read of size 8
==5443==    at 0x5C90EE: circt::sv::WireOp::getCanonicalizationPatterns(mlir::OwningRewritePatternList&, mlir::MLIRContext*)::DropDeadConnect::matchAndRewrite(circt::sv::WireOp, mlir::PatternRewriter&) const (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x63F61D: mlir::PatternApplicator::matchAndRewrite(mlir::Operation*, mlir::PatternRewriter&, llvm::function_ref<bool (mlir::Pattern const&)>, llvm::function_ref<void (mlir::Pattern const&)>, llvm::function_ref<mlir::LogicalResult (mlir::Pattern const&)>) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x6183BE: mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternList const&, unsigned int) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FB54: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /home/drom/work/github/llvm/circt/build/bin/firtool)
==5443==    by 0x69FC7F: mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /home/drom/work/github/llvm/circt/build/bi [... cut ...]
==5443==    indirectly lost: 0 bytes in 0 blocks
==5443==      possibly lost: 0 bytes in 0 blocks
==5443==    still reachable: 40 bytes in 1 blocks
==5443==         suppressed: 0 bytes in 0 blocks
==5443== Rerun with --leak-check=full to see details of leaked memory
==5443== 
==5443== Use --track-origins=yes to see where uninitialised values come from
==5443== For lists of detected and suppressed errors, rerun with: -s
==5443== ERROR SUMMARY: 10 errors from 5 contexts (suppressed: 0 from 0)
```

lattner, 2021-03-17:
I can't reproduce these valgrind issues on top of tree, can you still?

drom, 2021-03-18:
yes, I still see them

lattner, 2021-03-18:
Ok, I don't see this so I would appreciate it if someone else could take a look

youngar, 2021-03-19:
This should be fixed by https://github.com/llvm/circt/pull/792.  @drom You are going to have to rebuild LLVM this time :'(

drom, 2021-03-19:
@youngar challenge accepted!

lattner, 2021-03-19:
I wonder if any canonicalize speedups came in this merge

youngar, 2021-03-19:
I think we didn't have https://reviews.llvm.org/D98309 before

prithayan, 2021-03-25:
Still can't reproduce the valgrind issues on the gamma machines, with /work/plato/local/circt/bin/firtool

lattner, 2021-03-25:
I thought that @youngar fixed this by merging in some stuff from mlir trunk?

drom, 2021-03-25:
Yes, I don't see this issue anymore.

closed 2021-03-25 by drom