llvm/circt #778: [FIRRTL] mem 0-width (core dump)

drom, 2021-03-17:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input r0en: UInt<1>
    mem tmp41:
      data-type => UInt<8>
      depth => 10
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
      reader => r0
      writer => w0
    tmp41.r0.clk <= clock
    tmp41.r0.en <= r0en
    tmp41.r0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.clk <= clock
    tmp41.w0.en <= r0en
    tmp41.w0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.mask <= asUInt(SInt<25>(0))
    tmp41.w0.data <= asUInt(SInt<25>(0))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
1.       #0 0x000000000043cccd llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x43cccd)
 #1 0x000000000043aad4 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x43aad4)
 #2 0x000000000043ac33 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x43ac33)
 #3 0x00007f32ead06a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #4 0x000000000054673a mlir::detail::OperandStorage::OperandStorage(mlir::Operation*, mlir::ValueRange) (../../llvm/circt/build/bin/firtool+0x54673a)
 #5 0x000000000053b3b1 mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, unsigned int) (../../llvm/circt/build/bin/firtool+0x53b3b1)
 #6 0x000000000053b479 mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::BlockRange, mlir::RegionRange) (../../llvm/circt/build/bin/firtool+0x53b479)
 #7 0x000000000053b653 mlir::Operation::create(mlir::OperationState const&) (../../llvm/circt/build/bin/firtool+0x53b653)
 #8 0x00000000004f93ac mlir::OpBuilder::createOperation(mlir::OperationState const&) (../../llvm/circt/build/bin/firtool+0x4f93ac)
 #9 0x0000000000482ac5 circt::sv::ConnectOp mlir::OpBuilder::create<circt::sv::ConnectOp, mlir::Value&, mlir::Value&>(mlir::Location, mlir::Value&, mlir::Value&) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/Builders.h:399:31
#10 0x0000000000495d43 begin /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:253:45
#11 0x0000000000495d43 end /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:255:32
#12 0x0000000000495d43 ~SmallVector /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:1175:24
#13 0x0000000000495d43 (anonymous namespace)::FIRRTLLowering::visitDecl(circt::firrtl::MemOp) /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:1865:36
#14 0x0000000000497b34 operator()<circt::firrtl::MemOp> /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:211:53
#15 0x0000000000497b34 Case<circt::firrtl::MemOp, circt::firrtl::DeclVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchDeclVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:18)>&> /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:116:22
#16 0x0000000000497b34 Case<circt::firrtl::MemOp, circt::firrtl::NodeOp, circt::firrtl::RegOp, circt::firrtl::SMemOp, circt::firrtl::RegResetOp, circt::firrtl::WireOp, circt::firrtl::DeclVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchDeclVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:18)>&> /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:40:40
#17 0x0000000000497b34 Case<circt::firrtl::InstanceOp, circt::firrtl::MemOp, circt::firrtl::NodeOp, circt::firrtl::RegOp, circt::firrtl::SMemOp, circt::firrtl::RegResetOp, circt::firrtl::WireOp, circt::firrtl::DeclVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchDeclVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:18)>&> /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:41:42
#18 0x0000000000497b34 Case<circt::firrtl::CMemOp, circt::firrtl::InstanceOp, circt::firrtl::MemOp, circt::firrtl::NodeOp, circt::firrtl::RegOp, circt::firrtl::SMemOp, circt::firrtl::RegResetOp, circt::firrtl::WireOp, circt::firrtl::DeclVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchDeclVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:18)> > /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:41:42
#19 0x0000000000497b34 dispatchDeclVisitor /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:210:43
#20 0x0000000000497b34 visitInvalidStmt /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:270:37
#21 0x0000000000497b34 operator()<mlir::Operation*> /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:165:44
#22 0x0000000000497b34 Default<circt::firrtl::StmtVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchStmtVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:17)> > /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:125:21
#23 0x0000000000497b34 dispatchStmtVisitor /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:164:17
#24 0x0000000000497b34 visitInvalidExpr /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:267:37
#25 0x0000000000497b34 operator()<mlir::Operation*> /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:54:44
#26 0x0000000000497b34 Default<circt::firrtl::ExprVisitor<ConcreteType, ResultType, ExtraArgs>::dispatchExprVisitor<(anonymous namespace)::FIRRTLLowering, mlir::LogicalResult, {}>::<lambda(auto:15)> > /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/TypeSwitch.h:125:21
#27 0x0000000000497b34 dispatchExprVisitor /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:53:17
#28 0x0000000000497b34 dispatchVisitor /home/drom/work/github/llvm/circt/build/../include/circt/Dialect/FIRRTL/FIRRTLVisitors.h:262:37
#29 0x0000000000497b34 (anonymous namespace)::FIRRTLLowering::run() /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:1116:18
#30 0x0000000000499246 llvm::SmallVectorTemplateCommon<circt::sv::WireOp, void>::begin() /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:253:45
#31 0x0000000000499246 llvm::SmallVectorTemplateCommon<circt::sv::WireOp, void>::end() /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:255:32
#32 0x0000000000499246 llvm::SmallVector<circt::sv::WireOp, 6u>::~SmallVector() /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/ADT/SmallVector.h:1175:24
#33 0x0000000000499246 ~FIRRTLLowering /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:876:8
#34 0x0000000000499246 lowerModuleOperations /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:1098:3
#35 0x0000000000499246 (anonymous namespace)::FIRRTLModuleLowering::lowerModuleBody(circt::firrtl::FModuleOp, (anonymous namespace)::CircuitLoweringState&) /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:746:24
#36 0x000000000049b1f1 parallel_for_each_n<long unsigned int, (anonymous namespace)::FIRRTLModuleLowering::runOnOperation()::<lambda(auto:21)> > /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Parallel.h:162:25
#37 0x000000000049b1f1 parallelForEachN<(anonymous namespace)::FIRRTLModuleLowering::runOnOperation()::<lambda(auto:21)> > /home/drom/work/github/llvm/circt/build/../llvm/llvm/include/llvm/Support/Parallel.h:244:42
#38 0x000000000049b1f1 (anonymous namespace)::FIRRTLModuleLowering::runOnOperation() /home/drom/work/github/llvm/circt/build/../lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp:223:27
[... cut ...]
#43 0x00000000004080a6 processBufferIntoSingleStream /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:191:4
#44 0x00000000004080a6 main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:256:15
#45 0x00007f32ea774b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#46 0x000000000040989e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 26: 20292 Segmentation fault      (core dumped) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

firrtl-1.5.0-SNAPSHOT also crashes :(

lattner, 2021-03-17:
I'll take a look

lattner, 2021-03-17:
Fixed in 00ef911c, thanks!

closed 2021-03-17 by lattner