llvm/circt #781: [FIRRTL] Parse RawString ExtModule Parameters

seldridge, 2021-03-18:
The Scala FIRRTL Compiler supports passing "raw string" parameters to blackboxes. These have single quotes as opposed to double quotes.

As an example, the following:

```scala
module BlackBoxTypeParam :
  extmodule BlackBoxTypeParam :                                                                                    
    output out : UInt<1>                                                                                           
    defname = BlackBoxTypeParam                                                                                    
    parameter T = 'bit'                                                                                            
```

Becomes in the Scala FIRRTL Compiler:

```verilog
module Foo(
  output  out
);
  wire  bar_out;
  BlackBoxTypeParam #(.T(bit)) bar (
    .out(bar_out)
  );
  assign out = bar_out;
endmodule
```

We currently barf on the `'` when trying to parse this `.fir` file with `firtool`:

```
BlackBoxTypeParam.fir:5:19: error: unexpected character
    parameter T = 'bit'
```

As far as I can tell, if something is a raw string, it should be reproduced exactly in the Verilog output.

Ayushsunny, 2021-03-26:
Hello @seldridge I wanna work on this issue

seldridge, 2021-03-26:
@Ayushsunny: awesome, thanks. If you want to poke around, I think [this `switch` statement is missing a case to handle the raw string stuff](https://github.com/llvm/circt/blob/4d91f1bcaa5c3e1a59ba1136873bf45e873c7cda/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L2661).

The definition of a "RawString" is [here](https://github.com/chipsalliance/firrtl/blob/5a89fca6090948d0a99c217a09c692e58a20d1df/src/main/antlr4/FIRRTL.g4#L368).

Ayushsunny, 2021-03-27:
> @Ayushsunny: awesome, thanks. If you want to poke around, I think [this `switch` statement is missing a case to handle the raw string [...]

So do I have to add this to switch statement ?

``` CPP
   switch (getToken().getKind()) {
   case Token::l_paren:
   case Token::kw_memref:
   case Token::kw_tensor:
   case Token::kw_complex:
   case Token::kw_tuple:
   case Token::kw_vector:
   case Token::inttype:
   case Token::kw_bf16:
   case Token::kw_f16:
   case Token::kw_f32:
   case Token::kw_f64:
   case Token::kw_index:
   case Token::kw_none:
   case Token::exclamation_identifier:
     return failure(!(type = parseType()));

seldridge, 2021-03-30:
Hey, sorry. No, you shouldn't need all that. That switch statement is only handling the parsing of a FIRRTL external module parameter. All the possible cases are correctly handled except if there's a "raw string" using single quotes.

The specific issue, I think, is that that case needs to handle `FIRToken::raw_string` which is commented out here: https://github.com/llvm/circt/blob/4d91f1bcaa5c3e1a59ba1136873bf45e873c7cda/lib/Dialect/FIRRTL/Import/FIRTokenKinds.def#L55. (That probably needs to be uncommented...) I don't think the actual parsing is any different than the `FIRToken::string` case, but I'm not sure.

PR #1340 (2021-06-30): [FIRParser] Parse RawString ExtModule Parameters
Decided to tackle this as I'm familiarizing myself with the project. Please let me know of any needed changes.

Closes #781 and #874.
closed 2021-06-30 by lattner