llvm/circt #785: [FIRRTL] LowerTypes for Aggregate Register Inits

seldridge, 2021-03-18:
Vector register lowering appears to not work.

Consider the following circuit:

```scala
circuit Foo:
  module Foo:
    input clock: Clock
    input reset: UInt<1>
    input a_d: UInt<1>[2]
    output a_q: UInt<1>[2]

    wire init: UInt<1>[2]
    init[0] <= UInt<1>(0)
    init[1] <= UInt<1>(0)

    reg r: UInt<1>[2], clock with:
      reset => (reset, init)

    r <= a_d
    a_q <= r
```

This is crashing out in lower types:

```bash
irtool -lower-to-rtl -enable-lower-types Foo.fir
# Assertion failed: (entry && "bundle lowering was not set"), function getBundleLowering, file ../lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp, line 794.
```

I would expect this to generate something like what the Scala FIRRTL Compiler does here:

```verilog
module Foo(
  input   clock,
  input   reset,
  input   a_d_0,
  input   a_d_1,
  output  a_q_0,
  output  a_q_1
);
  reg  r_0;
  reg  r_1;
  assign a_q_0 = r_0;
  assign a_q_1 = r_1;
  always @(posedge clock) begin
    if (reset) begin
      r_0 <= 1'h0;
    end else begin
      r_0 <= a_d_0;
    end
    if (reset) begin
      r_1 <= 1'h0;
    end else begin
      r_1 <= a_d_1;
    end
  end
endmodule
```

seldridge, 2021-03-18:
h/t @youngar for triaging: We're just missing any handling of `RegResetOp` in `LowerTypes`. Oops.

prithayan, 2021-03-18:
I just checked that LowerTypes does not yet handle these firrtl declarations:, ```CMemOp, SMemOp, NodeOp, RegResetOp```. 
Not sure if how important the others are.

seldridge, 2021-03-18:
`CMemOp` and `SMemOp` are "CHIRRTL" (a super-dialect of FIRRTL IR that Chisel emits and includes those combinational and sequential mem accessor ops). These you can ignore. These need to be lowered to a `MemOp` before type lowering runs. 

`NodeOp` should be handled.

prithayan, 2021-03-22:
Fixed in https://github.com/llvm/circt/pull/790

closed 2021-03-22 by prithayan