llvm/circt #787: [FIRRTL] FIRRTL to RTL Memory Lowering Needs to Handle All Read Under Write Behaviors

seldridge, 2021-03-18:
Add support for the "old" read-under-write FIRRTL memory behavior. Currently, this is incorrectly emitted as "new".

The FIRRTL spec defines three read-under-write behaviors for memories. A read-under-write behavior is what happens if you try to read and write the same address on the same cycle These three behaviors are:

1. New (return the value in the memory when the read was made)
2. Old (return the value when the read was requested)
3. Undefined (technically anything, but the Scala FIRRTL Compiler treats this as "compiler's choice" and chooses "New")

Note that this only matters for read latencies > 0. These can be treated as equivalent if the read latency == 1.

lattner, 2021-03-18:
How do we know which behavior is required?  Is this known given a fir file or is this global?

If we have to support 2 or 3 things, we should probably model this as an attribute on the op.  It is ok for the FIR parser to have global behavior flags, but we don't want MLIR itself to have context sensitive behavior like this.

seldridge, 2021-03-18:
Sorry for the confusion. It's defined on the memory op:

```scala
    mem memory:
      data-type => UInt<8>
      depth => 16
      reader => r
      writer => w
      read-latency => 1
      write-latency => 1
      read-under-write => new
```

Where `read-under-write` can be any of `[new, old, undefined]`.

lattner, 2021-03-18:
oh awesome, already an attribute :)

mikeurbach, 2021-03-22:
> These can be treated as equivalent if the read latency == 1.

Should this say `== 0`? Just trying to understand what's expected.

I'm interested because HandshakeToFIRRTL uses `old` at the moment. It also uses read latency of 0 and write latency of 1, because that's all that used to be supported. My understanding is this should work right now regardless of this issue, but wanted to check.

https://github.com/llvm/circt/blob/72162617f62fff5ff12372fc94f81f8aa71b5af6/lib/Conversion/HandshakeToFIRRTL/HandshakeToFIRRTL.cpp#L1685-L1687

seldridge, 2021-03-22:
@mikeurbach: 
> Should this say == 0?

Yes. My bad. Editing that above.

mikeurbach, 2021-03-22:
:+1: no worries, thanks for clarifying.