llvm/circt #788: [FIRRTL] More alias copy propagation

seldridge, 2021-03-18:
This is really just more of #738 and #740.

CIRCT needs to work harder to eliminate or copy-propagate FIRRTL `Analog` types or Verilog `inout` types. `inout` support is _extremely_ spotty and weird across different tooling. 

Annoyingly, the typical situation where use of `Analog` or `inout` shows up in code is purely passing these from a module's IO to a submodule's IO. Doing the direct assignment is fine and supported by basically all tooling, but anything else is extremely spotty.

Consider something like:

```scala
circuit Foo:
  module Bar:
    input a: Analog<1>
  module Foo:
    input a: Analog<1>
    inst bar of Bar

    wire b: Analog<1>

    attach(b, a)
    attach(bar.a, b)
```

The `wire b: Analog<1>` is really just a temporary and can be trivially copy propagated through.

The Scala FIRRTL Compiler will get rid of this and produce:

```verilog
module Bar(
  inout   a
);
endmodule
module Foo(
  inout   a
);
  Bar bar (
    .a(a)
  );
endmodule
```

However, CIRCT is much less aggressive here and produces:

```verilog
module Bar(
  inout a);

endmodule

module Foo(
  inout a);

  wire _a_wire;	// Analog.fir:6:5
  wire b;	// Analog.fir:8:5

  Bar bar (	// Analog.fir:6:5
    .a (_a_wire)
  );
  `ifdef SYNTHESIS	// Analog.fir:10:5
    wire _T = b;	// Analog.fir:10:5
    assign b = a;	// Analog.fir:10:5
    assign a = _T;	// Analog.fir:10:5
    assign _a_wire = _T;	// Analog.fir:11:5
    assign b = _a_wire;	// Analog.fir:11:5
  `else
    `ifdef verilator	// Analog.fir:11:5
      `error "Verilator does not support alias and thus cannot arbitrarily connect bidirectional wires and ports"	// Analog.fir:10:5
      `error "Verilator does not support alias and thus cannot arbitrarily connect bidirectional wires and ports"	// Analog.fir:11:5
    `else
      alias b = a;	// Analog.fir:10:5
      alias _a_wire = b;	// Analog.fir:11:5
    `endif
  `endif
endmodule
```

This has consequences because just about any tool will handle the SFC code, but the CIRCT code is extremely problematic due to spotty tooling support (see below).

### Summary of Spotty Tooling

Thanks to @jackkoenig for finding and documenting all this. :+1:

Concretely, any tool is happy with:

```verilog
module Foo(inout bar);
  SubFoo SubFoo(bar);
endmodule
```

While entirely equivalent, if you introduce a temporary wire for `inout bar`, you're in trouble.

Synopsys Design Compiler will accept, but VCS will not accept:

```verilog
module Foo(inout bar);
  wire tmp;
  assign tmp = bar;
  assign bar = tmp;
  SubFoo SubFoo(tmp);
endmodule
```

VCS will accept this if you express this using `alias`:

```verilog
module Foo(inout bar);
  wire tmp;
  alias tmp = bar;
  SubFoo SubFoo(tmp);
endmodule
```

Verilator accepts neither the double assign nor alias.

lattner, 2021-03-19:
This is ridonculous but I'm not at all surprised by this.

lattner, 2021-03-19:
Fixing this is going to zap a ton of wires (virtually all of them), I'll start with something narrow.

lattner, 2021-03-19:
Part 1/2: 5f16ee51

lattner, 2021-03-19:
Fixed in 24ce8c9e, we now get:

module Foo(
  inout a);

  Bar bar (	// z.fir:6:5
    .a (a)
  );
endmodule

closed 2021-03-19 by lattner