llvm/circt #791: [FIRRTL] Lowertypes connect doesnot preserve source destination after lowering

prithayan, 2021-03-19:
The type lowering of aggregate connect operation does not preserve source and destination
```FIR
circuit Foo:
  module Foo:
    input clock: Clock
    input a_d: UInt<1>[2]
    output a_q: UInt<1>[2]

    reg r: UInt<1>[2], clock

    r <= a_d
    a_q <= r
```
generates this IR before lowering
```mlir
module  {
  firrtl.circuit "Foo" {
    firrtl.module @Foo(%clock: !firrtl.clock, %a_d: !firrtl.vector<uint<1>, 2>, %a_q: !firrtl.flip<vector<uint<1>, 2>>) {
      %r = firrtl.reg %clock {name = "r"} : (!firrtl.clock) -> !firrtl.vector<uint<1>, 2>
      firrtl.connect %r, %a_d : !firrtl.vector<uint<1>, 2>, !firrtl.vector<uint<1>, 2>
      firrtl.connect %a_q, %r : !firrtl.flip<vector<uint<1>, 2>>, !firrtl.vector<uint<1>, 2>
    }
  }
}
```
IR after LowerFIRRTLTypes, 
```mlir
firrtl.circuit "Foo" {
  firrtl.module @Foo(%clock: !firrtl.clock, %a_d_0: !firrtl.uint<1>, %a_d_1: !firrtl.uint<1>, %a_q_0: !firrtl.flip<uint<1>>, %a_q_1: !firrtl.flip<uint<1>>) {
    %r_0 = firrtl.reg %clock {name = "r_0"} : (!firrtl.clock) -> !firrtl.uint<1>
    %r_1 = firrtl.reg %clock {name = "r_1"} : (!firrtl.clock) -> !firrtl.uint<1>
    firrtl.connect %a_d_0, %r_0 : !firrtl.uint<1>, !firrtl.uint<1>
    firrtl.connect %a_d_1, %r_1 : !firrtl.uint<1>, !firrtl.uint<1>
    firrtl.connect %a_q_0, %r_0 : !firrtl.flip<uint<1>>, !firrtl.uint<1>
    firrtl.connect %a_q_1, %r_1 : !firrtl.flip<uint<1>>, !firrtl.uint<1>
  }
}
```
Here the ```firrtl.connect %a_d_0, %r_0 ``` is incorrect, it should instead be ```firrtl.connect  %r_0, %a_d_0```

youngar, 2021-03-19:
I believe this is fixed by https://github.com/llvm/circt/pull/688

prithayan, 2021-03-19:
This is now fixed,

closed 2021-03-19 by prithayan