llvm/circt #794: [FIRRTL] Memories with Multiple Write Ports Emit as Multiple Drivers

seldridge, 2021-03-19:
A memory with multiple write ports is incorrectly emitted as Verilog which looks like there are multiple drivers on the same signal. However, this is really just the appearance of multiple clocks that are all wired together and create multiple always blocks.

I think this is either one of two issues:

1. We need more copy-propagation that aggressively removes "temporaries" (something that a front end language created and marked as optimizable) or "synthetics" (something that CIRCT created). There's a balance here because we don't want to impair Verilog readability, but we also don't want to have temporary/synthetic soup.
2. Always block merging needs to be beefed up to look across wires.

This overlaps to some extent with the omnibus memory issue: #754.

Consider:

```scala
circuit Foo:
  module Foo:
    input clock: Clock
    input rAddr: UInt<4>
    input rEn: UInt<1>
    output rData: UInt<8>
    input wAddr: UInt<4>[2]
    input wEn: UInt<1>[2]
    input wMask: UInt<1>[2]
    input wData: UInt<8>[2]

    mem memory:
      data-type => UInt<8>
      depth => 16
      reader => r
      writer => w0,w1
      read-latency => 0
      write-latency => 1
      read-under-write => undefined

    memory.r.clk <= clock
    memory.r.en <= rEn
    memory.r.addr <= rAddr
    rData <= memory.r.data

    memory.w0.clk <= clock
    memory.w0.en <= wEn[0]
    memory.w0.addr <= wAddr[0]
    memory.w0.mask <= wMask[0]
    memory.w0.data <= wData[0]

    memory.w1.clk <= clock
    memory.w1.en <= wEn[1]
    memory.w1.addr <= wAddr[1]
    memory.w1.mask <= wMask[1]
    memory.w1.data <= wData[1]
```

This produces the following (trimmed) Verilog:

```verilog
  wire       memory_w0_clk;     // Foo.fir:12:5
  wire       memory_w1_clk;     // Foo.fir:12:5
  wire       _T_1;      // Foo.fir:12:5
  wire       _T_2;      // Foo.fir:12:5
  wire _T = memory_w0_en & wMask_0;     // Foo.fir:12:5
  always_ff @(posedge _T_1) begin       // Foo.fir:12:5
    if (_T)     // Foo.fir:12:5
      memory[memory_w0_addr] <= wData_0;        // Foo.fir:12:5
  end // always_ff @(posedge)
  wire _T_0 = memory_w1_en & wMask_1;   // Foo.fir:12:5
  always_ff @(posedge _T_2) begin       // Foo.fir:12:5
    if (_T_0)   // Foo.fir:12:5
      memory[memory_w1_addr] <= wData_1;        // Foo.fir:12:5
  assign _T_1 = memory_w0_clk;  // Foo.fir:12:5
  assign _T_2 = memory_w1_clk;  // Foo.fir:12:5
  assign memory_w0_clk = clock; // Foo.fir:26:19
  assign memory_w1_clk = clock; // Foo.fir:32:19
```

Tools will barf on this because "memory has multiple drivers". This should be emitted like how the SFC does this:

```verilog
  always @(posedge clock) begin
    if(memory_w0_en & memory_w0_mask) begin
      memory[memory_w0_addr] <= memory_w0_data;
    end
    if(memory_w1_en & memory_w1_mask) begin
      memory[memory_w1_addr] <= memory_w1_data;
    end
  end
```

lattner, 2021-03-20:
I see:

$ firtool z.fir -lower-to-rtl
z.fir:2:3: error: cannot lower this port type to RTL

seldridge, 2021-03-20:
@lattner wrote: 
> I see:
>
> $ firtool z.fir -lower-to-rtl
> z.fir:2:3: error: cannot lower this port type to RTL

This also needs an `-enable-lower-types` because I wrote the example with vector ports. 

However, that made me realize that this is actually unrelated to memories and actually an artifact of using temporary wires and passing around clocks. (It just happens that the memory lowering I put into LowerTypes tickles this.) Consider:

```mlir
module attributes {firrtl.mainModule = "Bar"}  {
  rtl.module @Bar(%data: i1, %clock: i1) -> (%rData: i1) {
    %reg = sv.reg  : !rtl.inout<i1>
    %0 = sv.read_inout %reg : !rtl.inout<i1>
    %clk1_wire = sv.wire  : !rtl.inout<i1>
    %clk2_wire = sv.wire  : !rtl.inout<i1>
    sv.connect %clk1_wire, %clock : i1
    sv.connect %clk2_wire, %clock : i1
    %clk1 = sv.read_inout %clk1_wire : !rtl.inout<i1>
    %clk2 = sv.read_inout %clk2_wire : !rtl.inout<i1>
    sv.alwaysff(posedge %clk1)  {
      sv.passign %reg, %data: i1
    }
    sv.alwaysff(posedge %clk2)  {
      sv.passign %reg, %data: i1
    }
    rtl.output %0 : i1
  }
}
```

This is treating each clock as separate when they actually aren't:

```verilog
module Bar(
  input  data, clock,
  output rData);

  reg  reg_0;   // Foo.mlir:3:12
  wire clk1_wire;       // Foo.mlir:5:18
  wire clk2_wire;       // Foo.mlir:6:18

  assign clk1_wire = clock;     // Foo.mlir:7:5
  assign clk2_wire = clock;     // Foo.mlir:8:5
  wire _T = clk1_wire;  // Foo.mlir:9:13
  wire _T_1 = clk2_wire;        // Foo.mlir:10:13
  always_ff @(posedge _T)       // Foo.mlir:11:5
    reg_0 <= data;      // Foo.mlir:12:7
  always_ff @(posedge _T_1)     // Foo.mlir:14:5
    reg_0 <= data;      // Foo.mlir:15:7
  assign rData = reg_0; // Foo.mlir:4:10, :17:5
endmodule
```

These always blocks should be merged, and ideally should wind up with an `always_ff @(posedge clock)` and not the temporary wire.

lattner, 2021-03-23:
This is because the `-firrtl-lower-types` pass is creating pointless temporary wires.  I will take a stab at extending the existing constant prop logic to also forward input ports, which will be a general solution.

seldridge, 2021-03-23:
> This is because the -firrtl-lower-types pass is creating pointless temporary wires.

Mea culpa.

seldridge, 2021-03-23:
Walking back the self-flagellation here a bit. The `*.mlir` snippet isn't _that_ weird. The following is totally legal FIRRTL to write and shouldn't produce multiple always blocks:

```scala
circuit Foo:
  module Foo:
    input clock: Clock
    input a: UInt<1>
    input b: UInt<1>
    output c: UInt<1>

    wire n: Clock
    n <= clock

    reg x: UInt<1>, clock
    reg y: UInt<1>, n

    x <= a
    y <= b

    c <= xor(x, y)
```

MFC:
```verilog
  always_ff @(posedge clock)    // Foo.fir:14:7
    x <= a;     // Foo.fir:14:7
  wire _T = n;  // Foo.fir:15:7
  always_ff @(posedge _T)       // Foo.fir:15:7
    y <= b;     // Foo.fir:15:7
  assign c = x ^ y;     // Foo.fir:2:3, :17:10
```

SFC:
```verilog
  always @(posedge clock) begin
    x <= a;
    y <= b;
  end
```

The MFC output isn't malformed here with multiple drivers, but lack of ability to see beyond the temporary wire produces a sub-optimal result.

Thanks for taking a look at this!

lattner, 2021-03-24:
Yeah, the actual issue here is that FIRRTL/SV isn't correctly composable.  Extraneous wires should be allowed to exist due to their electrical properties, but this abstraction is defeated by the pattern matching the tools are doing.

Yet another heinous day in the eda tools space.

The bulletproof way to handle this would be to make LowerTypes avoid generating the wire.  If there is a reasonable way to do that, please do.  We shouldn't depend on canonicalization for correctness.  However, it may not be "worth it" to be theoretically correct.

Cross reference Ram's points about wanting an -O0 though...

seldridge, 2021-03-24:
> We shouldn't depend on canonicalization for correctness.

100% agree.

I can revisit this to clean up the wire generation.

lattner, 2021-03-24:
Fixed in 0dff5b1d.

closed 2021-03-24 by lattner

jackkoenig, 2021-03-24:
> Cross reference Ram's points about wanting an -O0 though...

I have always struggled with what `-O0` in SFC should mean for this reason... We have this weak notion of "Legalization [for Verilog]" where we require certain limited optimizations even if others are disabled. It's a huge pain though as this example illustrates really well.