llvm/circt #796: [FIRRTL] Memory leak, possibly from creating bundle types

youngar, 2021-03-19:
In https://github.com/llvm/circt/issues/775, it looked like there was a memory leak when running fir tool with valgrind: 
```
==7544==    definitely lost: 67,728 bytes in 2 blocks
==7544==    indirectly lost: 0 bytes in 0 blocks
==7544==      possibly lost: 0 bytes in 0 blocks
==7544==    still reachable: 40 bytes in 1 blocks
==7544==         suppressed: 0 bytes in 0 blocks
```

I don't think there is a specific test case needed to reproduce. You can get more information out of valgrind by running `valgrind --leak-check=full --num-callers=50`

I briefly looked at the backtrace and it looked like something from a created bundle type was leaked. I don’t know what the issue is, but this might be relevant: https://github.com/llvm/llvm-project/commit/31bb8efd698304a8385ff79229ffbaa5613efdfb. This MLIR change is soon to be included in circt: https://github.com/llvm/circt/pull/792.

lattner, 2021-03-19:
cool thanks for catching and looking into this!  Do we need an equivalent of llvm_shutdown() ?

prithayan, 2021-03-24:
Seems like no more leakage, with the latest build. Run on gamma with /work/plato/local/circt/bin/firtool
```
==26293== Command: firtool --lower-to-rtl --enable-lower-types --verilog t.fir -o out.v
==26293==
==26293==
==26293== HEAP SUMMARY:
==26293==     in use at exit: 140,288 bytes in 2 blocks
==26293==   total heap usage: 2,781 allocs, 2,779 frees, 1,581,558 bytes allocated
==26293==
==26293== LEAK SUMMARY:
==26293==    definitely lost: 0 bytes in 0 blocks
==26293==    indirectly lost: 0 bytes in 0 blocks
==26293==      possibly lost: 0 bytes in 0 blocks
==26293==    still reachable: 140,288 bytes in 2 blocks
==26293==         suppressed: 0 bytes in 0 blocks
==26293== Reachable blocks (those to which a pointer was found) are not shown.
==26293== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==26293==
==26293== For counts of detected and suppressed errors, rerun with: -v
==26293== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```
For this input: 
```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input r0en: UInt<1>
    mem tmp41:
      data-type => UInt<8>
      depth => 10
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
      reader => r0
      writer => w0
    tmp41.r0.clk <= clock
    tmp41.r0.en <= r0en
    tmp41.r0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.clk <= clock
    tmp41.w0.en <= r0en
    tmp41.w0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.mask <= asUInt(SInt<25>(0))
    tmp41.w0.data <= asUInt(SInt<25>(0))
```

youngar, 2021-03-25:
Thanks for looking in to this!

closed 2021-03-25 by youngar