llvm/circt #797: Constant Prop to Avoid False Index Out of Bounds

seldridge, 2021-03-19:
Some downstream tools will emit warnings or errors if a vector can be indexed out of bounds. However, not all tools are smart enough to know when something is actually a constant and can't be out of bounds.

Consider the following two Verilog circuits. Both contain a single-element vector indexed by a 1-bit wire. However, `Foo` cannot cause an out-of-bounds error (unless you zap it with some Verilog `force` thing). `Bar` can result in an out-of-bounds error if `addr` is `1'b1`.

```verilog
module Foo(
  output rData);

  reg  memory[0:0];     
  wire memory_r_addr;   

  assign memory_r_addr = 1'h0;  
  assign rData = memory[memory_r_addr]; /* False index out-of-bounds */
endmodule
```

```verilog
module Bar(
  input  addr,
  output rData);

  reg memory[0:0];	

  assign rData = memory[addr]; /* Actual index out-of-bounds (from what is visible here...) */
endmodule
```

Emitting things like `Foo` is problematic because if I'm running a specific downstream tool, I may need to globally disable index out-of-bounds warnings/errors in order to get it to work. (I have a tool here that won't accept `Foo` unless I turn off out-of-bounds index warnings.) However, I'm then missing things like `Bar`. 

There are a couple of ways to handle this:

1) It would be good if this was constant propagated or copy propagated (I'm a broken record...). It generally seems like a good idea to move constants into indexes if possible. Note that this is weird because this is an optimization required for "correctness". See discussion below.
2) We could add per-tool options to emit the code that specific tools want or add Verilog attributes/comments that disable specific checks on specific lines. E.g., `--vendor=<vendor> --tool=<tool>`. This is going to get very tedious and a massive amount of work until we take over the world and EDA vendors start hiring people to improve our output to be compatible with their tools. Likely this is a punt for now.
3) We don't handle it and we start some documentation page/rationale doc to explain quirks of different tools and why we do/don't emit certain things a certain way. This may also include canned snippets for how to run things. This may bump up against legal problems if it exposes vendor tool APIs.

### Discussion

#### Constant Propagation Required for "Correctness"?

It's weird here that in order for our Verilog to be "correct" (by some definition of a downstream tool) we need to run an _optimization_ pass. (@youngar has brought up this exact point before and may have some thoughts.) I generally don't like this, but there may be some analogies from software compilers that we can use here? 

Or: hardware and the hardware ecosystem is a patchwork of different tools accepting different things and we turn on specific optimizations by default. Deviating from specific optimizations being on may cause problems with tools.

#### Scope of Analysis

In order for this to really be "correct" we need global constant propagation. E.g., the `Bar` circuit could be totally safe if it's instantiator is passing a `1'b0` into its `addr` port. This is problematic/annoying because: (1) it's a global analysis and (2) it turns a port into dead code which may be undesirable for an end user. E.g., some verification engineer may expect every instantiation of `Bar` to have an `addr` port. If we remove it or stop using it internal to `Bar` this may cause problems.

lattner, 2021-03-20:
I'm not sure how to fix this.  For the specific case mentioned, we can always emit array_index of a zero sized array with a zero literal (given that anything else would be UB).  However, I don't see how this solves the general non-pow-2 case.

What does SFC do?

darthscsi, 2021-03-23:
How do the tools with problems here react to non-power-of-two arrays?  This happens often enough in human code. 
If this is just a problem with 1-length arrays, we could canonicalize away the array or always use a constant as @lattner suggests.

seldridge, 2021-03-23:
Note that this is a size-1 array, not a size-0 array. (I was trying to make the example as simple as possible...) The problem generalizes to any non-power-of-2-sized-array.

> What does SFC do?

The SFC has constant propagation and dodges this specific manifestation (indexing by a constant) if optimizations are turned on.

It won't handle the general non-power-of-2-case. It *could* have a solution where waivers are automatically emitte if a non-power-of-2-sized array was indexed by an _interval_ type that the type system says cannot exceed the array size. (It does not do this now.) However, interval types are a very niche part of the SFC that were never widely used.

> How do the tools with problems here react to non-power-of-two arrays? 

You need to disable this check on a per-tool basis (either globally or with some per-line waiver).

-----

I think there's two separate issues here:

1. We need constant propagation to get rid of obviously fine code because downstream tooling is dumb. This is really just a special case of: https://github.com/llvm/circt/issues/804.
2. There might be a solution to the general non-power-of-2-case where waivers can be automatically emitted if some dialect has enough information in its type system to know what values a wire can take.

I propose constant prop discussion moves to #804 and this issue tracks a solution for (2). Or: we opt to close this.

seldridge, 2021-03-23:
Example: Formality won't run without the user either (1) globally suppressing this error or (2) making this error a warning. You can see code examples doing this with a [GitHub search](https://github.com/search?p=1&q=FMR_ELAB-147&type=Code).

lattner, 2021-03-23:
We need to improve constant propagation, I think it make sense to track that as part of this bug.

lattner, 2021-03-23:
This case is handled in eeed4fa8, but we also need to implement fold hooks for all the primitives: Issue #809

closed 2021-03-23 by lattner