llvm/circt #801: [comb] Canonicalize away comb.extract from comb.concat

lattner, 2021-03-20:
I see stuff like this in generated Verilog:

```
module foo(input  [127:0] io_in_data_1 ...)

wire [22:0] _T_38 = io_in_data_1[22:0];
wire [23:0] _T_39 = {1'h0, _T_38};
wire [4:0] _tmp_1030 = _T_39[3] ? 5'h14 : _T_39[2] ? 5'h15 : _T_39[1] ? 5'h16 : 5'h17;
```

Given our cost model, we can always forward substitute the things being concat'd into a comb.extract, so we should be able to simplify this to something like:

```
wire [4:0] _tmp_1030 = io_in_data_1[3] ? 5'h14 : io_in_data_1[2] ? 5'h15 : io_in_data_1[1] ? 5'h16 : 5'h17;`
```

lattner, 2021-03-20:
This looks like this in the IR:

```
   %22 = comb.extract %io_in_data_0 from 0 : (i128) -> i23
    %23 = comb.concat %false, %22 : (i1, i23) -> i24
    %24 = comb.extract %23 from 23 : (i24) -> i1
    %25 = comb.extract %23 from 22 : (i24) -> i1
    %26 = comb.extract %23 from 21 : (i24) -> i1
    %27 = comb.extract %23 from 20 : (i24) -> i1
    %28 = comb.extract %23 from 19 : (i24) -> i1
    %29 = comb.extract %23 from 18 : (i24) -> i1
    %30 = comb.extract %23 from 17 : (i24) -> i1
    %31 = comb.extract %23 from 16 : (i24) -> i1
    %32 = comb.extract %23 from 15 : (i24) -> i1
    %33 = comb.extract %23 from 14 : (i24) -> i1
    %34 = comb.extract %23 from 13 : (i24) -> i1
    %35 = comb.extract %23 from 12 : (i24) -> i1
    %36 = comb.extract %23 from 11 : (i24) -> i1
    %37 = comb.extract %23 from 10 : (i24) -> i1
    %38 = comb.extract %23 from 9 : (i24) -> i1
    %39 = comb.extract %23 from 8 : (i24) -> i1
    %40 = comb.extract %23 from 7 : (i24) -> i1
    %41 = comb.extract %23 from 6 : (i24) -> i1
    %42 = comb.extract %23 from 5 : (i24) -> i1
    %43 = comb.extract %23 from 4 : (i24) -> i1
    %44 = comb.extract %23 from 3 : (i24) -> i1
    %45 = comb.extract %23 from 2 : (i24) -> i1
    %46 = comb.extract %23 from 1 : (i24) -> i1
    %47 = comb.mux %46, %c-10_i5, %c-9_i5 : i5
    %48 = comb.mux %45, %c-11_i5, %47 : i5
    %49 = comb.mux %44, %c-12_i5, %48 : i5
    %50 = comb.mux %43, %c-13_i5, %49 : i5
    %51 = comb.mux %42, %c-14_i5, %50 : i5
    %52 = comb.mux %41, %c-15_i5, %51 : i5
    %53 = comb.mux %40, %c-16_i5, %52 : i5
    %54 = comb.mux %39, %c15_i5, %53 : i5
    %55 = comb.mux %38, %c14_i5, %54 : i5
    %56 = comb.mux %37, %c13_i5, %55 : i5
    %57 = comb.mux %36, %c12_i5, %56 : i5
    %58 = comb.mux %35, %c11_i5, %57 : i5
    %59 = comb.mux %34, %c10_i5, %58 : i5
```

fyquah, 2021-08-21:
I believe https://github.com/llvm/circt/pull/1423 addressed this ? The above example will be transformed via `extract(cat(extract(..))) ->(extract(extract(...))) -> extract(extract(..))`?

lattner, 2021-08-21:
Yes, thank you!

closed 2021-08-21 by lattner