llvm/circt #803: Implement Scala FIRRTL Compiler Dead Code Elimination

seldridge, 2021-03-22:
Add an option that implements the same policy/algorithm of the Scala FIRRTL Compiler for doing dead code elimination. 

The following tests should all pass (possibly with some options/annotations applied to either the Scala FIRRTL Compiler or to CIRCT): https://github.com/chipsalliance/firrtl/blob/master/src/test/scala/firrtlTests/DCETests.scala.

lattner, 2021-03-26:
I'll work on this a bit.  First step here: 06adb4f9

lattner, 2021-03-27:
MLIR has a very nice and generic SCCP pass, but after digging into it a bit, it won't handle FIRRTL's connect semantics and output arguments :-(.   Fortunately, our needs are a lot simpler than what it supports, but it is still a bummer.

seldridge, 2021-03-27:
Thanks for digging into this, @lattner. :+1: Guidance on what we can reuse vs. needs to be built is great.

dtzSiFive, 2023-06-05:
Closing due to our current IMDCE pass implementation handling this -- please re-open if that doesn't cover this.