llvm/circt #804: Implement Scala FIRRTL Compiler Constant Propagation

seldridge, 2021-03-22:
Add a pass to the FIRRTL Dialect that implements Scala FIRRTL Compiler-style constant propagation. 

The tests in the Scala FIRRTL Compiler should provide guidance here: https://github.com/chipsalliance/firrtl/blob/master/src/test/scala/firrtlTests/ConstantPropagationTests.scala

fabianschuiki, 2021-03-29:
Reminder: We should replace the current uses of `constFoldBinaryOp` with a custom function that handles zero-/sign-extension to a unified bit width, depending on the result type. Currently `constFoldBinaryOp` only folds constants of identical type, but FIRRTL allows for a mix-and-match.

dtzSiFive, 2023-06-05:
Okay to close, with our current IMCP implementation + folders?

fabianschuiki, 2023-06-05:
Yeah I think we cover this :+1:

closed 2023-06-05 by dtzSiFive