llvm/circt #806: [LLHD]llhd-sim unable to simulate generated llhd dialect

LindYu, 2021-03-22:
To simulate below llhd:

```
llhd.entity @adder(%clk: !llhd.sig, %res: !llhd.sig) → (%i5_output: !llhd.sig ) {
%0 = llhd.const #llhd.time<0s, 0d, 1e> : !llhd.time
%1 = llhd.const 16 : i32
llhd.drv %i5_output, %1 after %0 : !llhd.sig
}
```

shows this error:

```
Assertion failed: (idx < size()), function operator[], file …/llvm/llvm/include/llvm/ADT/SmallVector.h, line 277.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0. Program arguments: …/build/bin/llhd-sim addersvllhd.mlir -n 10 -r adder
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var LLVM_SYMBOLIZER_PATH to point to it):
0 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011dca593d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 61
1 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011dca5eeb PrintStackTraceSignalHandler(void*) + 27
2 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011dca3a6b llvm::sys::RunSignalHandlers() + 123
3 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011dca83f8 SignalHandler(int) + 232
4 libsystem_platform.dylib 0x00007fff6af445fd _sigtramp + 29
5 libdyld.dylib 0x00007fff6ad4a78f dyldGlobalLockRelease() + 0
6 libsystem_c.dylib 0x00007fff6ae1a808 abort + 120
7 libsystem_c.dylib 0x00007fff6ae19ac6 err + 0
8 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011db4fa48 llvm::SmallVectorTemplateCommon<circt::llhd::sim::Signal, void>::operator[](unsigned long) + 104
9 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011db4f912 driveSignal + 178
10 libcirct-llhd-signals-runtime-wrappers.dylib 0x000000011e1240f1 driveSignal + 6113425
11 llhd-sim 0x000000010acda78e main + 1518
12 libdyld.dylib 0x00007fff6ad4bcc9 start + 1
```

My command to llhd-sim is -n 10 -r adder and I think the expect behaviour should be a constant signal at the output (signal i5_output being constantly at value 16). I was able to trace the code till line 235 of engine.cpp 

(ps: this is not the full code for my sv; I have reduced it to the simplest form to see if I can get an output. Originally my code just aborted when the reaches the adder.adder entity in state.)

fabianschuiki, 2021-03-29:
Hi @LindYu, thanks for the nice minimalized test case! Will take a look :+1:

rodonisi, 2021-04-05:
Hi @LindYu , sorry for the late answer. The main issue I can see in this example is having a root with some input/output ports. In it’s current state `llhd-sim` kinda blindly assumes the root to always be a "portless testbench root entity", as is the usual case with `moore`’s output (when compiling a simulation testbench). So what happens here, I think, is that we’re pretty much trying to access undefined signals, as there's no LLHD defining what signals are connected to the root's ports. @fabianschuiki do you think it should always be the case that we have a portless root when simulating LLHD, and should thus enforce it, or should we have some way for `llhd-sim` to detect and instantiate undefined signals with some default value in the future?

For the time being, by adding such a portless root entity it should be possible to simulate this example. E.g., by adding the following entity:
```
llhd.entity @adder_tb() -> () {
  %0 = llhd.const 0 : i32
  %1 = llhd.sig "s" %0 : i32
  %2 = llhd.const 0 : i1
  %3 = llhd.sig "clk" %2 : i1
  llhd.inst "proc" @adder (%3, %1) -> (%1) : (!llhd.sig<i1>, !llhd.sig<i32>) -> (!llhd.sig<i32>)
}
```
I’m able to simulate the adder (using `-n 10 -r adder_tb` for the flags)  with output:
```
0ps 0d 0e  adder_tb/clk  0x00
0ps 0d 0e  adder_tb/proc/clk  0x00
0ps 0d 0e  adder_tb/proc/s  0x00000000
0ps 0d 0e  adder_tb/s  0x00000000
0ps 0d 1e  adder_tb/proc/s  0x00000010
0ps 0d 1e  adder_tb/s  0x00000010
Finished at 0ps 0d 2e (3 cycles)
```

LindYu, 2021-04-05:
Hi @rodonisi, thank you so much for your reply! With regarding to the moore usually generating a portless testbench, I used moore to convert a sv file to llhd without changing anything so I am not sure why it didn't generate a portless testbench this time. How should I structure my sv for moore to avoid this? 

I also noticed that the testbenches in LLHD/simulator are mostly portless with a single entity, is there a multi-module test case that users can reference in integration tests?

rodonisi, 2021-04-07:
Hi @LindYu, you're very welcome, thanks for bringing this up!
Regarding more complex integration tests, no, I don't think we currently any available in this repo, but you are welcome to refer to the compiled examples I used during the early development of `llhd-sim` [here]( https://github.com/rodonisi/llhd-extras/tree/master/examples), and the respective sv files [here]( https://github.com/rodonisi/llhd-extras/tree/master/examples-sv). @fabianschuiki would it also make sense to move them here as integration tests now that we have such a structure?

As for the general structure of the sv files when working with `moore`, I would let @fabianschuiki give a more detailed answer as I'm not very  SystemVerilog-savvy, but my general understanding is that one usually defines a SystemVerilog testbench in a separate sv file and compiles that together with the design, which should result in an llhd/mlir file that can be simulated.

maerhart, 2024-07-19:
Closing this in response to 2345382e672b652e3195449db3b2813b78622f7a

closed 2024-07-19 by maerhart