llvm/circt #808: Support Verilog Implementation Limits

seldridge, 2021-03-23:
The Verilog spec (1364-2005 was what I looked at) lets tools define limits on what they will accept. This issue covers building in the minimum limits as defaults with the ability to exceed them optionally. A user should be able to set a specific limit if they want.

The limits that I am aware of are:

- [ ] Implementations may set a limit on the maximum length of a vector, but the limit shall be at least
65,536 (2^16) bits.
- [ ] Implementations may limit the maximum size of an array, but they shall allow at least
16,777,216 (2^24) elements.
- [ ] Implementations may limit the maximum number of ports in a module definition, but the limit shall be at least 256.
- [ ] Implementations may set a limit on the maximum length of identifiers, but the limit shall be at least
1024 characters.

Identifiers is the easy one. 

The others require changes to the actual circuit, e.g., if a module has too many ports, then you could combine ports.

seldridge, 2021-08-17:
Note that we actually have a test that is over the implementation limit for vector length: https://github.com/llvm/circt/blob/415b84213f959db6fad3fff23765c2cef607c1fd/test/ExportVerilog/hw-dialect.mlir#L599.

lattner, 2021-08-18:
We can tweak the test to use some other "large in verilog" literal, it isn't precious.