llvm/circt #809: [FIRRTL] Need constant folding for all the primitives

lattner, 2021-03-23:
We're missing constant folding for a bunch of primitives, e.g. this should be handled by a `fold` method:

```
  firrtl.module @wire_cst_prop1(%out_b: !firrtl.flip<uint<9>>) {
      %c5_ui8 = firrtl.constant(5 : ui8) : !firrtl.uint<8>
      %0 = firrtl.add %c5_ui8, %c5_ui8 : (!firrtl.uint<8>, !firrtl.uint<8>) -> !firrtl.uint<9>
      firrtl.connect %out_b, %0 : !firrtl.flip<uint<9>>, !firrtl.uint<9>
    }
```

prithayan, 2021-03-24:
Should we add a fold method for every ```BinaryPrimOp``` ?  Currently it does not have ```hasFolder=1```,  
https://github.com/llvm/circt/blob/main/include/circt/Dialect/FIRRTL/FIRRTLExpressions.td#L152 
Why not add it to the base class, instead of specific ops only, 
```python
def AndPrimOp : BinaryPrimOp<"and", IntType, IntType, UIntType, [Commutative]> {
  let hasFolder = 1;
}
```

seldridge, 2021-03-24:
This should _at least_ implement any fold that sub-types [`SimplifyBinaryOp`, `FoldCommutativeOp`, or `SimplifyReductionOp` in the Scala FIRRTL Compiler](https://github.com/chipsalliance/firrtl/blob/09c4baf90ca61c06fd87467110f62db5b6a999c2/src/main/scala/firrtl/transforms/ConstantPropagation.scala#L127).

lattner, 2021-03-24:
yes, go further though - it should be added to `PrimOp` in tblgen.

prithayan, 2021-03-25:
Sounds good, we can include ``hasFolder =1`` in ``PrimOp``. 
I created a PR just for the Add op, will include each of the operators in separate PRs.
https://github.com/llvm/circt/pull/827

lattner, 2021-05-18:
Finished here: 299c8782

closed 2021-05-18 by lattner