llvm/circt #812: [FIRRTL] unexpected stdout

drom, 2021-03-24:
The following FIRRTL program

```python
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input r0en: UInt<1>
    mem tmp41:
      data-type => UInt<8>
      depth => 10
      read-latency => 0
      write-latency => 1
      read-under-write => undefined
      reader => r0
      writer => w0
    tmp41.r0.clk <= clock
    tmp41.r0.en <= r0en
    tmp41.r0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.clk <= clock
    tmp41.w0.en <= r0en
    tmp41.w0.addr <= asUInt(SInt<25>(0))
    tmp41.w0.mask <= asUInt(SInt<25>(0))
    tmp41.w0.data <= asUInt(SInt<25>(0))
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` produces unexpected `stdout` output:

```
%tmp41_r0_addr = sv.wire  : !rtl.inout<i4>
%tmp41_r0_addr = sv.wire  : !rtl.inout<i4>
%tmp41_r0_en = sv.wire  : !rtl.inout<i1>
%tmp41_r0_en = sv.wire  : !rtl.inout<i1>
%tmp41_r0_clk = sv.wire  : !rtl.inout<i1>
%tmp41_r0_clk = sv.wire  : !rtl.inout<i1>
%tmp41_r0_data = sv.wire  : !rtl.inout<i8>
%tmp41_r0_data = sv.wire  : !rtl.inout<i8>
%tmp41_w0_addr = sv.wire  : !rtl.inout<i4>
%tmp41_w0_addr = sv.wire  : !rtl.inout<i4>
%tmp41_w0_en = sv.wire  : !rtl.inout<i1>
%tmp41_w0_en = sv.wire  : !rtl.inout<i1>
%tmp41_w0_clk = sv.wire  : !rtl.inout<i1>
%tmp41_w0_clk = sv.wire  : !rtl.inout<i1>
%tmp41_w0_data = sv.wire  : !rtl.inout<i8>
%tmp41_w0_data = sv.wire  : !rtl.inout<i8>
%tmp41_w0_mask = sv.wire  : !rtl.inout<i1>
%tmp41_w0_mask = sv.wire  : !rtl.inout<i1>
```

youngar, 2021-03-24:
This was introduced by accident earlier today, fix here: https://github.com/llvm/circt/pull/814

drom, 2021-03-24:
LGTM

closed 2021-03-24 by drom

youngar, 2021-03-24:
Thanks Aleksei!