llvm/circt #813: [FIRRTL] mem (core dump)

drom, 2021-03-24:
The following FIRRTL program

```python
; seed: 27
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_b: SInt<1>
    input inp_d: UInt<11>
    input inp_g: UInt<18>
    input inp_h: SInt<3>
    input inp_i: SInt<1>
    reg tmp11: UInt<1>, clock
    reg tmp12: UInt<3>, clock
    reg _tmp14: UInt<1>, clock with: (reset => (reset, UInt<1>(0)))
    wire tmp15: UInt<18>
    reg tmp19: SInt<5>, clock with: (reset => (reset, SInt<5>(0)))
    wire _tmp22: SInt<20>
    wire tmp32: SInt<32>
    tmp11 <= neq(inp_i, inp_h)
    tmp12 <= not(inp_h)
    _tmp14 <= rem(asUInt(SInt<12>("b11100011011")), tmp11)
    tmp15 <= div(inp_g, UInt<6>("h18"))
    _tmp22 <= mul(inp_b, asSInt(UInt<19>("h2c97e")))
    tmp32 <= dshl(SInt<1>("b-1"), head(tmp15, 5))
    mem tmp33:
      data-type => UInt<5>
      depth => 12
      read-latency => 2
      write-latency => 1
      read-under-write => undefined
      reader => r0
      reader => r1
      reader => r2
      writer => w0
      writer => w1
    tmp33.r0.clk <= clock
    tmp33.r0.en <= head(UInt<26>("b1110100010010001011101110"), 1)
    tmp33.r0.addr <= UInt<7>("b1110111")
    tmp33.r1.clk <= clock
    tmp33.r1.en <= head(asUInt(SInt<26>("b-1010000101001100000001011")), 1)
    tmp33.r1.addr <= asUInt(_tmp22)
    tmp33.r2.clk <= clock
    tmp33.r2.en <= head(UInt<14>(11714), 1)
    tmp33.r2.addr <= inp_d
    tmp33.w0.clk <= clock
    tmp33.w0.en <= head(asUInt(SInt<25>("b10100111000010001001010")), 1)
    tmp33.w0.addr <= UInt<19>(357565)
    tmp33.w0.data <= UInt<9>(118)
    tmp33.w0.mask <= tmp12
    tmp33.w1.clk <= clock
    tmp33.w1.en <= head(asUInt(SInt<21>("o3002125")), 1)
    tmp33.w1.addr <= asUInt(SInt<5>("o10"))
    tmp33.w1.data <= asUInt(tmp32)
    tmp33.w1.mask <= tmp12
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
firtool
firtool: /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/ScopedHashTable.h:244: llvm::ScopedHashTableScope<K, V, KInfo, AllocatorTy>::~ScopedHashTableScope() [with K = mlir::Operation*; V = mlir::Operation*; KInfo = {anonymous}::SimpleOperationInfo; AllocatorTy = llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*> >]: Assertion `HT.TopLevelMap[ThisEntry->getKey()] == ThisEntry && "Scope imbalance!"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
 #0 0x00000000004bc7a7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004bc85e PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004ba85e llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004bc1fa SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007fc96f4f5a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #5 0x00007fc96ef794a5 raise (/lib64/libc.so.6+0x3d4a5)
 #6 0x00007fc96ef62864 abort (/lib64/libc.so.6+0x26864)
 #7 0x00007fc96ef62749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x00007fc96ef719b6 (/lib64/libc.so.6+0x359b6)
 #9 0x00000000006f65ce llvm::ScopedHashTableScope<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >::~ScopedHashTableScope() /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/ScopedHashTable.h:244:7
#10 0x00000000006f60e8 (anonymous namespace)::CSE::simplifyRegion(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >&, mlir::DominanceInfo&, mlir::Region&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:169:43
#11 0x00000000006f5f47 (anonymous namespace)::CSE::simplifyBlock(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >&, mlir::DominanceInfo&, mlir::Block*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:150:7
#12 0x00000000006f60d9 (anonymous namespace)::CSE::simplifyRegion(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >&, mlir::DominanceInfo&, mlir::Region&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:169:43
#13 0x00000000006f6381 (anonymous namespace)::CSE::runOnOperation() /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:219:3
#14 0x00000000008cda85 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:403:36
#15 0x00000000008cdcb1 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:444:15
#16 0x00000000008cfdf1 mlir::PassManager::run(mlir::Operation*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:896:74
#17 0x000000000040dd68 processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) (../../llvm/circt/build/bin/firtool+0x40dd68)
#18 0x000000000040b203 main (../../llvm/circt/build/bin/firtool+0x40b203)
#19 0x00007fc96ef63b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#20 0x000000000040b58e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: строка 26:  9875 Aborted                 (стек памяти сброшен на диск) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

firrtl-1.5.0-SNAPSHOT produces this Verilog:

```verilog
module top_mod(
  input         clock,
  input         reset,
  input         inp_b,
  input  [10:0] inp_d,
  input  [17:0] inp_g,
  input  [2:0]  inp_h,
  input         inp_i
);
endmodule
```

youngar, 2021-03-24:
A more minimal test case:
```mlir
  rtl.module @top_mod(%in0: i1, %in1 : i3) -> (i1, i1) {
    %0 = comb.and %in0, %4 : i1
    %1 = comb.and %in0, %3 : i1
    %3 = comb.extract %in1 from 0 : (i3) -> i1
    %4 = comb.extract %in1 from 0 : (i3) -> i1
    rtl.output %0, %1 : i1, i1
  }
```

CSE uses a hashtable of `Op -> Op` to keep track of seen values as it walks forward through a block.  There is the invariant that you can not change a key after it has been entered in to the hashtable.  When CSE runs it RAU of `%4` with `%3`, which modifies `%0`.  Since this will change the hash of `%0`, things kind of explode.

fabianschuiki, 2021-03-24:
My guess is that the CSE algorithm has been written with the assumption that the operations in the region follow def-before-use. Since the `rtl.module` is purely a data flow thing, there's probably no def-before-use order of the operations that we could establish before doing CSE (because there are going to be cycles through registers). So we'd need a way to update the hash under which an operation is stored in the table. One way that might work when eliminating an operation would be to temporarily remove all uses of the operation from the table of known values, replace the uses, then re-add the operations:

https://github.com/llvm/llvm-project/blob/f178c13fa89960c7247a6367269919acf87fd1b3/mlir/lib/Transforms/CSE.cpp#L116-L119

lattner, 2021-03-26:
FYI I disabled the RTL level CSE pass in this patch f9e71027 as a workaround for this.

youngar, 2021-04-01:
This is fixed upstream, and RTL level CSE has been enabled by https://github.com/llvm/circt/pull/867.

closed 2021-04-01 by youngar

lattner, 2021-04-01:
Awesome, thank you so much for fixing this @youngar !