llvm/circt #819: [FIRRTL] !FoundVal && "Key already in new map? (core dump)

drom, 2021-03-24:
The following FIRRTL program (Sorry it is quite long :( )

```python
; seed: 1184
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input arst: AsyncReset
    input inp_a: SInt<18>
    input inp_b: SInt<14>
    input inp_c: SInt<2>
    input inp_d: SInt<18>
    input inp_e: SInt<19>
    input inp_f: UInt<13>
    input inp_g: UInt<12>
    input inp_h: SInt<4>
    input inp_i: UInt<20>
    input inp_j: UInt<2>
    output tmp14: SInt<17>
    output _tmp15: UInt<16>
    output _tmp22: UInt<25>
    output tmp23: SInt<62>
    output _tmp32: UInt<24>
    output tmp36: UInt<1>
    wire tmp11: SInt<9>
    reg tmp12: SInt<9>, clock
    wire tmp13: SInt<9>
    reg tmp16: UInt<52>, clock
    wire tmp18: SInt<53>
    reg tmp19: UInt<7>, clock
    reg tmp21: SInt<10>, clock with: (reset => (arst, SInt<10>(0)))
    reg tmp24: SInt<48>, clock
    wire tmp26: SInt<3>
    reg tmp27: SInt<3>, clock with: (reset => (reset, SInt<3>(0)))
    wire tmp28: UInt<3>
    reg tmp29: SInt<136>, clock with: (reset => (reset, SInt<136>(0)))
    wire _tmp33: UInt<35>
    wire _tmp41: UInt<61>
    reg tmp44: SInt<8>, clock with: (reset => (reset, SInt<8>(0)))
    mem tmp10:
      data-type => SInt<9>
      depth => 5
      read-latency => 3
      write-latency => 1
      read-under-write => undefined
      reader => r0
      reader => r1
      reader => r2
      writer => w0
      writer => w1
    tmp10.r0.clk <= clock
    tmp10.r0.en <= head(UInt<20>("hed052"), 1)
    tmp10.r0.addr <= inp_f
    tmp10.r1.clk <= clock
    tmp10.r1.en <= head(asUInt(inp_h), 1)
    tmp10.r1.addr <= UInt<4>("h3")
    tmp10.r2.clk <= clock
    tmp10.r2.en <= head(asUInt(inp_b), 1)
    tmp10.r2.addr <= UInt<9>(407)
    tmp10.w0.clk <= clock
    tmp10.w0.en <= head(asUInt(SInt<28>("o-677523445")), 1)
    tmp10.w0.addr <= inp_i
    tmp10.w0.data <= inp_d
    tmp10.w0.mask <= asUInt(inp_h)
    tmp10.w1.clk <= clock
    tmp10.w1.en <= head(asUInt(SInt<24>(-2844022)), 1)
    tmp10.w1.addr <= inp_j
    tmp10.w1.data <= asSInt(inp_f)
    tmp10.w1.mask <= asUInt(inp_e)
[... 110 lines cut ...]
#10 0x00000000006f9bd3 llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::grow(unsigned int) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:816:53
#11 0x00000000006f9888 llvm::DenseMapBase<llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >, mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::grow(unsigned int) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:538:3
#12 0x00000000006f9265 llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*>* llvm::DenseMapBase<llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >, mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::InsertIntoBucketImpl<mlir::Operation*>(mlir::Operation* const&, mlir::Operation* const&, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*>*) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:582:22
#13 0x00000000006f8813 llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*>* llvm::DenseMapBase<llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >, mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::InsertIntoBucket<mlir::Operation* const&>(llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*>*, mlir::Operation* const&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:547:37
#14 0x00000000006f7b0e llvm::DenseMapBase<llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >, mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::FindAndConstruct(mlir::Operation* const&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:326:44
#15 0x00000000006f6d6b llvm::DenseMapBase<llvm::DenseMap<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >, mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*, (anonymous namespace)::SimpleOperationInfo, llvm::detail::DenseMapPair<mlir::Operation*, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>*> >::operator[](mlir::Operation* const&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/DenseMap.h:330:34
#16 0x00000000006f7002 llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >::insertIntoScope(llvm::ScopedHashTableScope<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >*, mlir::Operation* const&, mlir::Operation* const&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/ScopedHashTable.h:218:54
#17 0x00000000006f6670 llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >::insert(mlir::Operation* const&, mlir::Operation* const&) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/ADT/ScopedHashTable.h:196:3
#18 0x00000000006f5cf3 (anonymous namespace)::CSE::simplifyOperation(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >&, mlir::Operation*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:135:18
#19 0x00000000006f5dd6 (anonymous namespace)::CSE::simplifyBlock(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm::MallocAllocator, 4096ul, 4096ul, 128ul>, llvm::ScopedHashTableVal<mlir::Operation*, mlir::Operation*>, 32ul, 8ul> >&, mlir::DominanceInfo&, mlir::Block*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/CSE.cpp:142:18
#20 0x00000000006f6009 (anonymous namespace)::CSE::simplifyRegion(llvm::ScopedHashTable<mlir::Operation*, mlir::Operation*, (anonymous namespace)::SimpleOperationInfo, llvm::RecyclingAllocator<llvm::BumpPtrAllocatorImpl<llvm: [... cut ...] firtool+0x40dd68)
#28 0x000000000040b203 main (../../llvm/circt/build/bin/firtool+0x40b203)
#29 0x00007f7fe56c1b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#30 0x000000000040b58e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: строка 26: 12056 Aborted                 (стек памяти сброшен на диск) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```

firrtl-1.5.0-SNAPSHOT produces Verilog

drom, 2021-03-24:
has unique signature: `Assertion `!FoundVal && "Key already in new map?"' failed.` but may be related to: [#813]

stephenneuendorffer, 2021-03-24:
Seems like something worth trying mlir-reduce on?

drom, 2021-03-24:
@stephenneuendorffer that would be great. That was the smallest source that my fuzzer came out with for the same crash reason.

lattner, 2021-03-24:
Yeah, this is a known to me issue.  I'll try to first principles a testcase.

lattner, 2021-03-24:
This crashes with `mlir-opt z.mlir -allow-unregistered-dialect -cse`:

```
func @f(%a: i8, %b : i8) {
   "some_graph_region" () ({
     "other_graph_region" () ({
       "use"(%x) : (i8) -> ()
     }) : () -> ()

     %x = std.addi %a, %b : i8
     %q = std.addi %y, %y : i8
     %y = std.addi %a, %b : i8

     "other_graph_region" () ({
       "use"(%q) : (i8) -> ()
     }) : () -> ()
 
    }) : () -> ()
}
```

I'll take a look.

lattner, 2021-03-24:
@youngar points out that this is a likely dupe of Issue #813

youngar, 2021-04-01:
Fixed by https://github.com/llvm/circt/pull/867

closed 2021-04-01 by youngar