llvm/circt #820: [comb] Still missing lots of constant folders

lattner, 2021-03-24:
For example, this should fold but doesn't yet:

```
    %c13274935_i30 = rtl.constant 13274935 : i30
    %c-411106406_i30 = rtl.constant -411106406 : i30
    %60 = comb.divu %c13274935_i30, %c-411106406_i30 : i30
```

Ayushsunny, 2021-03-26:
> For example, this should fold but doesn't yet:
> 
> ```
>     %c13274935_i30 = rtl.constant 13274935 : i30
>     %c-411106406_i30 = rtl.constant -411106406 : i30
>     %60 = comb.divu %c13274935_i30, %c-411106406_i30 : i30
> ```

I would like to work on this issue

lattner, 2021-03-26:
Cool, please do!

Ayushsunny, 2021-03-26:
> Cool, please do!

So, what exact I have to do @lattner sir ? as I am a new to open source I need some help before start contributing

Ayushsunny, 2021-04-02:
? @lattner

lattner, 2021-04-02:
Hi @Ayushsunny.  I'm sorry, but I don't have the ability to give you a full development tutorial here.  I'd recommend that you read some of the MLIR tutorials and look for other examples to learn from how they worked.

uenoku, 2023-03-02:
Closed by https://github.com/llvm/circt/commit/27c9e6c94d1a7887a8288b7a4a6823719d797215

closed 2023-03-02 by uenoku