llvm/circt #821: [FIRRTL] (neg) mismatch

drom, 2021-03-24:
The following FIRRTL program

```python
; seed: 9902
circuit top_mod :
  module top_mod :
    input inp_f: UInt<2>
    output tmp32: SInt<3>
    tmp32 <= neg(inp_f)
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
// Standard header to adapt well known macros to our needs.

module top_mod(
  input  [1:0] inp_f,
  output [2:0] tmp32);

  assign tmp32 = 3'h0 - {inp_f[1], inp_f};	// a_top_mod.fir:3:3, :6:14
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input  [1:0] inp_f,
  output [2:0] tmp32
);
  assign tmp32 = 2'h0 - inp_f;
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
yosys 0
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
yosys 1
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
yosys 2
ERROR: Found 1 unproven $equiv cells in 'equiv_status -assert'.
```

darthscsi, 2021-03-24:
in_f should not be sign extended, it should be zero extended.

prithayan, 2021-03-25:
@darthscsi , Chris made the exact opposite change to fix https://github.com/llvm/circt/issues/364
The comment from https://github.com/llvm/circt/blob/main/lib/Conversion/FIRRTLToRTL/LowerToRTL.cpp#L2122
```
FIRRTL negate always adds a bit, and always does a sign extension even if  the input is unsigned.
```
https://github.com/llvm/circt/commit/2fc584ef92bd0b41139ec486cfb3437dba362bd7

darthscsi, 2021-03-28:
Closed in #831

closed 2021-03-28 by darthscsi