llvm/circt #826: LowerTypes is growing and needs some factoring

darthscsi, 2021-03-25:
A couple high-level points:
* Struct lowering and vector lowering should be separate as we may want to do one and not the other.  Specifically, I imagine vector lowering won't be something that we enable normally.
* There is a difference between dynamic indexes and everything else as dynamic indexes cause mux trees at the access op, where-as static indexes and field accesses are trivial replacements of the op with the lowered values.

We've been talking about restructuring things so that a single layer of a type is peeled at a time.  We've debated either doing all the logic at point of complex type value creation (args, instances) or doing all the work at use-points (index ops).

This is a tracking issue so we don't lose this and we can get discussion on it.

mikeurbach, 2021-03-25:
I have been wondering if perhaps this pass has outgrown the current visitor pattern. It has always been designed to take a single pass through every module's block arguments, and then take a single pass through the ops in the module body.

I think the debate about where to do the work is the right one to be having. Without taking either side, here are some thoughts about the pass at a high-level.

We could continue using FIRRTLVisitor, but consider splitting up the processing into more modular steps, visiting things more than once, etc.

I have also wondered if perhaps FIRRTLVisitor just isn't the best tool for a job of this complexity. It maybe be worth switching to use the upstream pattern rewriting infrastructure instead of visitors. This opens the door to cost models and different pattern drivers.

For example, we could potentially drive this pass within the Dialect Conversion framework. I initially explored this, but stopped for two reasons: 1) it is more complex than using visitors, and I didn't see the benefit at the start, when I was just looking at static cases, and 2) a key part of the upstream block argument conversion only worked for FuncOp, not FunctionLike ops including FModuleOp.

I think on both fronts, there has been enough improvement to reconsider using RewritePatterns and Dialect Conversion. For 1) I recently used Dialect Conversion to drive the conversion of RTLModuleOps [here](https://github.com/llvm/circt/commit/905445248b12e7303aa007e2e10f15daf7936b54), and it really wasn't that bad. For 2), I added this [upstream](https://github.com/llvm/llvm-project/commit/0a7a1ac73d095eacd4499e889ce35191a9d1c648) a while back. Even if the upstream support doesn't work here, the example from RTLModuleOp shows what it looks like when you do it yourself.

Dialect Conversion might be one approach (mark either the sources of Values as illegal, or the uses of Values), but there are certainly others. If we can frame this pass as iteration to a fixed point, the upstream Greedy pattern driver can work. There is also the possibility to use a novel driver for this, but keep using RewritePatterns at the core.

Just some food for thought, if we are taking a step back.

darthscsi, 2021-03-26:
Discussion for #748