llvm/circt #832: [FIRRTL] vector (core dump)

drom, 2021-03-26:
The following FIRRTL program

```python
circuit top_mod :
 module top_mod :
   input addr: UInt<8>
   input wdata: UInt<8>
   output rdata: UInt<8>
   wire vec0: UInt<8>[4]
   vec0[addr] <= wdata
   rdata <= vec0[addr]
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
firtool
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
1.       #0 0x000000000043d1cd llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (../../llvm/circt/build/bin/firtool+0x43d1cd)
 #1 0x000000000043afd4 llvm::sys::RunSignalHandlers() (../../llvm/circt/build/bin/firtool+0x43afd4)
 #2 0x000000000043b133 SignalHandler(int) (../../llvm/circt/build/bin/firtool+0x43b133)
 #3 0x00007f765eb03a30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #4 0x00000000005644c6 mlir::Type::getDialect() const (../../llvm/circt/build/bin/firtool+0x5644c6)
 #5 0x00000000005890a9 circt::firrtl::__mlir_ods_local_type_constraint_FIRRTL12(mlir::Operation*, mlir::Type, llvm::StringRef, unsigned int) /home/drom/work/github/llvm/circt/build/include/circt/Dialect/FIRRTL/FIRRTL.cpp.inc:143:3
 #6 0x00000000005a14ec circt::firrtl::SubaccessOp::verify() /home/drom/work/github/llvm/circt/build/include/circt/Dialect/FIRRTL/FIRRTL.cpp.inc:12308:7
 #7 0x000000000057599f mlir::Op<circt::firrtl::SubaccessOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::FIRRTLType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::NOperands<2u>::Impl, mlir::MemoryEffectOpInterface::Trait>::verifyInvariants(mlir::Operation*) /home/drom/work/github/llvm/circt/build/../llvm/mlir/include/mlir/IR/OpDefinition.h:1713:3
 #8 0x0000000000566d74 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) (../../llvm/circt/build/bin/firtool+0x566d74)
 #9 0x00000000005671f8 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) (../../llvm/circt/build/bin/firtool+0x5671f8)
#10 0x00000000005671f8 (anonymous namespace)::OperationVerifier::verifyOperation(mlir::Operation&) (../../llvm/circt/build/bin/firtool+0x5671f8)
#11 0x0000000000568387 mlir::verify(mlir::Operation*) (../../llvm/circt/build/bin/firtool+0x568387)
#12 0x00000000006a7eea mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x6a7eea)
#13 0x00000000006a80c0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x6a80c0)
#14 0x00000000006a74e2 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (../../llvm/circt/build/bin/firtool+0x6a74e2)
#15 0x00000000006a7d26 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (../../llvm/circt/build/bin/firtool+0x6a7d26)
#16 0x00000000006a80c0 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (../../llvm/circt/build/bin/firtool+0x6a80c0)
#17 0x00000000006ab144 mlir::PassManager::run(mlir::Operation*) (../../llvm/circt/build/bin/firtool+0x6ab144)
#18 0x000000000040c10c processBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer> >, llvm::StringRef, std::function<mlir::LogicalResult (mlir::OwningModuleRef)>) /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:210:3
#19 0x00000000004084ba processBufferIntoSingleStream /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:235:8
#20 0x00000000004084ba main /home/drom/work/github/llvm/circt/build/../tools/firtool/firtool.cpp:304:15
#21 0x00007f765e571b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#22 0x0000000000409cee _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: line 26:  7889 Segmentation fault      (core dumped) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```
Expected behavior: not to crash

darthscsi, 2021-03-26:
SubAcccessOp is missing from Lower types.  #746

darthscsi, 2021-03-28:
array index as a LHS is not low firrtl, changing milestone.

drom, 2021-03-28:
should it at lest not crash as part of milestone 1?

darthscsi, 2021-03-28:
> should it at least not crash as part of milestone 1?

yes, core dumps are bad.  I'll split this up tomorrow.

darthscsi, 2021-03-29:
Specifically, this is a lower-types issue, lower-to-rtl errors out gracefully.

commit 6662740a (2021-03-29): Fail gracefully on subaccessop in lowertypes.  Closes #832
closed 2021-03-29 by darthscsi