llvm/circt #833: [FIRRTL] Canonicalize and CSE don't have to run on the global ModuleOp

youngar, 2021-03-26:
The canonicalize and CSE passes are currently set to run on the ModuleOp.  I think they can effectively run on any IsolatedFromAbove operation.  To increase parallelism they should run on the the CircuitOp::FModule.  I don't think we can do the same thing for RTLModule because of the SV operations inserted into the global module scope.

lattner, 2021-03-26:
aha!  Great catch!  This should just require adding some `nest` things to firtool.cpp?

youngar, 2021-03-26:
Yeah, fix here: https://github.com/llvm/circt/pull/834.

Is there actually anything emitted in to the global scope of RTL that we need to run CSE+Canon on?  If not, we could make the same change to the RTL optimization phase.  This would most likely cause confusions down the line though.

lattner, 2021-03-26:
> Is there actually anything emitted in to the global scope of RTL that we need to run CSE+Canon on? 

No there isn't, everything should be in an rtl.module

closed 2021-03-30 by youngar