llvm/circt #838: wire generated in always block

darthscsi, 2021-03-29:
Probably long name spilling.

```
circuit testcase :
  module testcase :
    input clock : Clock
    input reset : UInt<1>
    input io_enq_bits_prediction_branch_base_prediction : UInt<1>
    input io_deq_ready : UInt<1>
    output io_deq_bits_prediction_branch_prediction : UInt<1>
    output io_count : UInt<1>

    mem ram_prediction_branch_base_prediction : @[Decoupled.scala 218:16]
      data-type => UInt<1>
      depth => 1
      read-latency => 0
      write-latency => 1
      reader => io_deq_bits_MPORT
      writer => MPORT
      read-under-write => undefined
```

seldridge, 2021-03-29:
A simplified MLIR example is:

```mlir
module attributes {firrtl.mainModule = "Foo"}  {
  rtl.module @Foo(%clock: i1, %in: i1) {
    %addr__________________________________________________ = sv.wire  : !rtl.inout<i1>
    %0 = sv.read_inout %addr__________________________________________________ : !rtl.inout<i1>
    %r = sv.reg  : !rtl.inout<uarray<1xi1>>
    %1 = sv.array_index_inout %r[%0] : !rtl.inout<uarray<1xi1>>, i1
    sv.alwaysff(posedge %clock)  {
        sv.passign %1, %in : i1
    }
  }
}
```

The above produces:

```verilog
  always_ff @(posedge clock) begin      // 838/Foo.mlir:7:5
    wire _tmp = addr__________________________________________________; // 838/Foo.mlir:4:10
    r[_tmp] <= in;      // 838/Foo.mlir:6:10, :8:9
  end // always_ff @(posedge)
```

The underlying problem is that the statement where something is going to be inlined needs to be examined to see if it's in a procedural region or not. The current line "line too long, dumping a temporary" is using the to-be-inlined expression's defining op to make this determination now. When this crosses a procedural boundary this bug shows up.

I think I can fix this specific case by including the statement in the expression emitter and then querying that when doing naming.

Longer term, I'd be curious about making the Verilog emitter do less and try to do some of this inlining preparation earlier where verifiers can kick in. E.g., expression inlining could be a pass that is moving `%addr___` into the region of the `sv.alwaysff`.

seldridge, 2021-03-30:
Note: there is a second issue here which can lead to double wires being emitted. 

A minimal example is here:

```mlir
module attributes {firrtl.mainModule = "Bar"}  {
  rtl.module @Bar(%clock: i1, %x: i1) {
    %0 = comb.and %1, %x : i1
    sv.alwaysff(posedge %clock)  {
      sv.if %0  {
        sv.verbatim "// hello"
      }
    }
    %aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa = sv.wire  : !rtl.inout<i1>
    %1 = sv.read_inout %aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa : !rtl.inout<i1>
  }
}
```

This produces the following (see double `wire _tmp`):

```verilog
module Bar(
  input clock, x);

  wire aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;	// 838/Bar.mlir:9:59

  wire _tmp = aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;	// 838/Bar.mlir:10:10
  wire _T = _tmp & x;	// 838/Bar.mlir:3:10
  always_ff @(posedge clock) begin	// 838/Bar.mlir:4:5
    if (_T) begin	// 838/Bar.mlir:5:7
      // hello	// 838/Bar.mlir:6:9
    end
  end // always_ff @(posedge)
  wire _tmp = aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa;	// 838/Bar.mlir:10:10
endmodule
```

seldridge, 2021-03-30:
I have fixes ready to go in #838. The first for the first circuit is in 68bad26a92115dcfdf39e32049a0ac52ce298c99. The second fix is in 07d0d8b2f80470ece54d8968bd55dce430828052. Slightly modified versions of the testcases shown here are included in that PR.

lattner, 2021-03-30:
Hey @seldridge, I think it is simpler & better to fix this the longer term way, through two changes:

1) Not inlining across operations across block uses.  We should have a prepass that sinks these users into the blocks anyway.

2) Duplicate and move the "must be inlined" ops as part of a prepass as you say.  

The end result of this will keep the verilog emitter simpler and more stable.  I'm going to work on some patches.

lattner, 2021-03-30:
Yeah, we're already doing #1.  The problem is the "must be inlined" ops like array index.  That should be a straight-forward fix.

seldridge, 2021-03-30:
Cool. Thanks for taking a look! I'll let you hack. If #846 is bound for closure, do use the test cases.

I agree that the longer term solution is the way we need to move. 

I'm for keeping the Verilog emitter as simple as possible. Ideally, as close to `Operation => String` as possible. Anything which is effectively an IR transformation (like inlining operations across blocks due to temporary spillage) is best to be avoided.

lattner, 2021-03-30:
Agreed, thanks!  I appreciate the testcases!

lattner, 2021-03-30:
ccfee708 is progress towards fixing this.

lattner, 2021-03-31:
Second issue fixed in a515f213, thanks!

closed 2021-03-31 by lattner

seldridge, 2021-03-31:
Awesome. Thanks, @lattner!