llvm/circt #839: multi-port memories can cause writes in different always blocks

darthscsi, 2021-03-29:
due to the clocks being assigned to wires and the wires not deliminated, the always blocks aren't merged.

darthscsi, 2021-03-29:
This should be a sufficient test case:
```
circuit top_mod :
 module top_mod :
   output foo : UInt<64>
   output bar : UInt<64>
   
   mem xrf : @[Core.scala 704:16]
     data-type => UInt<64>
     depth => 32
     read-latency => 0
     write-latency => 1
     reader => id_regs_MPORT
     reader => id_regs_MPORT_1
     reader => id_regs_MPORT_2
     reader => id_regs_MPORT_3
     writer => MPORT
     writer => MPORT_1
     writer => MPORT_2
     read-under-write => undefined
```

fabianschuiki, 2021-03-30:
Couldn't this actually be correct behaviour? The read/write ports have their own clock, and without hooking them up, these are just dangling wires. Connecting up the clocks of these ports produces nicely merged blocks:
```firrtl
; foo.fir
circuit top_mod :
  module top_mod :
    input clock : Clock

    mem xrf : @[Core.scala 704:16]
      data-type => UInt<64>
      depth => 32
      read-latency => 0
      write-latency => 1
      writer => write_0
      writer => write_1
      read-under-write => undefined

    xrf.write_0.clk <= clock
    xrf.write_1.clk <= clock
```
Compiled with `firtool foo.fir -enable-lower-types -lower-to-rtl -verilog` produces:
```systemverilog
module top_mod(
  input clock);

  reg  [63:0] xrf[0:31];	// Core.scala:704:16
  wire [63:0] xrf_write_0_data;	// Core.scala:704:16
  wire        xrf_write_0_mask;	// Core.scala:704:16
  wire [63:0] xrf_write_1_data;	// Core.scala:704:16
  wire        xrf_write_1_mask;	// Core.scala:704:16
  wire [4:0]  xrf_write_0_addr;	// Core.scala:704:16
  wire        xrf_write_0_en;	// Core.scala:704:16
  wire [4:0]  xrf_write_1_addr;	// Core.scala:704:16
  wire        xrf_write_1_en;	// Core.scala:704:16

  // [...]

  always_ff @(posedge clock) begin	// Core.scala:704:16
    if (_T_0)	// Core.scala:704:16
      xrf[xrf_write_0_addr] <= _T;	// Core.scala:704:16
    if (_T_2)	// Core.scala:704:16
      xrf[xrf_write_1_addr] <= _T_1;	// Core.scala:704:16
  end // always_ff @(posedge)

  // [...]
endmodule
```
What am I missing?

seldridge, 2021-03-30:
Good catch @fabianschuiki. This is the same thing as https://github.com/llvm/circt/issues/794, but with an undriven memory. (I expect this wasn't what @darthscsi intended...)

If the memory is dead then we should remove it. However, that's the recurring oddity of requiring optimizations for correctness (or lint correctness). This could be entirely avoided with a check that memories are driven (part of the general fix for https://github.com/llvm/circt/issues/564).

darthscsi, 2021-03-30:
yea, my reduction doesn't have the driving clock.  The case I was looking at was multiple write ports with the same clock wound up spilling the clock to different temporaries for each use, thus to different always blocks.

darthscsi, 2021-03-30:
The failure came up because the array representing the memory was written in two different always blocks (for the reasons above).

fabianschuiki, 2021-03-30:
Do you still have a concrete test case where the clocks fail to be merged?

fabianschuiki, 2021-03-31:
Dug through this a bit more and found the following failing example:
```
circuit top_mod :
  module top_mod :
    input clock : Clock
    input valid : UInt<1>

    mem xrf : @[Core.scala 704:16]
      data-type => UInt<64>
      depth => 32
      read-latency => 0
      write-latency => 1
      writer => write_0
      writer => write_1
      read-under-write => undefined

    xrf.write_0.clk <= validif(valid, clock)
    xrf.write_1.clk <= validif(valid, clock)
```
Compiled with `firtool foo.fir -enable-lower-types -lower-to-rtl -verilog` produces the following (rearranged for clarity):
```systemverilog
module top_mod(
  input clock, valid);

  reg  [63:0] xrf[0:31];	// Core.scala:704:16
  wire [63:0] xrf_write_0_data;	// Core.scala:704:16
  wire        xrf_write_0_mask;	// Core.scala:704:16
  wire [63:0] xrf_write_1_data;	// Core.scala:704:16
  wire        xrf_write_1_mask;	// Core.scala:704:16
  wire [4:0]  xrf_write_0_addr;	// Core.scala:704:16
  wire        xrf_write_0_en;	// Core.scala:704:16
  wire        xrf_write_0_clk;	// Core.scala:704:16
  wire        _T_3;	// Core.scala:704:16
  wire [4:0]  xrf_write_1_addr;	// Core.scala:704:16
  wire        xrf_write_1_en;	// Core.scala:704:16
  wire        xrf_write_1_clk;	// Core.scala:704:16
  wire        _T_4;	// Core.scala:704:16

  wire [63:0] _T = xrf_write_0_data;	// Core.scala:704:16
  wire _T_0 = xrf_write_0_en & xrf_write_0_mask;	// Core.scala:704:16
  wire [63:0] _T_1 = xrf_write_1_data;	// Core.scala:704:16
  wire _T_2 = xrf_write_1_en & xrf_write_1_mask;	// Core.scala:704:16

  assign _T_3 = xrf_write_0_clk;	// Core.scala:704:16
  assign _T_4 = xrf_write_1_clk;	// Core.scala:704:16
  assign xrf_write_0_clk = clock;	// minimal.fir:15:21
  assign xrf_write_1_clk = clock;	// minimal.fir:16:21

  always_ff @(posedge _T_3) begin	// Core.scala:704:16
    if (_T_0)	// Core.scala:704:16
      xrf[xrf_write_0_addr] <= _T;	// Core.scala:704:16
  end // always_ff @(posedge)

  always_ff @(posedge _T_4) begin	// Core.scala:704:16
    if (_T_2)	// Core.scala:704:16
      xrf[xrf_write_1_addr] <= _T_1;	// Core.scala:704:16
  end // always_ff @(posedge)
endmodule
```
The additional `validif` (which are present in the full failing test case as well) seems to be ignored in ExportVerilog for emission, but not for collapsing signals together.

Out of curiosity: Do we know why there are `validif`s in the clock paths to memories? That seems like asking for trouble.

fabianschuiki, 2021-03-31:
This can actually be triggered without any memories involved. Consider the following two modules:
```
circuit A :
  module A :
    input clock : Clock
    input valid : UInt<1>
    output out0 : Clock
    output out1 : Clock

    wire x : Clock
    wire y : Clock
    x <= validif(valid, clock)  // <--- here
    y <= validif(valid, clock)  // <--- here
    out0 <= x
    out1 <= y

  module B :
    input clock : Clock
    input valid : UInt<1>
    output out0 : Clock
    output out1 : Clock

    wire x : Clock
    wire y : Clock
    x <= clock  // <--- here
    y <= clock  // <--- here
    out0 <= x
    out1 <= y
```
Compiling this with `firtool -lower-to-rtl -enable-lower-types` produces:
```
rtl.module @A(%clock: i1, %valid: i1) -> (%out0: i1, %out1: i1) {
  %x = sv.wire  : !rtl.inout<i1>
  %y = sv.wire  : !rtl.inout<i1>
  sv.connect %x, %clock : i1
  sv.connect %y, %clock : i1
  %0 = sv.read_inout %x : !rtl.inout<i1>
  %1 = sv.read_inout %y : !rtl.inout<i1>
  rtl.output %0, %1 : i1, i1
}

rtl.module @B(%clock: i1, %valid: i1) -> (%out0: i1, %out1: i1) {
  rtl.output %clock, %clock : i1, i1
}
```
Notice how for `A` (with `validif`) we spilled into a separate wire, but for `B` (without `validif`) we directly folded everything to `%clock`. Missing constant folding case?

seldridge, 2021-03-31:
Great find @fabianschuiki. The SFC has an explicit pass for removing `validif` which converts circuit `A` to:

```scala
circuit A :
  module A :
    input clock : Clock
    input valid : UInt<1>
    output out0 : Clock
    output out1 : Clock

    node x = clock
    node y = clock
    out0 <= x
    out1 <= y
```

This the MFC will lower in the same way as circuit `B`.

It seems like the immediate fix is to add handling of `validif` during `LowerToRTL`.

That said, that circuit you show is a little concerning:

```mlir
rtl.module @A(%clock: i1, %valid: i1) -> (%out0: i1, %out1: i1) {
  %x = sv.wire  : !rtl.inout<i1>
  %y = sv.wire  : !rtl.inout<i1>
  sv.connect %x, %clock : i1
  sv.connect %y, %clock : i1
  %0 = sv.read_inout %x : !rtl.inout<i1>
  %1 = sv.read_inout %y : !rtl.inout<i1>
  rtl.output %0, %1 : i1, i1
}
```

This RTL dialect code is perfectly fine, but this isn't copy propagated to produce what we'd expect. Maybe we've been going about this the wrong with by packing copy propagation-like stuff into FIRRTL -> RTL path and need actual RTL/SV dialect copy propagation as an optimization?

fabianschuiki, 2021-03-31:
In the `FIRRTLFold.cpp`, the `foldSingleSetConnect` function tries to push values through wires and registers. This happens for simple things like constants or ports, but somehow `validif` does not seem to be propagated down. That then trips up the RTL/SV part which explicitly emits the wire and connects.

I agree @seldridge that it could make sense to do more work specifically in the RTL and SV dialects. As @darthscsi also noted, having SV rely on certain folds to happen one or two dialect levels up can make stuff like the memory emission very brittle.

fabianschuiki, 2021-03-31:
The culprit seems to be here:
https://github.com/llvm/circt/blob/74f99727c6ca9ec353a0a54a68d45f6e7fce49de/lib/Dialect/FIRRTL/FIRRTLFolds.cpp#L875-L890

Ports (i.e. just `%clock`) take the then-path through the if statement (`!srcValueOp`) and are appropriately forwarded through wires. Things like `validif` take the else-path and get stopped at the check for `ConstantOp` or `InvalidValuePrimOp`. That prevents the operation from being pushed further down. This blocks propagating the clock input port to the clock port of the memory directly, which then causes the RTL lowering to not detect that this is actually a clock.

One solution would be to just add a folder to the `validif` op for the time being which simply removes it? I didn't yet see the op used in any way so far.

PR #881 (2021-04-03): [FIRRTL] Fold away validif operations
Add a trivial folder for `firrtl.validif` operations that will always replace the operation by its second operand, which is the value being gated. This is not perfect but replicates the behaviour of the Scala FIRRTL compiler in all cases that we are interested in for now. Also fixes #839 and makes reasoning about the clock signal easier in some places. We'll want to revisit this at a later stage.
commit e57dd52c (2021-04-03): [FIRRTL] Fold away validif operations (#881)

* Add a trivial folder for `firrtl.validif` operations that will always
  replace the operation by its second operand, which is the value being
  gated. This is not perfect but replicates the behaviour of the Scala
  FIRRTL compiler in all cases that we are interested in for now. Also
  fixes #839 and makes reasoning about the clock signal easier in some
  places. We'll want to revisit this at a later stage.
closed 2021-04-03 by fabianschuiki