llvm/circt #844: [FIRRTL] mem (mismatch)

drom, 2021-03-29:
The following FIRRTL program

```python
; seed: 56
circuit top_mod :
  module top_mod :
    input clock: Clock
    input en: UInt<1>
    input addr1: UInt<17>
    input wdata: SInt<19>
    output tmp16: SInt<10>
    mem tmp14:
      data-type => SInt<10>
      depth => 16
      read-latency => 1
      write-latency => 1
      read-under-write => undefined
      reader => r1
      writer => w0
    tmp14.r1.clk <= clock
    tmp14.r1.en <= en
    tmp14.r1.addr <= addr1
    tmp14.w0.clk <= clock
    tmp14.w0.en <= en
    tmp14.w0.addr <= asUInt(SInt<7>("h-28"))
    tmp14.w0.data <= wdata
    tmp14.w0.mask <= UInt<25>("hffffff")
    tmp16 <= tmp14.r1.data
```

Compiled with `firtool --lower-to-rtl --enable-lower-types` produces this Verilog:

```verilog
...
module top_mod(
  input         clock, en,
  input  [16:0] addr1,
  input  [18:0] wdata,
  output [9:0]  tmp16);

  reg  [9:0] tmp14[0:15];	// a_top_mod.fir:9:5
  reg        tmp14_r1_en_pipe[0:0];	// a_top_mod.fir:9:5
  reg  [3:0] tmp14_r1_addr_pipe[0:0];	// a_top_mod.fir:9:5
  wire [3:0] tmp14_r1_addr;	// a_top_mod.fir:9:5
  wire [3:0] _T;	// a_top_mod.fir:9:5
  wire [9:0] _T_0;	// a_top_mod.fir:23:19

  always_ff @(posedge clock) begin	// a_top_mod.fir:9:5
    tmp14_r1_en_pipe[1'h0] <= en;	// a_top_mod.fir:9:5
    if (en)	// a_top_mod.fir:9:5
      tmp14_r1_addr_pipe[1'h0] <= _T;	// a_top_mod.fir:9:5
    if (en)	// a_top_mod.fir:9:5
      tmp14[4'h8] <= _T_0;	// a_top_mod.fir:9:5, :22:19
  end // always_ff @(posedge)
...
  assign _T = tmp14_r1_addr;	// a_top_mod.fir:9:5
  assign tmp14_r1_addr = addr1[3:0];	// a_top_mod.fir:19:19
  assign _T_0 = wdata[9:0];	// a_top_mod.fir:23:19
  assign tmp16 = tmp14[tmp14_r1_addr_pipe[1'h0]];	// a_top_mod.fir:3:3, :9:5
endmodule
```

Compiled with `firrtl-1.5-SNAPSHOT --dont-fold div` produces this Verilog:

```verilog
module top_mod(
  input         clock,
  input         en,
  input  [16:0] addr1,
  input  [18:0] wdata,
  output [9:0]  tmp16
);
...
  reg [9:0] tmp14 [0:15];
  wire [9:0] tmp14_r1_data;
  wire [3:0] tmp14_r1_addr;
  wire [9:0] tmp14_w0_data;
  wire [3:0] tmp14_w0_addr;
  wire  tmp14_w0_mask;
  wire  tmp14_w0_en;
  reg  tmp14_r1_en_pipe_0;
  reg [3:0] tmp14_r1_addr_pipe_0;
  assign tmp14_r1_addr = tmp14_r1_addr_pipe_0;
  assign tmp14_r1_data = tmp14[tmp14_r1_addr];
  assign tmp14_w0_data = wdata[9:0];
  assign tmp14_w0_addr = 4'h8;
  assign tmp14_w0_mask = 1'h1;
  assign tmp14_w0_en = en;
  assign tmp16 = tmp14_r1_data;
  always @(posedge clock) begin
    if (tmp14_w0_en & tmp14_w0_mask) begin
      tmp14[tmp14_w0_addr] <= tmp14_w0_data;
    end
    tmp14_r1_en_pipe_0 <= en;
    if (en) begin
      tmp14_r1_addr_pipe_0 <= addr1[3:0];
    end
  end
// Register and memory initialization
...
endmodule
```

`Yosys 0.9+3911 (git sha1 4b31223e, clang 11.0.1 -fPIC -Os)` reports formal mismatch:

```
yosys 0
Warning: Replacing memory \tmp14 with list of registers. See a_top_mod_new.sv:53
Warning: Replacing memory \tmp14_r1_addr_pipe with list of registers. See a_top_mod_new.sv:51
Warning: Replacing memory \tmp14_r1_en_pipe with list of registers. See a_top_mod_new.sv:49
ERROR: Found 10 unproven $equiv cells in 'equiv_status -assert'.
yosys 1
Warning: Replacing memory \tmp14 with list of registers. See a_top_mod_new.sv:53
Warning: Replacing memory \tmp14_r1_addr_pipe with list of registers. See a_top_mod_new.sv:51
Warning: Replacing memory \tmp14_r1_en_pipe with list of registers. See a_top_mod_new.sv:49
ERROR: Found 10 unproven $equiv cells in 'equiv_status -assert'.
yosys 2
Warning: Replacing memory \tmp14 with list of registers. See a_top_mod_new.sv:53
Warning: Replacing memory \tmp14_r1_addr_pipe with list of registers. See a_top_mod_new.sv:51
Warning: Replacing memory \tmp14_r1_en_pipe with list of registers. See a_top_mod_new.sv:49
ERROR: Found 10 unproven $equiv cells in 'equiv_status -assert'.
```

seldridge, 2021-03-29:
By inspection, there's some more aggressive constant propagation in `firtool` than in `firrtl` here. Also, there are some changes to the representation of the enable pipe.

Do you know if either of these could be causing the mismatch?

fabianschuiki, 2021-03-30:
> There are some changes to the representation of the enable pipe.

This could be the culprit. As far as I remember, registers are used mostly as fixed points for formal equivalence checking (like additional inputs and outputs). I think the tools make some attempt to match up registers and wires by name on a best-effort basis, but it could be that the `tmp14_r1_addr_pipe_0` vs `tmp14_r1_en_pipe[0:0]` here throws the tool off.

darthscsi, 2021-04-07:
The causes of this will be moved with #864 but we will need to reverify after that is in.

closed 2021-04-15 by darthscsi