llvm/circt #847: [FIRRTL] (geq) (core dump)

drom, 2021-03-30:
The following FIRRTL program

```python
; seed: 12
circuit top_mod :
  module top_mod :
    input clock: Clock
    input reset: UInt<1>
    input inp_h: UInt<0>
    wire tmp10: SInt<0>
    wire tmp25: SInt<29>
    wire tmp28: UInt<1>
    tmp10 <= asSInt(inp_h)
    tmp25 <= SInt<29>(-1)
    tmp28 <= geq(tmp10, tmp25)
```

Compiled with `firtool --lower-to-rtl --enable-lower-types --verilog` gives an error:

```mlir
firtool
firtool: ../llvm/llvm/include/llvm/ADT/APInt.h:280: llvm::APInt::APInt(unsigned int, uint64_t, bool): Assertion `BitWidth && "bitwidth too small"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=a_top_mod_new.sv
1.      2.       #0 0x00000000004be4f7 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00000000004be5ae PrintStackTraceSignalHandler(void*) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00000000004bc5a4 llvm::sys::RunSignalHandlers() /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Signals.cpp:71:20
 #3 0x00000000004bdf4a SignalHandler(int) /home/drom/work/github/llvm/circt/llvm/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007f8b67bbaa30 __restore_rt (/lib64/libpthread.so.0+0x13a30)
 #5 0x00007f8b6763e4a5 raise (/lib64/libc.so.6+0x3d4a5)
 #6 0x00007f8b67627864 abort (/lib64/libc.so.6+0x26864)
 #7 0x00007f8b67627749 _nl_load_domain.cold (/lib64/libc.so.6+0x26749)
 #8 0x00007f8b676369b6 (/lib64/libc.so.6+0x359b6)
 #9 0x0000000000727d4b circt::firrtl::GEQPrimOp::fold(llvm::ArrayRef<mlir::Attribute>) (../../llvm/circt/build/bin/firtool+0x727d4b)
#10 0x000000000071cda4 mlir::LogicalResult mlir::Op<circt::firrtl::GEQPrimOp, mlir::OpTrait::ZeroRegion, mlir::OpTrait::OneResult, mlir::OpTrait::OneTypedResult<circt::firrtl::UIntType>::Impl, mlir::OpTrait::ZeroSuccessor, mlir::OpTrait::NOperands<2u>::Impl, mlir::MemoryEffectOpInterface::Trait>::foldSingleResultHook<circt::firrtl::GEQPrimOp>(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) (../../llvm/circt/build/bin/firtool+0x71cda4)
#11 0x00000000006bbdb6 mlir::AbstractOperation::foldHook(mlir::Operation*, llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) const /home/drom/work/github/llvm/circt/llvm/mlir/include/mlir/IR/OperationSupport.h:128:3
#12 0x00000000006b4f12 mlir::Operation::fold(llvm::ArrayRef<mlir::Attribute>, llvm::SmallVectorImpl<mlir::OpFoldResult>&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/IR/Operation.cpp:553:30
#13 0x000000000078be1f mlir::OperationFolder::tryToFold(mlir::OpBuilder&, mlir::Operation*, llvm::SmallVectorImpl<mlir::Value>&, llvm::function_ref<void (mlir::Operation*)>) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:198:13
#14 0x000000000078b7a4 mlir::OperationFolder::tryToFold(mlir::Operation*, llvm::function_ref<void (mlir::Operation*)>, llvm::function_ref<void (mlir::Operation*)>, bool*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/FoldUtils.cpp:101:13
#15 0x00000000007718cb (anonymous namespace)::GreedyPatternRewriteDriver::simplify(llvm::MutableArrayRef<mlir::Region>, int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:192:21
#16 0x0000000000771bee mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:256:35
#17 0x0000000000771ada mlir::applyPatternsAndFoldGreedily(llvm::MutableArrayRef<mlir::Region>, mlir::FrozenRewritePatternSet const&) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Utils/GreedyPatternRewriteDriver.cpp:236:1
#18 0x0000000000703ca4 (anonymous namespace)::Canonicalizer::runOnOperation() /home/drom/work/github/llvm/circt/llvm/mlir/lib/Transforms/Canonicalizer.cpp:36:3
#19 0x00000000008ed8bb mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:407:36
#20 0x00000000008edae7 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:448:15
#21 0x00000000008ee7c2 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)::operator()(llvm::MutableArrayRef<mlir::OpPassManager>) const /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:624:68
#22 0x00000000008f1a0b mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>) std::for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)) /usr/include/c++/10/bits/stl_algo.h:3838:7
#23 0x00000000008f19ac void llvm::parallel::detail::parallel_for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/Support/Parallel.h:143:16
#24 0x00000000008f0ed3 void llvm::parallelForEach<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)) /home/drom/work/github/llvm/circt/llvm/llvm/include/llvm/Support/Parallel.h:233:40
#25 0x00000000008eecd7 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:601:24
#26 0x00000000008ee203 mlir::detail::OpToOpPassAdaptor::runOnOperation(bool) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:526:1
#27 0x00000000008ed89c mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:404:28
#28 0x00000000008edae7 mlir::detail::OpToOpPassAdaptor::runPipeline(llvm::iterator_range<llvm::pointee_iterator<std::unique_ptr<mlir::Pass, std::default_delete<mlir::Pass> >*, mlir::Pass> >, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:448:15
#29 0x00000000008ee7c2 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)::operator()(llvm::MutableArrayRef<mlir::OpPassManager>) const /home/drom/work/github/llvm/circt/llvm/mlir/lib/Pass/Pass.cpp:624:68
#30 0x00000000008f1a0b mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>) std::for_each<llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)>(llvm::SmallVector<mlir::OpPassManager, 1u>*, llvm::SmallVector<mlir::OpPassManager, 1u>*, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::'lambda'(llvm::MutableArrayRef<mlir::OpPassManager>)) /usr/include/c++/10/bits/stl_algo.h:3838:7
[... cut ...] ild/bin/firtool+0x40e193)
#39 0x000000000040b3d1 main (../../llvm/circt/build/bin/firtool+0x40b3d1)
#40 0x00007f8b67628b25 __libc_start_main (/lib64/libc.so.6+0x27b25)
#41 0x000000000040b76e _start /home/abuild/rpmbuild/BUILD/glibc-2.33/csu/../sysdeps/x86_64/start.S:122:0
./process1.sh: строка 26: 13260 Aborted                 (стек памяти сброшен на диск) ../../llvm/circt/build/bin/firtool a_top_mod.fir --lower-to-rtl --enable-lower-types --verilog -o=$VFILE2
```
Expected behavior: not to crash

fabianschuiki, 2021-03-30:
This seems to crash in `circt::firrtl::GEQPrimOp::fold`, which I recently added as part of constant folding. Will take a look.

fabianschuiki, 2021-03-30:
0-width types cause trouble here.

fabianschuiki, 2021-03-30:
Pushed a fix in #850.

commit ce929f78 (2021-03-30): [FIRRTL] Handle folding of 0-width LEQ/LT/GEQ/GT (#850)

* Fix an issue in comparison folding in case the left-hand argument was
  a non-constant value of width 0. Fixes #847.

* Handle additional constant folding cases of 0-width values for
  comparison operation. The Scala FIRRTL compiler generally treats such
  values as a constant 0 for the sake of constant folding. This extends
  the folders for LEQ/LT/GEQ/GT/EQ/NEQ to replicate this behaviour.

* Add additional test cases for the 0-width constant foldings.