llvm/circt #856: [FIRRTL] Name strings shouldn't be optional

lattner, 2021-03-31:
Per [this change to sv dialect](https://github.com/llvm/circt/commit/1955aa3b90e9b88894efa66add2dde81dd22c185), we should also require names on the firrtl.mem, firrtl.reg etc operations.  Missing a name can be represented with an empty string attribute instead of a `None`.

In fact we should look at each of the uses of `OptionalAttr` to see how many we can eliminate

seldridge, 2021-03-31:
Because FIRRTL requires things to have names, I think this is great. :100:

I'm generally not a huge fan of use of an empty string to represent `None` (if you really mean "none"). However, I don't think this matters. FIRRTL requires names and those names should verify against FIRRTL naming rules (roughly Verilog's `[a-zA-Z_][a-zA-Z0-9_$]*`). This would then make the empty string invalid.

youngar, 2021-04-18:
I created a PR #928 to make the string attributes non-optional, but it still allows empty names. If we keep allowing empty names, a follow up would be to not print `{name = ""}` in the same way we are ellidiing the annotation attribute.

lattner, 2021-04-18:
My understanding is that empty names are verboten according to chisel/sfc.  If so, we should reject them in the verifier.  @seldridge is this true?

seldridge, 2021-04-18:
Yes, we should reject these.

Most SFC IR nodes all have a name. We can double check this by looking for anything which mixes in the `HasName` trait either [in the source](https://github.com/chipsalliance/firrtl/blob/master/src/main/scala/firrtl/ir/IR.scala) or by looking at ["known subclasses" of `HasName` in the ScalaDoc](https://www.chisel-lang.org/api/firrtl/latest/firrtl/ir/HasName.html). Verification statements are the only things that have a truly "optional" name in unreleased FIRRTL 1.5.

tl;dr: instance, module, memory, node, register, wire, external module, port are what I'm aware of and we should verify.

youngar, 2021-04-18:
This means we should carry the temporary wire names like `_T10` to the final verilog output and we should remove the function `filterUselessNames()` from the parser, is that right?

seldridge, 2021-04-18:
If something is an actual node, then it should have a name. However, true temporaries (name begins with `_T` and un-annotated) _may_ be okay to promote to nameless SSA values.

lattner, 2021-04-18:
Oh good point.  I don’t think there is any point at all in carrying around these pointless names - they just waste memory.  I think that representing them with an empty string is a good way to go.  This should get mentioned in the rationale though

youngar, 2021-04-18:
This is what i'm thinking right now, but I have no idea what to do for wires and registers.  If wires don't get generated with "_T" names, then we can just require names.
```
instances -> required
memories  -> required
nodes     -> _T nixed in the parser, otherwise required
wires     -> not required
registers -> not required
```

seldridge, 2021-04-18:
One caveat, if the Chisel naming plugin is used (it's supposed to be), then Chisel hardly ever generates a name with `_T_<int>`. All nodes wind up with a prefix related to a local "good" name that is intentionally constructed to be both semantically meaningful and to avoid large diffs in the output Verilog. (E.g., adding a node doesn't cause every `_T_<int>` to bump.

E.g., 

```scala
val a = RegNext(b)
```

Becomes:

```scala
reg b_REG : UInt<1>, clock with :
      reset => (UInt<1>("h0"), b_REG) @[main.scala 32:15]
```

Actual node temporaries will then have good names, e.g.:

```scala
  val x = a + 1.U
  val y = x + 1.U

  b := y
```

Will become:

```scala
    node _b_T = add(a, UInt<1>("h1")) @[main.scala 11:10]
    node _b_T_1 = tail(_b_T, 1) @[main.scala 11:10]
    node _b_T_2 = add(_b_T_1, UInt<1>("h1")) @[main.scala 11:16]
    node _b_T_3 = tail(_b_T_2, 1) @[main.scala 11:16]
    b <= _b_T_3 @[main.scala 11:5]
```

This pushes towards "all node names must be preserved" except for the now rare case of `_T_<int>`.

youngar, 2021-04-18:
This might be a bit outside the scope of the discussion, but should any name with a leading `_` (not a `_T`) be treated as a temporary for optimization?

If the `_T` names are few, then we can simplify by just keeping around all names.

```
instances -> required
memories  -> required
nodes     -> "_" nixed in the parser, otherwise required
wires     -> required
registers -> required
```

Does this sound ok?

youngar, 2021-04-20:
We are finding that the DontTouchAnnotation can be applied to things with temporary names, and that the annotation implies that those names shouldn't be dropped.   PR for this here: https://github.com/llvm/circt/pull/938/files.

youngar, 2021-04-22:
I think we can close this. Right now the name strings are non-optional, but can be empty "".  Temporary names like "_T_0" are stripped out by the parser unless the object has a DontTouchAnnotation.  We might revisit this issue if we discover that we can't strip out any names, ever, but this is working for us so far.

closed 2021-04-22 by youngar

lattner, 2021-04-25:
> I think we can close this. Right now the name strings are non-optional, but can be empty "". Temporary names like "_T_0" are stripped out by the parser unless the object has a DontTouchAnnotation. [...]

This sounds great, thank you!