llvm/circt #872: [RTL/FIRRTL] Store port names more efficiently

lattner, 2021-04-02:
We currently store port name attributes in an extremely inefficient format, as generalized argument attributes.  Consider a module with three input ports like this:

`module @foo(%a: i1, %b: i1, %c: i1) {`

Storing these as argument attributes causes a ton of extra dictionaries to be created, and causes them to be indexed by strings, see the logic that builds this up in `buildModule` in RTLOps.cpp.  The argument attributes are stored something like this:

```
"rtl.module" () { name = @foo,  // name attribute
                            arg0 = { rtl.name = "a"}, 
                            arg1 = { rtl.name = "b"},
                            arg2 = { rtl.name = "c" }} 
```

This is four dictionaryattrs (one for each argument and one overall, and means the op level dictionary attr has #ports entries in it, which slows down everything that is looking up the name for a module - a fairly common operation.  This also means that we can't efficiently look up names by index, because we have to create the "arg1" strings and unique them to do a lookup.

Instead, we should store the names as a separate attribute with a single array attribute, something like:

```
"rtl.module" () { name = @foo,  // name attribute
                            portnames = [ "a", "b", "c"] }
```

This would allow directly indexing port names, make the # entries in the module dictionary constant time.  This would dramatically reduce memory usage and make the compiler significantly faster.

This affects both the RTL and the FIRRTL dialect modules.

seldridge, 2021-04-02:
This relates to annotation storage in FIRRTL, too. #823 adds a `firrtl.annotations` key to each arg attribute dictionary to deal with port annotations. I agree that using an array here is way better. Per-port Annotations could then be packed into a second ArrayAttr.

(The patch series in #823 is clean enough that port annotation support can be easily removed if a fix for this lands first.)

lattner, 2021-04-02:
I’m working on the rtl dialect part of this

teqdruid, 2021-04-02:
This seems like an efficiency issue in how MLIR stores arg/result attributes. Isn't this sort of thing specifically what arg/result attributes are for?

lattner, 2021-04-03:
There is nothing that says we need to use argument attributes to store names, we can choose to store them however we like.  However, I think that the way MLIR does arg attributes in general is bad too :-)

lattner, 2021-04-03:
RTL half of this is done in b6b3cb9f

prithayan, 2021-04-05:
I'm working on the FIRRTL half of this

lattner, 2021-04-05:
awesome, thanks

seldridge, 2021-04-28:
FIRRTL half is done with #926.

closed 2021-04-28 by seldridge