llvm/circt #879: Add Integration Tests for Icarus Verilog

seldridge, 2021-04-02:
Add support to integration tests for [Icarus Verilog](https://github.com/steveicarus/iverilog). This is the main open source Verilog _event based_ simulator that I'm aware of.

### Motivation

We'd like to guarantee that we can produce Verilog that can (1) be consumed by existing tools and (2) simulates correctly with existing tools. We currently have integration tests that do this for Questa and Verilator. It would be good to expand this as wide as tractable without introducing tons of maintenance overhead for us.

teqdruid, 2021-04-02:
My understanding is the iverilog SystemVerilog support is partial and
experimental, so we'll likely run into compatibility issues which
necessitate the sv "flavors" we keep discussing. I'm not saying we
shouldn't do it, but just pointing out a potential blocker.

hasheddan, 2021-04-27:
I'd be happy to work on this!

closed 2022-03-26 by nandor