llvm/circt #880: test/CAPI/ir.c failing with assertions enabled

lattner, 2021-04-02:
I see this when running tests:

```
$ build/bin/circt-capi-ir-test
@registration
0
@rtltypes
error: invalid element for rtl.inout type
Assertion failed: (succeeded(ConcreteT::verify(getDefaultDiagnosticEmitFn(ctx), args...))), function get, file ../llvm/mlir/include/mlir/IR/StorageUniquerSupport.h, line 97.
zsh: abort      /Users/chrisl/Projects/circt/build/bin/circt-capi-ir-test
```

It isn't obvious to me what is going on here though.  The element type looks like an integer type.  Perhaps things aren't getting registered correctly?

lattner, 2021-04-02:
@GeorgeLyon do you have any idea?

GeorgeLyon, 2021-04-02:
Hmm, I haven't touched this code in a while. I think I remember @mikeurbach adding something around inout types. Let me see if I can find the PR.

GeorgeLyon, 2021-04-02:
Looks like its this one: https://github.com/llvm/circt/pull/845

GeorgeLyon, 2021-04-02:
Hmm, I'm not able to reproduce:

```
📁 ~/Developer/Projects/CIRCT/main
🔸 git log --pretty=oneline | head -n1
1de1cfd0211de1c521cdd6361dac2a36345e42fa (HEAD -> main, circt/main) [ExportVerilog] Change NameCollector to handle instance port names in bulk.

📁 ~/Developer/Projects/CIRCT/main
🔸 build/bin/circt-capi-ir-test
@registration
0
@rtltypes
0
```

mikeurbach, 2021-04-02:
Hmm I can take a look. This test should be run on every PR/commit to main, as well as the full matrix of CMake options and compilers that we run nightly. It seems to pass in those cases, and on George's test. @lattner besides "assertions enabled" do you mind sharing all the CMake configurations you're using so we can try to reproduce?

teqdruid, 2021-04-02:
None of our CI tests run on Mac...

lattner, 2021-04-04:
Ok, I think I see something going on.  It appears to fail when I build MLIR with release mode.  When I build MLIR with debug mode, the problem goes away.  Unfortunately, that makes it harder to repro.  Can anyone else see this?

Here are the two cmake lines I use specifically:

LLVM:
```(cd $CIRCT_PATH/llvm/build; cmake -G Ninja ../llvm -DLLVM_ENABLE_PROJECTS="mlir" -DLLVM_TARGETS_TO_BUILD="X86"  -DLLVM_ENABLE_ASSERTIONS=ON -DCMAKE_BUILD_TYPE=Release)```

CIRCT:
```(cd $CIRCT_PATH/build; cmake -G Ninja .. -DMLIR_DIR=$CIRCT_PATH/llvm/build/lib/cmake/mlir -DLLVM_DIR=$CIRCT_PATH/llvm/build/lib/cmake/llvm -DLLVM_ENABLE_ASSERTIONS=ON -DCMAKE_BUILD_TYPE=DEBUG)```

lattner, 2021-04-04:
Incidentally, I think the root cause also has something to do with `TypeID::get<type>()` in mlir/lib/Support.

lattner, 2021-04-04:
Same thing happens with MLIR is debug mode and CIRCT in release mode.

GeorgeLyon, 2021-04-05:
`check-circt` is succeeding for me in Release at `1de1cfd0211de1c521cdd6361dac2a36345e42fa`
Here are the exact arguments I'm using: 
```
    "cmake.configureArgs": [
      // cmake
      "-DCMAKE_BUILD_TYPE=Release",
      "-DCMAKE_OSX_DEPLOYMENT_TARGET=10.15",

      // LLVM
      "-DLLVM_ENABLE_ASSERTIONS=ON",
      "-DLLVM_ENABLE_PROJECTS=mlir",
      "-DLLVM_TARGETS_TO_BUILD=host",

      // LLVM (performance)
      "-DLLVM_OPTIMIZED_TABLEGEN=ON",
      "-DLLVM_USE_SPLIT_DWARF=ON",

      // LLVM (ccache)
      "-DLLVM_CCACHE_BUILD=ON",

      // Circt
      "-DLLVM_EXTERNAL_PROJECTS=circt",
      "-DLLVM_EXTERNAL_CIRCT_SOURCE_DIR=${workspaceFolder:CIRCT}",
    ],
```

lattner, 2021-04-05:
@darthscsi is seeing something similar in his config:

```
******************** TEST 'CIRCT :: CAPI/ir.c' FAILED ********************
Script:
--
: 'RUN: at line 10';   /home/andrew/src/circt/build/bin/circt-capi-ir-test 2>&1 | /home/andrew/src/circt/llvm/build/bin/FileCheck /home/andrew/src/circt/test/CAPI/ir.c
--
Exit Code: 1
Command Output (stderr):
--
/home/andrew/src/circt/test/CAPI/ir.c:106:12: error: CHECK: expected string not found in input
 // CHECK: 0
           ^
<stdin>:3:10: note: scanning from here
@rtltypes
         ^
Input file: <stdin>
Check file: /home/andrew/src/circt/test/CAPI/ir.c
-dump-input=help explains the following input dump.
Input was:
<<<<<<
           1: @registration 
           2: 0 
           3: @rtltypes 
check:106              X error: no match found
           4: dialect has no registered attribute printing hook 
check:106     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           5: UNREACHABLE executed at /home/andrew/src/circt/llvm/mlir/include/mlir/IR/Dialect.h:95! 
check:106     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

darthscsi, 2021-04-05:
I'm seeing that with debug/debug builds w/ clang 11.0.1 and gold

lattner, 2021-04-05:
build/bin/circt-capi-ir-test is valgrind clean for me btw.

darthscsi, 2021-04-05:
I'm not seeing it a different machine with debug/debug and clang 10.0.0

darthscsi, 2021-04-05:
here is my failing config:
```
LLVM:
CC=clang CXX=clang++ cmake -G Ninja ../llvm \
    -DLLVM_ENABLE_PROJECTS="mlir" \
    -DLLVM_TARGETS_TO_BUILD="X86;RISCV" \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DLLVM_USE_LINKER=gold
    
    MLIR:
CC=clang CXX=clang++ cmake -G Ninja .. \
    -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir \
    -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm \
    -DLLVM_ENABLE_ASSERTIONS=ON \
    -DCMAKE_BUILD_TYPE=DEBUG \
    -DLLVM_USE_LINKER=gold
    ```

lattner, 2021-04-05:
George, are you able to repro it with a mixed debug/release build?

GeorgeLyon, 2021-04-06:
Yes, this fails in a mixed build for me. 

I'm at a bit of a loss as to what is happening here though, maybe some of the dialect-loading code is being optimized in a weird way?

GeorgeLyon, 2021-04-06:
Because it is CMake, I'm not 100% certain this did what I want, but I ran the MLIR build in "Release" with "-Onone" added to CMAKE_CXX_FLAGS and the failure persisted. This seems to indicate that the problem is more likely to be in an explicit `#if DEBUG`.

mikeurbach, 2021-04-07:
> Incidentally, I think the root cause also has something to do with TypeID::get<type>() in mlir/lib/Support.

This is making my palms sweat. I know something weird happened here before that Stella resolved. I am hopeful it is not whatever that was.

I am normally building a little differently, but I'm going to try and reproduce.

lattner, 2021-04-07:
Thank you Mike.  The key thin about TypeID::get is that it expects mangled names to be merged by the linker according to ODR rules.  Maybe there is something weird about the build that is breaking this?  @stellaraccident is this a battle you've previously fought?

mikeurbach, 2021-04-09:
Ok, sorry for the delay. I tried to reproduce this using the setup Chris posted here: https://github.com/llvm/circt/issues/880#issuecomment-813050898. I could _not_ repro, even with MLIR built in Release mode (with assertions) and CIRCT built in Debug mode (also with assertions).

This is on Ubuntu 20.04. Since I was following the commands exactly, I didn't specify anything about the C++ compiler or linker, and on my machine, that defaults to GCC 9.3.0 and GNU ld. I have a feeling it was implied to use Clang for the C++ compiler, and that's what I normally do, so I'll test that as well. Hopefully that will be a reproducer.

In the meantime, for those who hit this with their CMake variable / compiler setup. Would it make sense to just not run this test for now? https://github.com/llvm/circt/blob/4c4d9482f863d07676f2cfd1a7f0e70932011b84/test/CAPI/ir.c#L99

I can't confirm that will fix it, but that is relatively new, and if removing it sweeps the problem back under the rug for now, I guess we should revert it while we dig further.

GeorgeLyon, 2021-04-09:
I have no problem removing the test temporarily.

lattner, 2021-04-09:
I'm a bit confused: is this a broken test, or is this a problem the test is exposing?

GeorgeLyon, 2021-04-10:
I think it is the latter, but I don't want this to impact peoples workflows (especially if this is the only place we encounter this issue)

mikeurbach, 2021-04-10:
Yeah, I think it is the latter as well. The test runs on every PR and commit to master and passes with the cmake configuration our CI uses. So I think this issue is about not blocking development, which is why I suggested just commenting out that test or something, if it helps.

I still cannot reproduce this with Clang 12.0.0 and gold or lld. I can try installing an older Clang.

GeorgeLyon, 2021-04-10:
OK I dug a bit deeper and the issue seems to be here:
https://github.com/llvm/circt/blob/efb9239e271fa184ef13e8f1e4e4134f8d7e27da/lib/Dialect/RTL/RTLTypes.cpp#L50
`type.isa<IntegerType>()` is returning `false`, despite `type.dump()` printing `i8`. This is peculiar since the type is created using `IntegerType::get` here:

https://github.com/llvm/llvm-project/blob/f3bf5c053b06c756f75ff07517d1e54c48cc99c0/mlir/lib/CAPI/IR/BuiltinTypes.cpp#L29

Indeed, the two type IDs in this line are not equal:

https://github.com/llvm/llvm-project/blob/b614ada0e80fe1af00294e8460f987dc6a7e4d5b/mlir/include/mlir/IR/StorageUniquerSupport.h#L81

Another clue is that `mlirTypeIsAInteger` correctly returns `true` for `i8type`.

I'm not familiar enough with C++ to guess as to why this would be happening in release and not in debug, but this does have a spooky smell to it. My best guess is that because `getTypeID()` in `StorageUniquerSupport.h` yields a different value depending on whether or not the header file is imported as release or debug. I'm not sure why this hasn't surfaced for folks not using the C API.

GeorgeLyon, 2021-04-10:
This may also be relevant, even though it is about a different type of linkage: https://llvm.discourse.group/t/inconsistent-typeid-after-linking-libmlir-so/2122/3

teqdruid, 2021-06-28:
#1331 mitigates this.

closed 2021-06-28 by teqdruid

lattner, 2021-06-29:
Verified, thank you again @fyquah !!