llvm/circt #887: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

teqdruid, 2021-04-05:
A bunch of the integration tests are broken with error messages like this:

```
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:10:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @B(%a: i1 { rtl.inout }) -> (i1 {rtl.name = "b"}, i1 {rtl.name = "c"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:10:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @B(%a: i1 { rtl.inout }) -> (i1 {rtl.name = "b"}, i1 {rtl.name = "c"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:16:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @A(%d: i1, %e: i1) -> (i1 {rtl.name = "f"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:21:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @AAA(%d: i1, %e: i1) -> (i1 {rtl.name = "f"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:26:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @AB(%w: i1, %x: i1) -> (i1 {rtl.name = "y"}, i1 {rtl.name = "z"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:26:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @AB(%w: i1, %x: i1) -> (i1 {rtl.name = "y"}, i1 {rtl.name = "z"}) {
^
/home/jodemme/circt/integration_test/EmitVerilog/lint.mlir:32:1: error: unexpected error: 'rtl.module' op Found port without a name. Port names are required for Verilog synthesis.

rtl.module @shl(%a: i1) -> (i1 {rtl.name = "b"}) {
^

--
```

lattner, 2021-04-05:
This is because I changed the representation of port names in rtl.module.  I'll take a look later today when I have a chance.  Sorry for the breakage!

lattner, 2021-04-05:
actually, this was quick.  1429a9dd

closed 2021-04-05 by lattner

teqdruid, 2021-04-05:
Huh. The error message was different from the one I got for module.extern
for the same issue... whatever.